arendst/Tasmota #7554: Is it possible add support for INA3221?

itProfi, 2020-01-19:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
Is it possible to support sensor INA3221?

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
https://esphome.io/components/sensor/ina3221.html

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

mike2nl, 2020-01-20:
In principle this is a 4 channel INA219.
Next important thing is that all, only one not and very pricy, have the ground connected to each other. So not usable for solar installations as Example or you manipulate the PCB with a dremel and a knife (xacto).
Had done a lot of investigations on this.

itProfi, 2020-01-21:
Why 4 channels? It has 3 channels..But in other situations, it can lead to savings of as many as 2 esp8266 and 2 pieces of INA 319 (battery chargers, etc.)

Jason2866, 2020-01-22:
A PR is always welcome.

itProfi, 2020-01-23:
> A PR is always welcome.
If i can do this, i am not open this issue..)

Jason2866, 2020-01-23:
Maybe someone writes sometime a driver.  The INA219 is a supported.

TrDA-hab, 2020-01-27:
@itProfi 
Now it does not make sense only the time spent!

1. INA3221 modules that are sold on Aliexpress - now contain a big bug, and are inoperative.
2. Custom modules INA3221 that are sold on third-party resources have not been tested, and not contain positive reviews.
3. If you wish, you can create your own custom INA3221 modules and get positive challenges later.

For information only - Custom modules INA3221:
http://www.yesyes.info/index.php/electronics-products/ina3221-r100/
https://easyeda.com/explore/ina3221

closed 2020-01-27 by itProfi