arendst/Tasmota #7556: I2cSetDevice warning behavior

BASM, 2020-01-19:
### PROBLEM DESCRIPTION

Behavior in different i2s drivers implies different.

Some device implies:
* Check that the device responds in the bus (by I2cSetDevice), and then make checks to uniquely identify the device. Then do I2CSetActive (if success). One does not imply that I2cSetActive is done automatically.

Drivers: xdrv_28_pcf8574, xdsp_02_ssd1306, xdsp_03_matrix.ino, xdsp_07_sh1106, xnrg_07_ade7953, xsns_16_tsl2561, xsns_32_mpu6050, xsns_57_tsl2591, xsns_59_ds1624,
xsns_44_sps30, xsns_45_vl53l0x, xsns_46_MLX90614

* If device responds in the bus by I2cSetDevice, drivers imply that I2cSetActive is done automatically.

Drivers: xdrv_10_scripter, xdsp_01_lcd, xdsp_08_ILI9488, xdsp_10_RA8876, 

I think need remove I2cSetActive from function I2cSetDevice in the support.ino and fix all drivers,
which have mean automatic I2cSetActive.

I can fix it myself, but I think it's better for the author to do it. But I can do PR if you want.

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): all
- [x] Tasmota binary firmware version number used: all
  - [x] Pre-compiled
  - [x] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here: any
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here: none
```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here: none

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here: none
```

### TO REPRODUCE
lm75ad is block the ds1624 on the address 0x48 ever time.

### EXPECTED BEHAVIOUR

drivers must block only self devices

### SCREENSHOTS

none

### ADDITIONAL CONTEXT

none

arendst, 2020-01-19:
To solve your issue you need to use command ``I2cDriver`` to disable the LM75AD from the initial probe.

BASM, 2020-01-19:
Each driver should only take devices that it explicitly supports. At now in almost all drivers this is implemented successful. But they block addreses because they misunderstand the logic of the I2cSetDevice function. The taking of the address occurs before checking the correctness of the device.

Command I2cDrivers solve only my problem.  Will not solve the problem as a whole.

BASM, 2020-01-19:
I can do PR if you want.

arendst, 2020-01-19:
Yes pls do so I can see what you mean.

ascillato2, 2020-01-19:
Closing this issue as it is solved. Thanks.

closed 2020-01-19 by ascillato2