arendst/Tasmota #7559: Invert WS2812 Data Signal

tomy42, 2020-01-19:
**Have you looked for this feature in other issues and in the docs?**  
Yes, I looked in the Tasmota documentation if it is possible to change the behavior. 

**Is your feature request related to a problem? Please describe.**  
I want to control a WS2812 with an RGB PWM controller

**Describe the solution you'd like**  
Invert the WS2812 Data Signal on the GPIO Pin. So it is possible with cheap LED RGB PWM devices such as MagicHome RGB ZJ-WFMN-A V1.1 LED controller to control a WS2812 LED. You only need a pullup resistor on the data line.

**Describe alternatives you've considered**  
to use another LED Pixel Controller (Magic House Wifi Pixel 5V 3pin), but this has an additional microcontroller on the board to control the WS2812 LED. I have not found any information on how to control this microcontroller.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-01-22:
Hi, 

I like your idea because it can make a magichome device to control a ws2812 through the actual output it has. But the control of the gpio is being done by the neopixelbus library. AFAIK, seems that this library doesn't have that option. So, I think it should be better to address this to the library developers and then just update the library into Tasmota.
Neopixelbus library: https://github.com/Makuna/NeoPixelBus

ascillato2, 2020-01-24:
@tomy42 

Hi, were you able to ask in the NeoPixelBus Libray repository?

ascillato2, 2020-02-02:
@tomy42

Hi, any news on this? Are you still interested on this?

tomy42, 2020-02-02:
@ascillato2 

Oh Sorry, i was on vacation and had no time to look at NeoPixelBus Library repository.

Today I checked there, in the Latest Features list on October 17th, 2019 the feature "Inverted signal support, makes it easy for a cheap and simple level shifter" was listed.
Wouldn't it be exactly what you would need?

I only found nothing about it in the API Documentation. But I don't know that very well :-(

greetings

ascillato2, 2020-02-10:
It is supported in the library: https://github.com/Makuna/NeoPixelBus/issues/174#issuecomment-536209816

but if you look in older issue https://github.com/arendst/Tasmota/issues/3771 you will find that the timming was not reliable and that the final solution was a direct connection. So a software inverting was never required.

closed 2020-02-10 by ascillato2

tomy42, 2020-02-11:
Yes, a direct connection is possible, I had already tested that. But it works with an incorrect voltage level of 3.3V. The WS2812 works with a voltage level of 5V. So my idea was to use the FET from the PWM output with a pullup resistor to 5V supply voltage from the WS2812. But for that the signal would have to be inverted.

Another idea is to use the LED SPI controller, remove the microcontroller there and connect the 74HC245 bus transeiver directly to the ESP8266.

ascillato, 2020-02-11:
> WS2812 works with a voltage level of 5V

Sorrt, that is wrong. Data pin can be controlled directly from any gpio of the esp8266 with 3.3v. Only the power of the strip must be 5v

tomy42, 2020-02-12:
Yes, it works, but it's (just) out of the range that is first specified in the data sheet for a WS2812B.
VDD range: +3.5 - + 5.3 V
VIH Range: 0.7 * VDD - VDD + 0.5V
VIL Range: 0 - 0.3 VDD
By VDD = 5V it is just outside the range

The better way will be to make your own board with all the other components you need. But the quickest way would have been to use an existing board, even if it doesn't fit 100%.