arendst/Tasmota #7560: Firmware upgrade from file fails with Upload buffer miscompare

ed-holland, 2020-01-19:
### PROBLEM DESCRIPTION
Started with compiled tasmota 8.1 with only changes to my_user_config.h. It flashes and starts without a problem. After some adjustments to my_user_config.h and build, the upload of the firmware fails with the above message.
Flashing the binary is no problem and works as expected.
Even uploading the same version fails with the same message.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _8.1____
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: platform.io
- [X] Flashing tools used: tasmonizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
18:46:42 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
18:46:42 SRC: Backlog
18:46:42 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
18:46:42 MQT: tasmota/sonoff/sonoff_s20_5014/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
18:46:42 SRC: Backlog
18:46:42 CMD: Group 0, Index 1, Command "MODULE", Data ""
18:46:42 MQT: tasmota/sonoff/sonoff_s20_5014/stat/RESULT = {"Module":{"8":"Sonoff S2X"}}
18:46:42 SRC: Backlog
18:46:42 CMD: Group 0, Index 1, Command "GPIO", Data "255"
18:46:42 MQT: tasmota/sonoff/sonoff_s20_5014/stat/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
18:48:35 MQT: tasmota/sonoff/sonoff_s20_5014/stat/STATUS4 = {"StatusMEM":{"ProgramSize":600,"Free":400,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","8FDAE3D7","043683A1","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
18:48:35 MQT: tasmota/sonoff/sonoff_s20_5014/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff-s20-5014","IPAddress":"10.250.2.193","Gateway":"10.250.2.253","Subnetmask":"255.255.255.0","DNSServer":"10.250.2.247","Mac":"CC:50:E3:57:53:96","Webserver":2,"WifiConfig":4}}
18:48:35 MQT: tasmota/sonoff/sonoff_s20_5014/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.250.2.200","MqttPort":8883,"MqttClientMask":"sonoff_s20_%04d","MqttClient":"sonoff_s20_5014","MqttUser":"sonoff","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
18:48:35 MQT: tasmota/sonoff/sonoff_s20_5014/stat/STATUS7 = {"StatusTIM":{"UTC":"Sun Jan 19 17:48:35 2020","Local":"Sun Jan 19 18:48:35 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":99,"Sunrise":"08:11","Sunset":"16:59"}}
18:48:35 MQT: tasmota/sonoff/sonoff_s20_5014/stat/STATUS10 = {"StatusSNS":{"Time":"2020-01-19T18:48:35"}}
18:48:35 MQT: tasmota/sonoff/sonoff_s20_5014/stat/STATUS11 = {"StatusSTS":{"Time":"2020-01-19T18:48:35","Uptime":"0T00:09:06","UptimeSec":546,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WENMAKERS","BSSId":"98:9B:CB:65:9C:BB","Channel":6,"RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Upload firmware

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Weblog logging set to More Debug:
18:50:43 HTP: Firmware Upgrade
18:50:49 CFG: Saved to flash at FB, Count 10, Bytes 4096
18:50:49 UPL: File tasmota.bin ...
18:51:15 HTP: Upload done
18:51:15 MQT: Attempting connection...
**(Please, remember to close the issue when the problem has been addressed)**

closed 2020-01-19 by ascillato2

ascillato, 2020-01-19:
If your firmware is big, you need to upload it gzipped or you need to upload minimal firmware first to free flash memory and then OTA the your full version.

Duplicated issue. Further explanations in the docs.

hvbommel, 2020-03-23:
I experienced the same problem indeed. Installing the tasmota-minimal.bin first followed by the tasmota-with-tls.bin, both OTA, worked fine.

smartboart, 2020-05-14:
same problem from udate tasmota sensors 8.2.0 to tasmota sensors 8.3.0 and i also installed minimal bin before.

ChrisVonFrosta, 2020-05-16:
Same Problem.
Come from 8.2.0. Upgrade to 8.3.0 minimal -> fine. Upgrade to 8.3.0 Sensors -> failed
"Upload Failed
Upload buffer miscompare"

5chufti, 2020-05-20:
had similar problem.
1.) do OTA upgrade to "http://thehackbox.org/tasmota/release/tasmota-minimal.bin"
2.) do upgrade to your own fw, preferably gzipped (with e.g. 7zip)
somehow selecting e.g. "http://thehackbox.org/tasmota/release/tasmota-minimal.bin.gz" for OTA didn't work for me (wrong bin header ...)

smartboart, 2020-05-20:
8.3.1 should work

sbroersen, 2020-05-20:
I just loaded 8.3.1, but I still have the same problem. I tried uploading tasmota-minimal.bin by file or by http, but both times I get an buffer miscompare error. 

The information page gives the following information.
```
Program Version	8.3.1(tasmota)
Build Date & Time	2020-05-20T14:07:50
Core/SDK Version	2_6_3/2.2.2-dev(38a443e)
Uptime	0T00:32:36
Flash write Count	116 at 0xFB000
Boot Count	20
Restart Reason	External System
Friendly Name 1	Tasmota
 
AP1 SSId (RSSI)	xxxxx (52%, -74 dBm)
Hostname	Tasmota-test
IP Address	192.168.178.198
Gateway	192.168.178.1
Subnet Mask	255.255.255.0
DNS Server	84.116.46.20
MAC Address	2C:F4:32:44:67:87
 
MQTT Host	192.168.178.63
MQTT Port	1883
MQTT User	mqtt
MQTT Client	DVES_446787
MQTT Topic	tasmota-test
MQTT Group Topic 1	cmnd/tasmotas/
MQTT Full Topic	cmnd/tasmota-test/
MQTT Fallback Topic	cmnd/DVES_446787_fb/
 
Emulation	None
mDNS Discovery	Disabled
 
ESP Chip Id	4482951
Flash Chip Id	0x164020
Flash Size	4096kB
Program Flash Size	4096kB
Program Size	672kB
Free Program Space	264kB
Free Memory	23kB

```

meingraham, 2020-05-20:
@Quest24 

It does not look like you using the pre-compiled binaries. Your binary is using an old Arduino Core (2.6.3).

Quest24, 2020-05-20:
Yes, that is correct. I will update this to the latest version. Are you sure this is the problem?

meingraham, 2020-05-20:
It's one variable to eliminate.

Even if you want to eventually use binaries that you compile yourself, for testing to see if it alleviates your issue you should try the pre-compiled binary - http://thehackbox.org/tasmota/tasmota.bin. If it does, then you can start isolating what part of your compilation may be causing the issue.

Quest24, 2020-05-20:
I upgraded core to 2.7.1. No effect. It must be something else.  When I update by file I get  an buffer miscompare error. When I update by HTTP to tasmota-minimal.bin, I get no error message, but the device just restarts with the old version.

meingraham, 2020-05-20:
Did you try a pre-compiled binary yet?

thamidi, 2020-05-22:
On a NodeMCU, I was getting **Upload buffer miscompare** when trying to upgrade the firmware via OTA from 6.7.1 to 8.1.0 Firmware.
My fix was described above and I just upload the tasmota-minimal firmware and after that I was able to successfully upgrade to 8.1.0,
Thanks to the good folks on this thread.

Quest24, 2020-05-22:
The answer is very simple. The free program space in your device needs to more than the tasmota-minimal.bin size of 378kb. So I switched of all options I don't really need. Once the free program space was  380kb it worked perfect. As you can see in my information dump above, the free space was only 264kb. Case closed for me.

meingraham, 2020-05-22:
Note that if you are on 8.2 or greater, you can used the gzip binaries thus need less free space to a new binary uploaded.

@Quest24 
- 264KB free! Wow, you must have a LOT of featured compiled into your binary. I would investigate whether you are actually using all those features.
- Case closed until you find that you _need_ to upgrade to a newer version. Just leaves a bit of the research for your setup not quite resolved.

Cheers!

Quest24, 2020-05-23:
Thnx for the tip. I tested with gzip files. Works perfect in the 8.3.1. version, but 8.2.0 gives error #define D_UPLOAD_ERR_6. I don't know what that means.

refthoom, 2020-06-10:
Tested OK with precompiled, compressed bin.gz from v8.2.0 to 8.3.1 with file upload in webUI on a Sonoff Basic. 
When using uncompressed bin file I get the TS error "Upload buffer miscompare".

reynos, 2020-07-13:
Careful folks... I followed this advice with 8.3.1, uploaded a bin.gz and seem to have bricked my RBG Bulb. Any ideas on recovery?

5chufti, 2020-07-15:
you are sure it was a compatible fw? In general zipping yourself is not a problem and if it was bad, you should have got at least sthg like "wrong bin header". If it does no longer connect to wifi the only way would be physical access to the board.

eleutscher, 2020-08-17:
Had a simular issue. Could update to the full version but not to the sensor version thru 'Upgrade by file upload'. Had to use OTA URL http://thehackbox.org/tasmota/release/tasmota-minimal.bin

gkawanami, 2020-09-20:
Guys.
If I flash tasmota-minimal, will I be able to flash tasmota.bin OTA afterwards?  How, because I understand that mininal does not create a wifi connection...

ascillato, 2020-09-20:
Minimal will connect to wifi. It is ONLY for upgrading.

Minimal is not for first flash
The first time, you need to flash Tasmota.bin.
Afterwards, if you want to upgrade, you need to upgrade to minimal to free space and then to the new version. 

Anyway, if you use OTAurl, this is not needed because Tasmota will do that automatically 

Also, if you upload a compressed firmware like Tasmota.bin.gz, you don't need minimal and you can do that in just one step. The firmware will be decompressed automatically inside your chip.

gkawanami, 2020-09-20:
Hey Adrian. Thanks for the response.
I have some Sonoff minis flashed with 8.3.1, but I´m trying to upgrade them to Tasmota-Hubitat 8.3.1.
I tried flashing by uploading a gzip file, but I got the "UpLoad Buffer Miscompare" message.

I have then flashed it with Tasmota minimal, but I can´t connect to it anymore. The device is creating an ESP_*** wifi network, but when I connect to it the interface does not show up, even using the router and/or it´s IP adress...

Any tips / tricks?

ascillato, 2020-09-20:
Minimal is only for updating using Tasmota webUI. If you have flashed minimal, your device is bricked and you need to erase by serial and flash by serial. You should now use directly the firmware you want to be flashed by serial.

gkawanami, 2020-09-20:
OK, I´m sorry I´m taking a lot of your time...
Just to make things clear so that I brick another unit.... (Making it OTA)
. I should open Tasmota menu thru the Mini´s IP 
. Choose update firmware through file upload and upload tasmota minimal (OTA)
. Wait for it to install Tasmota minimal and then do what?  How will I be able to upload the new final firmware OTA?

ascillato, 2020-09-20:
To start, you need to flash by serial your device with Tasmota.bin
IF YOU FLASH TASMOTA-MINIMAL.BIN, YOUR DEVICE WON'T WORK due to Tasmota-minimal.bin is ONLY for in-between updates and ONLY when you upload firmwares by OTA-File that are bigger than the available free flash space. Minimal must not be used when updating using OTAurl, when using compressed firmwares or when you have enough free flash space (like having in flash tasmota-lite.bin and updating to a newer tasmota-lite.bin)

Then, for updating:
From the Tasmota WebUI, you have 2 options for updating:

1- From an URL
2- From a file

* From an URL (that is OTA URL) forget about minimal. Tasmota will use what is called OTAMagic and will download minimal automatically and then the file you asked for. The official OTA URL are: http://ota.tasmota.com/tasmota/tasmota.bin (for latest version-recommended) and http://ota.tasmota.com/tasmota/release/tasmota.bin (for latest release).

* From file, and form a working device, you have to upgrade to latest tasmota-minimal.bin. After it restarts, you have to go to the WEBUI (will work because it uses the stored wifi settings - settings are not erased when updating) and go to firmware update and upload the firmware you want.

gkawanami, 2020-09-20:
Adrian,
Did from URL: Minimal flashed fine.
When I try flashing the new tasmota it reponds Upload Buffer Miscompare again.  Tried even with the original tasmota lite....

Any chance I should type a console command to wipe residual stuff?

gkawanami, 2020-09-20:
Just tried Tasmota-lite-8.1.0 and still got the Upload buffer miscompare messsage....

robustini, 2021-01-27:
Is there a way to make the version with the previously flashed language installed during OTA, or OTA only writes the default version in English?

ascillato, 2021-01-27:
The language is in the file your are updating by OTA.

If you flash Tasmota.bin, you will have it in English.
If you flash Tasmota-ES.bin you will have it in Spanish and so on. Please, check the releases notes for the explanation of each different firmware type.

robustini, 2021-01-27:
> 
> 
> The language is in the file your are updating by OTA.
> 
> If you flash Tasmota.bin, you will have it in English.
> If you flash Tasmota-ES.bin you will have it in Spanish and so on. Please, check the releases notes for the explanation of each different firmware type.

I know this, my question is another.

ascillato, 2021-01-27:
If you want the IT version just go to the Tasmota UI, click on firmware update, put http://ota.tasmota.com/tasmota/tasmota-IT.bin.gz and then click on update.

This will be saved so, for the next update whenever you want to update, just click on update and you will always have the IT version.

This is just a one-time configuration. 

By default it uses Tasmota.bin and you must tell Tasmota which other version you want (other language, sensors version, lite, KNX, etc etc)

robustini, 2021-01-27:
Verified now, saved after the reboot, thanks!

Gemax22, 2021-11-28:
What I did was uploading the minimal first but the GZ version 
http://ota.tasmota.com/tasmota/tasmota-minimal.bin.gz

was the only one that didn't give any error then the Sensor
http://ota.tasmota.com/tasmota/tasmota-sensors.bin

And I jump from 8.3.6.1 to 10.0.0.3

barbudor, 2021-11-28:
Please follow exactly the upgrade path as described in the doc using  the compressed binary tasmota-lite.bin.gz

https://tasmota.github.io/docs/Upgrading/