arendst/Tasmota #7562: witty and no AP coming up web server active on tasmota-4082 with IP address (IP unset)

riker65, 2020-01-20:
Hi

I flashed Tasmota on a witty via OTA .

Witty was formerly running ESPEasy firmware.

AP is not coming up. Do not see the SSID.

Tasmota Version 8.1.0.2(tasmota)-2_6_1

`web server active on tasmota-4082 with IP address (IP unset)`

Did I miss anything?

see serial log with Status 0

After running status 0 the witty restarted.

Thanks Thomas

`07:20:43.943 -> 00:00:00 CFG: Loaded from flash at F6, Count 6
07:20:43.943 -> 00:00:00 Project tasmota Tasmota Version 8.1.0.2(tasmota)-2_6_1
07:20:44.109 -> 00:00:00 WIF: WifiManager active for 3 minutes
07:20:44.607 -> 00:00:00 HTP: Web server active on tasmota-4082 with IP address (IP unset)
07:23:46.374 -> 00:03:02 APP: Restarting
07:23:46.473 -> 
07:23:46.473 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
07:23:46.473 -> 
07:23:46.507 -> load 0x4010f000, len 1384, room 16 
07:23:46.507 -> tail 8
07:23:46.507 -> chksum 0x2d
07:23:46.507 -> csum 0x2d
07:23:46.507 -> v482516e3
07:23:46.507 -> ~ld
07:23:46.639 -> 00:00:00 CFG: Loaded from flash at F5, Count 7
07:23:46.639 -> 00:00:00 Project tasmota Tasmota Version 8.1.0.2(tasmota)-2_6_1
07:23:46.772 -> 00:00:00 WIF: WifiManager active for 3 minutes
07:23:47.304 -> 00:00:00 HTP: Web server active on tasmota-4082 with IP address (IP unset)
07:26:49.077 -> 00:03:02 APP: Restarting
07:26:49.176 -> 
07:26:49.176 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
07:26:49.176 -> 
07:26:49.209 -> load 0x4010f000, len 1384, room 16 
07:26:49.209 -> tail 8
07:26:49.209 -> chksum 0x2d
07:26:49.209 -> csum 0x2d
07:26:49.209 -> v482516e3
07:26:49.209 -> ~ld
07:26:49.308 -> 00:00:00 CFG: Loaded from flash at F4, Count 8
07:26:49.342 -> 00:00:00 Project tasmota Tasmota Version 8.1.0.2(tasmota)-2_6_1
07:26:49.474 -> 00:00:00 WIF: WifiManager active for 3 minutes
07:26:49.974 -> 00:00:00 HTP: Web server active on tasmota-4082 with IP address (IP unset)
07:29:51.779 -> 00:03:02 APP: Restarting
07:29:51.856 -> 
07:29:51.856 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
07:29:51.856 -> 
07:29:51.935 -> load 0x4010f000, len 1384, room 16 
07:29:51.935 -> tail 8
07:29:51.935 -> chksum 0x2d
07:29:51.935 -> csum 0x2d
07:29:51.935 -> v482516e3
07:29:51.935 -> ~ld
07:29:52.014 -> 00:00:00 CFG: Loaded from flash at FB, Count 9
07:29:52.014 -> 00:00:00 Project tasmota Tasmota Version 8.1.0.2(tasmota)-2_6_1
07:29:52.179 -> 00:00:00 WIF: WifiManager active for 3 minutes
07:29:52.681 -> 00:00:00 HTP: Web server active on tasmota-4082 with IP address (IP unset)
07:32:45.141 -> 00:02:53 CMD: status 0
07:32:45.141 -> 00:02:53 RSL: stat/tasmota/STATUS = {"Status":{"Module":18,"FriendlyName":["Tasmota"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
07:32:45.174 -> 00:02:53 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:02:54","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":8,"SaveCount":10,"SaveAddress":"FA000"}}
07:32:45.257 -> 00:02:53 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.1.0.2(tasmota)","BuildDateTime":"2019-12-30T19:07:34","Boot":17,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"287/699"}}
07:32:45.257 -> 00:02:53 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
07:32:45.257 -> 00:02:53 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":452,"Free":548,"Heap":34,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0F400082","04048000","00000005","000000C0","00000080"],"Drivers":"1,2,3,4,9,10,12,16","Sensors":"1,2,3"}}
07:32:45.257 -> 00:02:53 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-4082","IPAddress":"(IP unset)","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"4C:11:AE:0E:8F:F2","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
07:32:45.311 -> 00:02:53 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0E8FF2","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
07:32:45.311 -> 00:02:53 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:02:53 1970","Local":"Thu Jan 01 00:02:53 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:03"}}
07:32:45.342 -> 00:02:53 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:02:53"}}
07:32:45.342 -> 00:02:53 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:02:53","Uptime":"0T00:02:54","UptimeSec":174,"Heap":34,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"","BSSId":"B1:C8:A1:D8:91:E8","Channel":1,"RSSI":100,"Signal":31,"LinkCount":0,"Downtime":"0T00:00:00"}}}
07:32:54.686 -> 00:03:02 APP: Restarting
07:32:54.786 -> 
07:32:54.786 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
07:32:54.786 -> 
07:32:54.842 -> load 0x4010f000, len 1384, room 16 
07:32:54.842 -> tail 8
07:32:54.842 -> chksum 0x2d
07:32:54.842 -> csum 0x2d
07:32:54.842 -> v482516e3
07:32:54.842 -> ~ld
07:32:54.952 -> 00:00:00 CFG: Loaded from flash at FA, Count 10
07:32:54.952 -> 00:00:00 Project tasmota Tasmota Version 8.1.0.2(tasmota)-2_6_1
07:32:55.118 -> 00:00:00 WIF: WifiManager active for 3 minutes
07:32:55.617 -> 00:00:00 HTP: Web server active on tasmota-4082 with IP address (IP unset)
`

closed 2020-01-20 by close-issue-app[bot]

Jason2866, 2020-01-20:
yes ;-)  Just look for the AccessPoint Tasmota-xxxx to configure wifi.
All described in documentation. http://tasmota.com

riker65, 2020-01-20:
Hi 

thanks

that is the problem AP not coming up....no Tasmota-xxxx while scanning for APs

ascillato, 2020-01-20:
As your device had another firmware before, may be some incorrect data has been left in the sdk area (as seems that the sdk used was another version) you will need to erase that part.

You can try command reset 3, then after it reboots, you need to de-energize the device and energize it again in order to let the sdk to recalibrate itself.

If you need further assistant on that, please address this to the Tasmota support chat. Thanks