arendst/Tasmota #7563: Problem in makeing energy data showing for a new (unsupported) socket - Burnett G1

sztsian, 2020-01-20:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Problem trying to make energy data showing for a new (unsupported) socket - Burnett BN-G1

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): __Burnett BN-G1__
- [x] Tasmota binary firmware version number used: __8.1.0__
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: __`esptool --port /dev/ttyUSB2 write_flash -fs 1MB -fm dout 0x0 tasmota.bin `under Fedora 31___
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
Module:
 {"Module":{"0":"BN-G1"}}
GPIO 255: 
{"GPIO0":{"146":"CSE7766 Rx"},"GPIO1":{"0":"None"},"GPIO2":{"145":"CSE7766 Tx"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

Backlog Template:
I've tried both
{"NAME":"BN-G1","GPIO":[145,0,146,0,0,0,0,0,21,56,0,0,0],"FLAG":0,"BASE":41}

And

{"NAME":"BN-G1","GPIO":[146,0,145,0,0,0,0,0,21,56,0,0,0],"FLAG":0,"BASE":41}

Unfortunately there is no energy status shows for any of the two.
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
I masked MAC address

15:54:01 CMD: STATUS 0
15:54:01 RSL: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"tasmota","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:54:01 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:02:21","StartupUTC":"2020-01-20T14:51:40","Sleep":50,"CfgHolder":4617,"BootCount":25,"SaveCount":77,"SaveAddress":"FB000"}}
15:54:01 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"324/699"}}
15:54:01 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"MqttLog":0,"SysLog":3,"LogHost":"","LogPort":514,"SSId":["d2-825f","TP-LINK_E87910"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
15:54:01 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
15:54:01 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota","IPAddress":"192.168.130.150","Gateway":"192.168.130.129","Subnetmask":"255.255.255.128","DNSServer":"192.168.130.129","Mac":"xx:xx:xx:xx:xx:xx","Webserver":2,"WifiConfig":4}}
15:54:01 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_BCEDBD","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
15:54:01 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Mon Jan 20 14:54:01 2020","Local":"Mon Jan 20 15:54:01 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:34","Sunset":"17:27"}}
15:54:01 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-01-20T15:54:01"}}
15:54:01 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-01-20T15:54:01","Uptime":"0T00:02:21","UptimeSec":141,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":2,"SSId":"TP-LINK_E87910","BSSId":"yy:yy:yy:yy:yy:yy","Channel":6,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
00:00:00 CFG: Loaded from flash at F9, Count 79
00:00:00 QPC: Flag 0E
00:00:00 CFG: CR 324/699
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 8.1.0(tasmota)-2_6_1
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP2 TP-LINK_E87910 in mode 11N as tasmota...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Attempting connection...
00:00:06 QPC: Reset
00:00:07 WIF: Checking connection...
00:00:07 WIF: Connected
00:00:07 HTP: Web server active on tasmota-3517 with IP address 192.168.130.150
15:57:01 HTP: Main Menu
15:57:01 NTP: Drift 0, (UTC) Mon Jan 20 14:57:01 2020, (DST) Sun Mar 29 02:00:00 2020, (STD) Sun Oct 25 03:00:00 2020
15:57:01 APP: Boot Count 26
15:57:01 CFG: Saved to flash at F8, Count 80, Bytes 4096
15:57:03 HTP: Console
15:57:14 HTP: Main Menu
15:57:16 HTP: Console
15:57:20 WIF: Checking connection...
15:57:20 WIF: Connected

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

* Flash the Burnett BN-G1 like [what I do for BN-K01 before](https://github.com/arendst/Tasmota/wiki/Burnett-BN-K01).
* Config wifi
* Config module like shown above
* Check the main page to see if power consumption shows up or not

### EXPECTED BEHAVIOUR
_The energy status should be shown._

### SCREENSHOTS
_
[image]
._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
This is a newly bought device. I tried the official app and energy consumption can be shown there.
I teared it down and confirm it is using a **CSE7759B** energy chip. I see it is not listed in the supported sensors so I tried a much similar CSE7766. Do note, I find that the CSE7759B have the TI pin (pin 6) is connected to GPIO0 along with the button. After configed like above, there is no energy status showing up.

I'd like to know if CSE7759B is supported or not. If it is supported, is there anything wrong with my setup or am I missing anything?

Thanks.
**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-01-20:
Sonoff POW v.2 uses this chip too. So in general it is supported

sztsian, 2020-01-21:
I see. So in this way, my question is if this is a bug, or this is my config issue.

Jason2866, 2020-01-21:
Since POW 2 is working fine, it is the config

arendst, 2020-01-21:
Try Sonoff Pow R2 module

arendst, 2020-01-21:
The current CSE7766/CSE7759 driver ONLY supports hardware serial and therefore only GPIO01 and GPIO03

arendst, 2020-01-21:
If Pow R2 doesn't work re-try your current settings with the latest dev release.
This version does allow SoftwareSerial and should work on GPIO0 and GPIO2.

sztsian, 2020-01-22:
Oh thanks for the feedback.
I think I probably forgot to mention, there is nothing connected to the RI pin of the CSE7759B. I set it to GPIO2 only because I was wondering if I need to set both to make it work.

I'm travelling this week and will test and comment again in early Feb.

arendst, 2020-01-22:
OK.

With the latest changes you only need to configure ``CSE7766 Rx (146)``. The Tx is not needed anymore.

ascillato2, 2020-01-24:
@sztsian 

Hi, any news on this?

sztsian, 2020-01-29:
> OK.
> 
> With the latest changes you only need to configure `CSE7766 Rx (146)`. The Tx is not needed anymore.

Thanks for the great work!
With the git head of today (I compiled from tasmocompiler container), the energy status shows up. And I find that the pin connection I detected using a multi-meter is not exact what it need on the template. Now I am using
`{"NAME":"BN-G1","GPIO":[17,0,0,146,0,0,0,0,21,56,0,0,0],"FLAG":0,"BASE":41}`
Which means the Rx is on GPIO3.
I flashed back to the origin 8.1.0 and use GPIO3 for CSE7766 Rx and random pin for CSE7766 Tx, it also works. So this means my detection using the multi-meter is wrong in the beginning. Sorry for the extra work.

I'm closing the issue since my origin question is solved. I'll post the guide for this device shortly.

sztsian, 2020-01-29:
> Try Sonoff Pow R2 module

Oh also tried Sonoff Pow R2, it works too.

closed 2020-01-29 by sztsian