arendst/Tasmota #7566: Revert to original Sonoff firmware possible? How?

Stooovie, 2020-01-21:
Hi, is it possible to revert from Tasmota to original Sonoff firmware? I have a TH10 with AM2301 sensor that worked perfectly prior to flashing Tasmota but doesn’t work at all now. It definitely looks like a software issue and I’d like to try fixing it with reverting to original Sonoff FW. How would I do it? Thanks.

closed 2020-01-21 by close-issue-app[bot]

meingraham, 2020-01-21:
Only if you made a backup of the firmware binary from that specific device. If you did, you can flash it back via the serial programming interface.

fastiuk, 2023-12-09:
> Only if you made a backup of the firmware binary from that specific device. If you did, you can flash it back via the serial programming interface.

From that specific device, or from that device model?
For example, can I flash Sonoff Basic 2 firmware I dumped to other Sonoff Basic 2 device, or only that one I dumped from?

barbudor, 2023-12-09:
It can vary from device to device
On some you need the exact dump of the device
On others any binary saved from the same model could work

But the trends is the 1st case as many time the binary dump is not identical from one device to the other.