arendst/Tasmota #7567: ESP32 flash

pepeEL, 2020-01-21:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  
Hi
Is any chance to flash ESP32 with tasmota ?

meingraham, 2020-01-21:
No. Only the ESP8266 & ESP8285 can be flashed with Tasmota. The ESP32 is a completely different chip architecture.

closed 2020-01-21 by meingraham

pepeEL, 2020-01-21:
But is any chance in future to implement also support this ESP32 ?

meingraham, 2020-01-21:
Not likely. Tasmota was forked and ported a couple of years ago but that project withered and died.

pepeEL, 2020-01-21:
Ok thanks