arendst/Tasmota #7569: Add Timers status

ppetppet, 2020-01-21:
**Have you looked for this feature in other issues and in the docs?**  
Yes, I didn't find it.

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I want to get timers status locally and display them using LED. I want to set timers status (on or off) using button attached to the device.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

Right now you have 3 options:
Timers | Timers control
0 = disable all timers 
1 = enable all timers
2 = toggle all timers
Is it possible to add fourth option:
3 = get timers status

**Describe alternatives you've considered**  

_A clear and concise description of any alternative solutions or features you've considered._
Using Mqtt, but I want to use device locally only.
**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2020-01-21:
Sending the command `Timer<x>` without parameters returns the status of timer `<x>`.
Sending the command `Timers` (without parameter) returns the status of all timers.

closed 2020-01-21 by meingraham

ppetppet, 2020-01-21:
This solution is not usable using rules.

meingraham, 2020-01-21:
You edited the original post to add rules and change your request:
```
I want to get timers status locally and display them using LED. I want to set timers status (on or off) using button attached to the device.

Describe the solution you'd like

Right now you have 3 options:
Timers | Timers control
0 = disable all timers 
1 = enable all timers
2 = toggle all timers
Is it possible to add fourth option:
3 = get timers status
```

For what you want, you may be better off managing the timers with an automation hub like openHAB or Home Assistant.

meingraham, 2020-01-21:
Yes, you want everything locally controlled. And you want LED outputs and button inputs for each timer. You may have to compromise and use a HAB to maintain each timer's state. Or you could use the MQTT broker to retain each timer's state.

If you want complete autonomous operation without MQTT or a HAB, ... then code probably needs to be modified, i.e., you need to fork the repository and propose a PR.

@ascillato would something like `ON Timer1#Data.Arm` be possible with the current command?
```
CMD: Timer1
MQT: /RESULT = {"Timer1":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0}}
```

ppetppet, 2020-01-21:
I want to know state of timers (all of them), not the individual one.

meingraham, 2020-01-21:
Yes... just using `ON Timer1#Data.Arm` for discussion. If this is a valid approach, then `ON Timer<x>#Data.Arm` rules for each timer can be used.

ascillato, 2020-01-21:
Hi,

`ON Timer1#Data.Arm` is not supported and it has very low priority to be added soon because it is just for one use case. Sorry.

meingraham, 2020-01-21:
Agreed. Lower priority.

I was only wondering if that syntax worked already.

meingraham, 2020-01-23:
@ppetppet 

Try `ON Timer1#Arm DO Var1 %value% ENDON`. This should store the state of `Timer1` in `Var1`. You will only get this message as a response to the Timer command. In other words, you invoke `Timer1` and the rule should be triggered. You can also use a condition to trigger only when the state changes to a particular value. For example: `ON Timer1#Arm=0 DO <perform a specific action> ENDON`

ppetppet, 2020-01-23:
OK, I will try it at the weekend and report back. Thank you.

ppetppet, 2020-01-24:
If you need more information, just ask.
Thank you.