arendst/Tasmota #7571: Jura Café machines as a potential new devices

BlazejosP, 2020-01-21:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
I'm an owner of such Café Machine Jura Impressa Z5 
[[image]](url)
[https://www.youtube.com/watch?v=ZRYpX1xgYQU](url)

But to the point such whole family of devices from that manufacturer are quite interesting from hacker point of view. They have a serial port originally intended by manufacturer for coin operations and also to work with their super, smart etc... (expensive) Bluetooth adapter with special manufacturer app for android [https://www.harveynorman.com.au/jura-smart-connect-bluetooth-transmitter.html](url) [https://www.youtube.com/watch?v=T6rdAZrGW8Q](url)

But thanks to the work done by some peoples around there is a open source solution to use that connector with adapter in your own software and have all functionality and data access.
Serial ports  standards in Jura café machines there are 3 variations of physical connector in various models done in last 15 years.
- 4 pins
- 7 pins
- DB9 RS-232 but 5V instated of 12V
So all are identical from electrical parameters TTL 
[http://protocoljura.wiki-site.com/index.php/Serial_interfaces](url)

**Describe the solution you'd like**  
When we have our arudino or pc connected we can operated on serial port with pretty easly there is also hacked set of commands for serial port which can operate whole functionality of such device 
[http://protocoljura.wiki-site.com/index.php/Commands_for_coffeemaker](url) Most of them is the same on various models but some of them have various functionality between different devices so after connection is established is necessary to test  this commands what they really do on your model I found already variations between my Z5 and default there IMPRESSA X7 . 
There is also other problem already solved by hackers. Jura is not working with plain text serial commands as Tasmota can send to Serial port but special shifted and prepared send in some sequence binary translations.
http://protocoljura.wiki-site.com/index.php/Protocol_to_coffeemaker

Translation function from best software which translate plain text to Jura understandable commands
and also back Jura device responses to plain text.
[https://community.home-assistant.io/t/control-your-jura-coffee-machine/26604](url)
[https://gitlab.com/Blueforcer/HA2JURA/snippets/1674496](url)
[https://github.com/hn/jura-coffee-machine/blob/master/cmd2jura.ino](url)

`String cmd2jura(String outbytes)_A clear and concise description of what the problem is._ {
  String inbytes;
  int w = 0;

  while (softserial.available()) {
    softserial.read();
  }

  outbytes += "\r\n";
  for (int i = 0; i < outbytes.length(); i++) {
    for (int s = 0; s < 8; s += 2) {
      char rawbyte = 255;
      bitWrite(rawbyte, 2, bitRead(outbytes.charAt(i), s + 0));
      bitWrite(rawbyte, 5, bitRead(outbytes.charAt(i), s + 1));
      softserial.write(rawbyte);
    }
    delay(8);
  }

  int s = 0;
  char inbyte;
  while (!inbytes.endsWith("\r\n")) {
    if (softserial.available()) {
      byte rawbyte = softserial.read();
      bitWrite(inbyte, s + 0, bitRead(rawbyte, 2));
      bitWrite(inbyte, s + 1, bitRead(rawbyte, 5));
      if ((s += 2) >= 8) {
        s = 0;
        inbytes += inbyte;
      }
    } else {
      delay(10);
    }
    if (w++ > 500) {
      return "";
    }
  }

  return inbytes.substring(0, inbytes.length() - 2);
}
`
There is also other solution to the same problem  [https://github.com/sklas/coffi](url)
Function in arduino file written from scratch but doing the same translation.
Also third solution for translation function [https://github.com/cubetech/arduino.coffeehack](url)

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  

Finally tough is that is possible to add to tasmota repository new module probably Jura TX RX which will switch serial port to this unusually command translation and operated on Jura coffee-maker of course individual commands functions will be necessary to discover for each device model. But there is function to check model of device in eeprom so I can even add a template of commands already discovered for that specific Z5 model. 

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-01-21:
Checking your link https://community.home-assistant.io/t/control-your-jura-coffee-machine/26604 you have there a complete dedicated solution to make your coffe maker a wifi appliance.

At this moment you have 2 solutions if you want to do it with Tasmota:

* Tasmota supports serial port and with _serial bridge to mqtt feature_ you can use it in your home automation software.
* If you want Tasmota to control it, you can use _serial port_ and you can set _rules_ to add the features you want.

If you want to develop an specific driver for this coffe maker, a Pull Request is very welcome. As this is very specific, may be should be better to be added to experimental Tasmota drivers at [Tasmota Playground](https://github.com/arendst/Sonoff-Tasmota-Playground)

BlazejosP, 2020-01-21:
Yes that true a working solution already exist and working great in my case [Link to HA2jura.ino] (https://gitlab.com/Blueforcer/HA2JURA/snippets/1674496) simple text field with send command button. I just tough that will be possible integrating this solution inside tasmota. I will be able to just plug ESP in that case because of powering from jura serial port 5V -> best device is Wemos D1 mini.  In Tasmota serial port either software or hardware on that moment can't working with Jura because this specific bytes shifting and timing of commands when sends is impossible to be achieved without adding driver for Jura device which will change behaviour of serial ports (probably one of them) when while be on. After that if tasmota will be able to speak through serial with Jura cafe-maker is possible to use any home automation software for example: domoticz through mqtt.

Is interesting question if rules in Tasmota are able so much changing behaviour of this software that writing special plug-in will be unnecessary? I personally tough that they aren't so powerfully meaby is necessary to check that?

ascillato2, 2020-01-24:
Hi,

As this is very specific usecase, and as you are the one that owns the machine, you will need to make the driver. There are plenty of driver examples in the codebase.

If you need assistance with the coding, you can start with a driver, make some modifications to it and make a pull request to Tasmota. We can help in the PR.

closed 2020-01-24 by ascillato2

ryanalden, 2020-06-22:
If you're willing to forego Tasmota, I put together an ESPhome custom component to accomplish this: https://github.com/ryanalden/esphome-jura-component

MCdobber, 2020-11-08:
Hi Schluden,

have you already managed to get the Jura E60 to work? I have one too, just can't get the right commands out yet.

I have the connection to the Jura thanks to the great esphome-jura-component.

schluden, 2020-11-14:
Hey @MCdobber, 
sorry to say: no unfortunately I didn't even try it out yet.

Currently, there are some other things that keep me busy.
But thank you for the reminder. I will keep this on track.

bernd37, 2021-01-28:
Hi @MCdobber & @schluden 

Any updates on this awesome project? I've got the Jura E60 also but could use some help in this case.

Dutch greetings!

MCdobber, 2021-02-06:
Hi @bernd37

Where you need help with?

BlazejosP, 2021-02-10:
Hello as I said in first post from Tasmota point of view is necessary to add only special modified com port driver with specific Jura bit shifting as in first post this processing code. Individually command varies between each Café machine and are not in scope of cover them all by Tasmota. This simple will be cover by user and his automation software. And Tasmota will be only a getaway which will receive command shift them and send to Jura machine. I need someone  to contact who can help add this (slightly modified) driver  to Tasmota repository. Any proposition who will be interested?

BlazejosP, 2021-02-10:
> Hi @bernd37
> 
> Where you need help with?

I need take ordinary standard Serial port driver in Tasmota and this function mentioned in first post `String cmd2jura(String outbytes)' paste into it this function will make bit shifting when we send something and when we receive something from Serial port. Than saved this As a new version of Serial Port for example (Serial Port Jura Cafe) and that it. If you can integrate this. I try myself but don't know how include this new driver in my own compilation. I have this  cafe machine so I can test this if you will be able include this function inside new Serial Port driver and of course send this to repository.

kugelkopf123, 2021-02-10:
Perhaps the scripting thing is something that could help? Perhaps it is possible to adjust this to your needs? [tasmota docs / scripting ](https://tasmota.github.io/docs/Smart-Meter-Interface/)

BlazejosP, 2021-02-11:
May be usefully but is necessary to write this bit shift function based on processing language translate this tasmota script but there is some problem. This is intended to be used with smart meters which shows automatically data on serial port in time period. Jura don't shows anything if is not asked with commands.

bernd37, 2021-02-15:
> Hi @bernd37
> 
> Where you need help with?

Same thing as your message 8 November 2020. I can't get the right commands out yet.
If got the rest up and running (ESPHome, hardware, wires, etc) but the only thing; commands.
When I hit the button "Make a coffee" my Jura E60 jumps into the menu page II and when I hit the button "Make hot water" my Jura E60 jumps into the menu page I. Do you know what to do @MCdobber ?

Kind regards,
Bernd

ryanalden, 2021-02-16:
> I can't get the right commands out yet.
> If got the rest up and running (ESPHome, hardware, wires, etc) but the only thing; commands.
> When I hit the button "Make a coffee" my Jura E60 jumps into the [...]
 
Hi @bernd37

That's great you've got the hardware set up, that's the hard part :)

Have you experimented with sending different commands to your machine?  It sounds like you've been working with my original code which is for an Impressa J6.  You'll have to experiment to find the correct commands for your E60.  It's a trial and error process, modifying the ESPhome yaml each time and testing to see the result.

For example, [here in a fork for an F70](https://github.com/fabicodes/esphome-jura-component/blob/master/jura_coffee.h), you can see the author has slightly different commands. His machine makes coffee with FA:03, whereas mine needs FA:09 to do the same thing. 

Here are some additional links with various commands for your experimentation:

[Jura XS90](https://github.com/PromyLOPh/juramote)
[Impressa X7](http://protocoljura.wiki-site.com/index.php/Commands_for_coffeemaker)
[S95](https://github.com/psct/sharespresso)

If you figure out a command to press the rinse button, let me know, I'd like to try it out on my J6 :)

bernd37, 2021-02-16:
Hi @ryanalden 

Thanks for your quick reply! Also thanks for the explanation and [helpfull](https://github.com/COM8/esp32-jura) links.
I'm now able to call the service "Coffee" and "Espresso" from my Home Assistant setup; as you can imagine this made me laugh :-)

For the Jura E60 the following calls and code are working fine:

**FA:04** _(Espresso)_
      - uart.write: [0xFB, 0xDF, 0xDB, 0xDF]  ## 'F'
      - delay: 8ms
      - uart.write: [0xDF, 0xDB, 0xDB, 0xDF]  ## 'A'
      - delay: 8ms
      - uart.write: [0xFB, 0xFB, 0xFF, 0xDB]  ## ':'
      - delay: 8ms
      - uart.write: [0xDB, 0xDB, 0xFF, 0xDB]  ## '0'
      - delay: 8ms
      - uart.write: [0xDB, 0xDF, 0xFF, 0xDB]  ## '4'
      - delay: 8ms
      - uart.write: [0xDF, 0xFF, 0xDB, 0xDB]  ## '\r'
      - delay: 8ms
      - uart.write: [0xFB, 0xFB, 0xDB, 0xDB]  ## '\n'

**FA:07** _(Coffee)_
      - uart.write: [0xFB, 0xDF, 0xDB, 0xDF]  ## 'F'
      - delay: 8ms
      - uart.write: [0xDF, 0xDB, 0xDB, 0xDF]  ## 'A'
      - delay: 8ms
      - uart.write: [0xFB, 0xFB, 0xFF, 0xDB]  ## ':'
      - delay: 8ms
      - uart.write: [0xDB, 0xDB, 0xFF, 0xDB]  ## '0'
      - delay: 8ms
      - uart.write: [0xFF, 0xDF, 0xFF, 0xDB]  ## '7'
      - delay: 8ms
      - uart.write: [0xDF, 0xFF, 0xDB, 0xDB]  ## '\r'
      - delay: 8ms
      - uart.write: [0xFB, 0xFB, 0xDB, 0xDB]  ## '\n'

Watertank and tray don't seem to work. This is something I'm still working on, I also do need to find the rinse button for you.

To be continued. Many thanks! Dutch greetings.

influenist, 2022-04-19:
Hi guys, @bernd37, 

Would be interested in this matter too. With our new home we are going to switch everything smart with HASS and seen I own a JURA ENA 9 One Touch, I was thinking to turn it smart. Starting with ESP8266 wemos d1 mini. Whats the advantage of using Tasmota for this?

Greetings from a Dutch guy in Warsaw :-)

BlazejosP, 2022-04-24:
Hello @influenist 
I saw that and I must tell you that is well done and research is quite good. But in my opinion ESP32 is too powerfully tool for that. ESP8266 is all what we need. I myself use https://github.com/hn/jura-coffee-machine extended by myself and is all what I need to job be done with my Jura Z5 machine. What will be necessary to have is some password on wifi interface to protect against unauthorised coffee production.  That is why I look also on Tasmota because with tasmote we can have webacess protected by Tasmota itself. Secondly that will be some standardized interface and tasmota my store some txt config files with specific commands already discovered by community so If you have Jura ena 9 when you discover all the command you will be able share them with rest of comunity. That is why I d'like to implement Jura Com Port inside tasmota.