arendst/Tasmota #7572: RGB color calibration for Alexa colors

mano1979, 2020-01-21:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._I have standard 5050 RGB ledstrip connected to a device of my own making. 
Everything works fine except for the colors. Green and blue are more prominent than red, and therefor i need to adjust the output of those channels.
I use Alexa to ocntrol my lights and the colors i can setup are limited. I prefer orange as color, but the orange is more a lime green in my case. Pink is a blue-ish white and purple is more blue.

I have tried adjusting the gamma settings which made it worse. I tried to adjust the RGBWWTable, which did nothing (that is probably not what it's used for), and tried pretty much every command even remotely connected to color, just to see if it did something. But nothing...

All i want is to lower the blue and green values so the red is more prominent than it is now.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): _____generic esp8266 on custom device
- [x ] Tasmota binary firmware version number used: _____8.0.1.4
  - [ x] Pre-compiled 
  - [ x] Self-compiled 
    - [x ] IDE / Compiler used: _____arduino 1.8.9 and 1.8.5
- [x ] Flashing tools used: _____tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

19:59:51 RSL: stat/tasmota/RESULT = {"NAME":"WiFiDomo","GPIO":[0,0,0,0,37,38,0,0,39,0,0,0,0],"FLAG":0,"BASE":34}
19:59:51 RSL: stat/tasmota/RESULT = {"Module":{"0":"WiFiDomo"}}
19:59:51 RSL: stat/tasmota/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"37":"PWM1"},"GPIO5":{"38":"PWM2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"39":"PWM3"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
20:01:30 RSL: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["WiFiDomo"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:01:30 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T19:29:14","StartupUTC":"2020-01-20T23:32:16","Sleep":50,"CfgHolder":4617,"BootCount":6,"SaveCount":86,"SaveAddress":"F6000"}}
20:01:30 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.1.0.4(tasmota)","BuildDateTime":"2020-01-20T20:14:37","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(bb83b9b)","Hardware":"ESP8266EX","CR":"346/699"}}
20:01:30 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SitecomD5DDF8 ",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","2805C8000180060000005A00000000000000","00000000","00000000"]}}
20:01:30 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":586,"Free":416,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashMode":2,"Features":["00000809","8FDAE397","003683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
20:01:30 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-4587","IPAddress":"192.168.192.243","Gateway":"192.168.192.1","Subnetmask":"255.255.255.0","DNSServer":"62.179.104.196","Mac":"5C:CF:7F:03:D1:EB","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:01:30 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_03D1EB","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
20:01:30 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Tue Jan 21 19:01:30 2020","Local":"Tue Jan 21 20:01:30 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:33","Sunset":"17:29"}}
20:01:30 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-01-21T20:01:30"}}
20:01:30 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-01-21T20:01:30","Uptime":"0T19:29:14","UptimeSec":70154,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":100,"Color":"FFA600","HSBColor":"39,100,100","Channel":[100,65,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"SitecomD5DDF8 ","BSSId":"B0:BE:76:71:C7:84","Channel":3,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I just would like to know how to adjust or calibrate the pwm output or internal color list to reflect the colors selected by alexa.

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-01-21:
This is precisely what RGBWWTABLE is for. What do you mean : it didn't work. 

Please try again and set `weblog 4` and report the logs here with different settings of rgbwwtable.

mano1979, 2020-01-21:
Here with adjusted values: 255,190,190,0,0
`22:12:55 CMD: RGBWWTABLE 255,190,190,0,0
22:12:55 RSL: stat/tasmota/RESULT = {"RGBWWTable1":"255,190,190,0,0"}
22:13:13 CMD: weblog 4
22:13:13 RSL: stat/tasmota/RESULT = {"WebLog":4}
22:13:14 CFG: Saved to flash at F4, Count 96, Bytes 4096
22:13:24 WIF: Checking connection...
22:13:24 WIF: Connected
22:13:44 WIF: Checking connection...
22:13:44 WIF: Connected
22:14:04 WIF: Checking connection...
22:14:04 WIF: Connected
22:14:06 HTP: Hue API (/3d1eb/lights/4005553/state)
22:14:06 HTP: Hue POST args ({"hue":7100,"sat":254})
22:14:06 RSL: stat/tasmota/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FFA600","HSBColor":"39,100,100","Channel":[100,65,0]}
22:14:06 HTP: Hue Result ([{"success":{"/lights/1/state/hue":7100}},{"success":{"/lights/1/state/sat":254}}])
22:14:07 HTP: Hue API (/3d1eb/lights/4005553)
22:14:07 /lights path=/lights/4005553
22:14:07 HTP: Hue User-Agent: Dalvik/2.1.0 (Linux; U; Android 5.1.1; AEOBC Build/LVY48F), gen=1
22:14:07 HTP: Hue Result ({"state":{"on":true,"bri":254,"colormode":"hs","xy":[0.55969,0.42362],"hue":7100,"sat":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"WiFiDomo","modelid":"LCT007","uniqueid":"5c:cf:7f:03:d1:eb:00:11-1","swversion":"5.50.1.19085"})
22:14:24 WIF: Checking connection...
22:14:24 WIF: Connected
`

And here with default values: 255 255,255,0,0
`22:17:58 CMD: RGBWWTABLE 255,255,255,0,0
22:17:58 SRC: WebConsole from 192.168.192.143
22:17:58 CMD: Group 0, Index 1, Command "RGBWWTABLE", Data "255,255,255,0,0"
22:17:58 RSL: stat/tasmota/RESULT = {"RGBWWTable1":"255,255,255,0,0"}
22:17:58 CFG: Saved to flash at FB, Count 97, Bytes 4096
22:18:04 WIF: Checking connection...
22:18:04 WIF: Connected
22:18:19 HTP: Hue API (/3d1eb/lights/4005553/state)
22:18:19 HTP: Hue POST args ({"hue":7100,"sat":254})
22:18:19 RSL: stat/tasmota/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FFA600","HSBColor":"39,100,100","Channel":[100,65,0]}
22:18:19 HTP: Hue Result ([{"success":{"/lights/1/state/hue":7100}},{"success":{"/lights/1/state/sat":254}}])
22:18:19 HTP: Hue API (/3d1eb/lights/4005553)
22:18:19 /lights path=/lights/4005553
22:18:19 HTP: Hue User-Agent: Dalvik/2.1.0 (Linux; U; Android 5.1.1; AEOBC Build/LVY48F), gen=1
22:18:19 HTP: Hue Result ({"state":{"on":true,"bri":254,"colormode":"hs","xy":[0.55969,0.42362],"hue":7100,"sat":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"WiFiDomo","modelid":"LCT007","uniqueid":"5c:cf:7f:03:d1:eb:00:11-1","swversion":"5.50.1.19085"})
22:18:21 HTP: Hue API (/qbWuKzRbedt0XfbJtK6r1oB3C5me03L1zw0dHj8D/lights/1)
22:18:21 /lights path=/lights/1
22:18:21 HTP: Hue User-Agent: Dalvik/2.1.0 (Linux; U; Android 5.1.1; AEOBC Build/LVY48F), gen=1
22:18:21 HTP: Hue Result ({"state":{"on":true,"bri":254,"colormode":"hs","xy":[0.55969,0.42362],"hue":7100,"sat":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"WiFiDomo","modelid":"LCT007","uniqueid":"5c:cf:7f:03:d1:eb:00:11-1","swversion":"5.50.1.19085"})
22:18:24 WIF: Checking connection...
22:18:24 WIF: Connected
`

I hope this is what you meant, it's my first time using tasmota.

mano1979, 2020-01-21:
There is no physical change in color, whatever i set the table to.

s-hadinger, 2020-01-22:
A difference of brightness from 255 to 190 is barely visible, especially when Gamma correction is off. Can you try with a more aggressive `RGBWWTable 255,50,50,255,255`.

Also don't set the last two values to zero, it enables white-blend mode.

mano1979, 2020-01-22:
I tried with 255,50,50,255,255 (and gamma correction is on) but that makes no difference either.
i did use weblog 4 but i am not on the development branche i am using the latest release.

I even have tried to set the blue and green to 0, but nothing changes in the color. Not in the console, and not physically.

for now i will update to the development branche, but let me know if there is anything else i can do...

s-hadinger, 2020-01-22:
I looked at the code, there is indeed a bug and RGBWWtable is ignored. I'm fixing it now.

PR #7579 (2020-01-22): Fix ``RGBWWTable`` ignored (#7572)
## Description:

**Related issue (if applicable):** fixes #7572

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.6.1
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).
closed 2020-01-22 by arendst