arendst/Tasmota #7577: "Calibrate linear/polynomial" function as ESPhome

tiagofreire-pt, 2020-01-22:
**Have you looked for this feature in other issues and in the docs?**  
Yes.

**Is your feature request related to a problem? Please describe.**  
Yes, I have to add an huge set of rules, with more than 40 "else if" expressions or compile myself the firmware and set one rule with "expressions enables". This is a mess, as every time Tasmota upgrades the code I have to re-compile it several times, each one for every single tasmotized plug. 

Every Blitzwolf plug has a very linear current value ranging (currentcal) - e.g. SHP6 15A model:
100W = 3016
3650W = 4096

**Describe the solution you'd like**  

As ESPHome, a calibration function option: https://gist.github.com/frenck/341323998fbc2557d44608b5bd634b88#file-calibrate_current-yaml

https://esphome.io/components/sensor/index.html?highlight=linear#calibrate-linear
https://esphome.io/components/sensor/index.html?highlight=linear#calibrate-polynomial

**Describe alternatives you've considered**  
Huge and inefficient set of rules and re-compiling...

**Additional context**  
N/A

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2020-01-22:
Once you have the plug's calibration settings, you should not have to adjust them even if you upgrade the firmware.

Finally, if the plug functions as desired, it is best to not mess with it - i.e., leave it alone and do not upgrade it unless there is a problem or a specific feature you require.

> every time Tasmota upgrades the code I have to re-compile it several times, each one for every single tasmotized plug

Isn't this always the case for ESPHome?

tiagofreire-pt, 2020-01-23:
@meingraham , I see your point. 

I'm just proposing a new and more precise way to calibrate devices with linear or polynomial behaviour, rather than a static one. Not even close a comparative way between Tasmota and ESPHome general approaches.

A possible solution for easy implementation (not by recompiling code - Tasmota approach), without an huge program memory compromise, would be a command on console accepting a JSON like:

Linear (e.g., current, voltage, power) function (order = 1, y = a*x + b):
`{"Order":"1","Parameters":[**value_for_a**, **value_for_b**]}`

Polynomial function (e.g., order = 2, y = a*x^2 + b*x + c):
`{"Order":"2","Parameters":[**value_for_a**, **value_for_b**, **value_for_c**]}`

Logaritmic (e.g., natural logaritm, y = a*ln(b*x) + c):
`{"Order":"ln","Parameters":[**value_for_a**, **value_for_b**, **value_for_c**]}`

Logaritmic (e.g., logaritm base 10, y = a*log(b*x) + c):
`{"Order":"log","Parameters":[**value_for_a**, **value_for_b**, **value_for_c**]}`

And so on.

tiagofreire-pt, 2020-01-23:
This effort for precise measurements is vital for automations on devices with non-linear power rating and usage, useful for determining states related to electric dimensions,  for improving the data collected over time for better energy consumption data, voltage following for power grid quality of service determination, safety regarding power or current ratings and even voltage (yes, many sensitive devices may need power off on extreme voltage levels), etc.

Jason2866, 2020-01-23:
A PR for this idea is highly welcome.

meingraham, 2020-01-23:
> Not even close a comparative way between Tasmota and ESPHome general approaches.

I was referring to having to always compile to "re-configure", not to how each firmware approaches power calibration.

But, yes, If you want to take a stab at coding this (i.e., a PR), I think it would be a great enhancement to Tasmota.

Cheers!

Mike

closed 2020-02-22 by stale[bot]