arendst/Tasmota #7578: Change Moisture to AdcRange

robin13, 2020-01-22:
**Have you looked for this feature in other issues and in the docs?**  
yes

**Is your feature request related to a problem? Please describe.**  
Currently the ADC range feature is named as a "Moisture" sensor.  In the mean time I have used this type for
- Moisture sensor
- Air quality sensor
- Voltage sensor
- Audio volume sensor

I see that it is a very useful feature, BUT it is a bit bothersome that each of these values are displayed (and passed on to Home Assistant) as a moisture sensor.  

**Describe the solution you'd like**  
I think it would be more useful for all if this feature was something generic such as "ADCRange" - this would allow more people to correctly find the right type and then adjust the display appropriately in Hassio etc. rather than have to document somewhere "Use the 'Moisture sensor' - it's a nice hack to be able to display measured XYZ"

**Describe alternatives you've considered**  
Currently everything is "Moisture" ... it works, but it's a bit annoying...

**Additional context**  
 
See original issue which resulted in the Moisture sensor: https://github.com/arendst/Tasmota/issues/7182

CC @effelle @ascillato as you were involved in the previous case too.

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-01-23:
Sorry to insist, but we have `SCALEx` for that. 
I understand perfectly your point of view and have a more generic text instead of `Moisture` on webui (and HA discovery in your case) makes sense but I really prefer to avoid duplicated code.

meingraham, 2020-01-23:
As answered.

effelle, 2020-01-23:
@ascillato I had a talk with Theo today about it and I will change the component from `ADC Moisture` asap with the suggested name `ADC Range`. 
We change a bit the formula without adding decimals @robin13 because we mutually agreed it is not necessary.

PR #7589 (2020-01-24): ADC Range (replaces adc Moisture)
## Description:
New definitions and translations for both console and webui added to update `ADC Moisture` to a more generic `ADC Range`. 
As discussed with the team leader the new formula will returns just int values.
Usage for calibration:
```
AdcParam 6, fromLow, fromHigh, toLow, toHigh
```
reset to default configuration [6,0,1023,0,100]:
```
AdcParam 6
```

**Related issue (if applicable):** fixes #7578 and solve definitely #7402 .

## Checklist:
  - [X] The pull request is done against the latest dev branch
  - [X] Only relevant files were touched
  - [X] [... cut]
closed 2020-01-24 by arendst