arendst/Tasmota #7582: HA discovery template creates invalid entries for dual energy monitoring on shelly 2.5

glance-, 2020-01-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
HA discovery templates returns list of values for current, power, apparent power and power factor which breaks graphing and numeric triggering.
For example the template `value_json['ENERGY'].ApparentPower` now returns [278, 0] instead of just 278 as it would before upgrade to tasmota capable of dual energy monitoring.

### Proposed fix
I'd suggest that we generate discovery for two different sensors, one for each value.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Shelly 2.5
- [x] Tasmota binary firmware version number used: 8.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: upgraded from tasmota 5-6 something, back then flashed with esptool
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
23:28:35 MQT: shelly1/stat/RESULT = {"NAME":"Shelly 2.5 swi","GPIO":[56,255,17,255,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}
23:28:36 MQT: shelly1/stat/RESULT = {"Module":{"0":"Shelly 2.5 swi"}}
23:28:36 MQT: shelly1/stat/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"0":"None"},"GPIO2":{"17":"Button1"},"GPIO3":{"0":"None"},"GPIO4":{"21":"Relay1"},"GPIO5":{"83":"Switch2n"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"6":"I2C SDA"},"GPIO13":{"82":"Switch1n"},"GPIO14":{"5":"I2C SCL"},"GPIO15":{"22":"Relay2"},"GPIO16":{"156":"ADE7953 IRQ"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
Not applicable
```
- [ ] Provide the output of this command: `Status 0`:
```
23:29:08 MQT: shelly1/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["shelly1","shelly1b"],"Topic":"shelly1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:29:08 MQT: shelly1/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:28:39","StartupUTC":"2020-01-22T22:00:29","Sleep":0,"CfgHolder":4617,"BootCount":39,"SaveCount":971,"SaveAddress":"FB000"}}
23:29:08 MQT: shelly1/stat/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"348/699"}}
23:29:08 MQT: shelly1/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"xxxxxxxxxxx","LogPort":514,"SSId":["yyyyyyy",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["002A880B","0105C8000100060000005AFF000000000000","00000600","00000000"]}}
23:29:08 MQT: shelly1/stat/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
23:29:08 MQT: shelly1/stat/STATUS5 = {"StatusNET":{"Hostname":"shelly1","IPAddress":"xxxxxxxxxxxxx","Gateway":"xxxxxxxxxxx","Subnetmask":"255.255.255.0","DNSServer":"xxxxxxxxxxx","Mac":"xxxxxxxxxxxxxxxxx","Webserver":2,"WifiConfig":4}}
23:29:08 MQT: shelly1/stat/STATUS6 = {"StatusMQT":{"MqttHost":"xxxxxxxxxxx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"xxxxxxxxxxx","MqttUser":"nnnnnnnnnnnnnn,"MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
23:29:08 MQT: shelly1/stat/STATUS7 = {"StatusTIM":{"UTC":"Wed Jan 22 22:29:08 2020","Local":"Wed Jan 22 23:29:08 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:32","Sunset":"17:30"}}
23:29:08 MQT: shelly1/stat/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
23:29:08 MQT: shelly1/stat/STATUS10 = {"StatusSNS":{"Time":"2020-01-22T23:29:08","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":62.5},"ENERGY":{"TotalStartTime":"2019-07-09T19:42:51","Total":896.804,"Yesterday":4.875,"Today":3.708,"Power":[278,0],"ApparentPower":[278,0],"ReactivePower":[4,0],"Factor":[1.00,0.00],"Frequency":50,"Voltage":237,"Current":[1.114,0.000]},"TempUnit":"C"}}
23:29:08 MQT: shelly1/stat/STATUS11 = {"StatusSTS":{"Time":"2020-01-22T23:29:08","Uptime":"0T00:28:39","UptimeSec":1719,"Heap":26,"SleepMode":"Normal","Sleep":0,"LoadAvg":999,"MqttCount":1,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"nnnnnnn","BSSId":"yyyyyyyyyyyyyyyyy","Channel":1,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Run tasmota 8.1.0 on a shelly 2.5 with SetOption19 1 against home-assistant.

### EXPECTED BEHAVIOUR
Sensors with scalar values

### SCREENSHOTS
`shelly1 ENERGY ApparentPower [277, 0]`

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-01-23:
That's a special case currently not handled, thanks for reporting.
For Shelly 2.5 was decided to have both the metering values stored in the same key separated by comma to keep the `json` shorter.
For now your only solution is to ignore/hide the sensors discovered by Tasmota and write some YAML code. Add the sensors with an index on `value_template` to have separated values.
Example:

```
{{ value_json.ENERGY.ApparentPower[0] }}
{{ value_json.ENERGY.ApparentPower[1] }}
```
`0` will return the first value on list, `1` the second.

meingraham, 2020-01-23:
As answered.

glance-, 2020-01-23:
Yes, I know that that's a workaround, man meanwhile I use a different workaround, but I still suggest that we should generate discovery templates which produce the correct information in the sensor.

effelle, 2020-01-23:
Noted.
Let's put in this way, for every special case to handle I need to use extra flash and memory.
Now, differently from special sensors not present on main firmware you choose to add in your own build, `Home Assistant Discovery` is enabled on the most important ones ( `tasmota.bin`, `knx.bin`, `sensors.bin` and `ir.bin`).
As you probably know the ESP offers a very limited space so every byte counts and right now I need to reduce the code size.
I'll have a look on next update to see if I can handle the `Shelly2.5` with minimal impact.

PR #7678 (2020-02-06): New HAss Discovery (1/4): Sensors
## Description:
**Added**:
-  New code lookup to identify and create entities with the correct unit of measure and class (or icon) for almost every sensor type;
- Support for multiple payload (like Shelly 2.5 or SDM630 energy meters);
- Optimization to save space on flash;
- New layout allows easy updates.

**Removed**:
- support for MPR121 as a `binary sensor` . Even if it will be created as normal sensors using the fallback sensor procedure it will not work as expected and is better add it manually using YAML code.

**Related issue (if applicable):** fixes # [... cut]
closed 2020-02-06 by arendst