arendst/Tasmota #7583: updating dimmer value in wakeup

balli1187, 2020-01-23:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
When using `wakeup` command the `dimmer` will not be updated (or send via mqtt). Therefore status of the device is shown wrong in this case.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Please update the `dimmer` value at least at the end (when completing a wakeup cycle) to actual dimmer state so external systems and device are aware of right device status.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
could be done within a rule (not sure) but sending the actual device status in all cases should be core feature without need to build a workaround.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-01-23:
Our ability to provide assistance is greatly hampered since you did not take a few minutes to complete the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) with the requested information. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.
 
Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).
 
Thanks.
 
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

balli1187, 2020-01-23:
Im sorry guys but i just used template that is given when opening a feature request 
(which is definately different from bug report template).
If this is not the right one pleas insert the right template and i will give information i can.

Jason2866, 2020-01-23:
Have you looked in the last enhancements in development path?
There are changes regarding wakeup. Maybe it is just what you want?
Since you did not wrote any info about your current version we dont know.
It is not a feature request it is a issue. So please use issue template

ascillato, 2020-01-23:
Hi,

I could reproduce it using latest Tasmota codebase (8.1.0.4). When using `wakeup` command, dimmer status is not republished, only POWER status.

Console output (with `weblog 4`):

```yaml
18:22:43 CMD: wakeup
18:22:43 SRC: WebConsole from 192.168.1.121
18:22:43 CMD: Group 0, Index 1, Command "WAKEUP", Data ""
18:22:43 SRC: Light
18:22:43 MQT: stat/cocina/RESULT = {"POWER3":"ON"}
18:22:43 MQT: stat/cocina/POWER3 = ON
18:22:43 MQT: stat/cocina/RESULT = {"Wakeup":"Started"}
18:22:44 CFG: Saved to flash at 3F8, Count 195, Bytes 4096
18:22:45 WIF: Checking connection...
18:22:45 WIF: Connected
18:22:46 MQT: tele/cocina/WAKEUP = {"Wakeup":"Done"}
18:22:47 CFG: Saved to flash at 3F7, Count 196, Bytes 4096
```

when using `dimmer` command, the output is:

```yaml
18:39:56 CMD: dimmer 20
18:39:56 SRC: WebConsole from 192.168.1.121
18:39:56 CMD: Grupo 0, Índice 1, Comando "DIMMER", Datos "20"
18:39:56 MQT: stat/cocina/RESULT = {"POWER3":"ON","Dimmer":20,"Color":"000233","HSBColor":"238,100,20","Channel":[0,1,20]}
```

So, it makes sense to have also this output when `wakeup` is finished.

ascillato2, 2020-01-24:
Closing issue as Theo has added the requested output. Thanks.

closed 2020-01-24 by ascillato2