arendst/Tasmota #7586: multiple dimmer/sliders on same tuyamcu

burnshomeautomation, 2020-01-23:
**Have you looked for this feature in other issues and in the docs?**  Yes, I have looked and as far as i can tell, with the latest development release 8.1.0.4, this is not possible.

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._  Yes and No, the problem is, I don't currently have a way to control both the brightness and white temperature from Tasmota due to the inability to have more than one dimmer/slider on the same tuyamcu.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._  Would it be possible to have a second fnid, in addition to fnid 21 that could be used for a second dimmer, or color temperature?

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
I have considered some sort of rule based solution that might work, just not sure how to go about setting it up.
**Additional context**  
_Add any other context or screenshots about the feature request here._ Here is a link to the lamp in question....
https://www.zemismart.com/wifi-timer-desk-lamp-dimmerable-alexa-google-home-tuya-smart-life-manual-touch-control-black-led-table-light-with-usb-port-p0206-p0206.html
**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-02-10:
Can be you request solved by https://github.com/arendst/Tasmota/issues/7709#issuecomment-584245560 ?

ascillato, 2020-02-14:
@burnshomeautomation

burnshomeautomation, 2020-02-17:
I am not sure, how do go about testing this exactly?

ascillato2, 2020-02-20:
seems that your request can e done using rules as in https://github.com/arendst/Tasmota/issues/7646

burnshomeautomation, 2020-02-20:
im still not sure how to implement that?  How do I assign 2 different dpids?

closed 2020-03-21 by stale[bot]