arendst/Tasmota #7592: SML Driver: Different baudrates for sending and receiving to support even more meters.

Kludi2019, 2020-01-24:
**Have you looked for this feature in other issues and in the docs?**  

yes

**Is your feature request related to a problem? Please describe.**  

I have an electricity meter named L + G ZMR120AReS2R2sfCS which, like some other electricity meters, needs a character string to send the current data. However, the initial request ("/ ?!" as Hex + CR LF) must be sent at 300 baud as described in the manual. After that you can read the meter's response at 9600 baud.

Since there is a lot of data, it takes a long time to receive it when I'am still on 300 Baud.

The SML driver doesn't support this option at the moment. So far, you have to decide for a baud rate that is used for sending and receiving.

There are other meters that also recommend different baud rates for sending and receiving. Another example is: Landis & Gyr UH550.

**Describe the solution you'd like**  

It would be great if you could set the receiving and sending speed separately. It may be easiest to implement it if one happens via hardware serial and the other via soft serial.

**Describe alternatives you've considered**  
-
**Additional context**  
-
**(Please, remember to close the issue when the problem has been addressed)**  

Thanks!

gemu2015, 2020-01-25:
@Kludi2019 

i added an sml function to scripter 

res=sml(meter 0 baud)
sets the baud rate of the serial port belonging to meter (1..n)

res=sml(meter 1 "hex string")
sends the hex string (as binary) to the serial port belonging to meter (1..n)

you may define SML with 300 baud and omit the global init string
then in section >B of the script
\>D
res=0
\>B
; restart sml
=>sensor53 r
; send init string once
res=sml(1 1 "2F3F210D0A")
; wait for the number of chars to send with 300 baud (30 chars/s)
; 5 chars 170 ms
delay(200)
; switch to 9600 baud
res=sml(1 0 9600)

before making a pr try my fork https://github.com/gemu2015/Sonoff-Tasmota

files changed scripter and sml

to enable use
\#define USE_SML_SCRIPT_CMD

Kludi2019, 2020-01-29:
Thank you for the quick program change.
Unfortunately, the SM is still transmitting at 300 baud. I have some questions about script execution. 
1.	What do I have to change under> M? The baud rate is still defined at 300 baud.

2.	There is a pull sequence that should be "2F3F210D0A", and as soon as the counter outputs its model number, another field sequence should be sent, which should then be "063030300D0A". But how do I do that with the two separate sequences with the new program change?

3.	In the old version, the request sequence was permanently sent every 2 seconds. This is also necessary because the SM no longer transmits after the end of the entire sequence and needs the request again

I have attached the operating instructions of the L&G and the current script as an attachment and hope that I have expressed myself clearly and thank you for the support.

[Script Stromzähler.txt](https://github.com/arendst/Tasmota/files/4127454/Script.Stromzahler.txt)
[E230-User-Manual.pdf](https://github.com/arendst/Tasmota/files/4127461/E230-User-Manual.pdf)

Best wishes
Gerhard

Kludi2019, 2020-02-03:
Thank you for the great help!

I have to test the adapter for a few days now, but the function is great so far

Best wishes Gerhard

PR #7663 (2020-02-04): Sml script dynamic baudrate and serial transmit support
## Description:

**Related issue (if applicable):** fixes #7592

script support for dynamic baud rates and serial transmit in sml meter driver
required for some special meters

enabled by 
#define USE_SML_SCRIPT_CMD

adds 2 script cmds

res=sml(meter 0 baud)
sets baud rate of meter (1..N) to serial baud rate
res=sml(meter 1 "hexstr")
send binary code of hexadecimal byte string to meter (1..N)

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix  [... cut]
closed 2020-02-03 by Kludi2019