arendst/Tasmota #7596: Add I2C Sensor (Asair) AHT10 / AHT15 (Temperatur/Humidity)

ludwich66, 2020-01-24:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Add a new I2C Temp/Humidity Sensor equal to BMP280
Library on https://github.com/Thinary/AHT10 

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
-
**Additional context**  
_Add any other context or screenshots about the feature request here._
find on Github link
**(Please, remember to close the issue when the problem has been addressed)**  
[image]

device111, 2020-02-18:
Hi,
you can try my fork but its not tested yet and i'm not verry similar with C-Programming. 
I'm waiting for Hardware to test it.

ludwich66, 2020-02-18:
Hi device111 I do not se a light way to compile your fork.
Is it possible to send me an precompiled binary? 
My Platform is an ESP 8266 12F

device111, 2020-02-18:
Hi ludwich,
here is it:

[tasmota.zip](https://github.com/arendst/Tasmota/files/4221329/tasmota.zip)

ludwich66, 2020-02-18:
Great - I will be back on Thursday and will solder and test it

device111, 2020-02-18:
Cool! 
thank you for the test!

ok, i have looked in the datasheet. The AHT series have no CHIP-ID. So there is no difference between the two Chips. I can call it AHT10/15.

I will change the repo and send you a new Version. 
So it would be great that you can test for a couple of hours, to see if its stable.

ludwich66, 2020-02-19:
Hi device111 - no problem - the last version works.
Temperature looks good - Humidity is a little bit too high, but think its not a Problem with your library. 
Did the sensor work on both addresses? Hex x038 / Hex x039
I have 2 Sensors -  tomorrow is soldering day ;-)

Please send me the new version.

device111, 2020-02-20:
Hi ludwich,

sounds good. No, at the moment is only working with one sensor Address. (0x38). If the libary works good, i will try to implemented both Adresses in the libary with autosearch.
I will check the humidity formula, if its wrong.

You can send me an Email, to perform more tests an we don't spam this thread: device@gmx.at
Here the new Version:
[tasmota.zip](https://github.com/arendst/Tasmota/files/4229267/tasmota.zip)

device111, 2020-02-20:
Here is the first try to autosearch and poll both sensors with 0x38 and 0x39.
Believe me, i don't no if that works... :-)

if no Display of sensors on the website, give me the console log from startup.

[tasmota.zip](https://github.com/arendst/Tasmota/files/4230318/tasmota.zip)

device111, 2020-02-20:
ok, formula for humidity is ok. I think there is the senor itself. It's a verry cheap sensor and has no calibration table to use.
one thing i found...at the moment i initialize the sensor with the "calibration" bit:

[image]

i dont' no what this happens but in the original repo from Thinary its the same and i found no Infos about the calibration Bit in the Datasheet.
ok, let us test if the other function works.. :-)

device111, 2020-02-25:
#7797

closed 2020-02-26 by ascillato2

catalintabirca, 2020-03-02:
testing a AHT10 sensor on a Sonoff Basic. it replaced a BME280. loadAvg has gone up from 20 to 27-30. just letting you know. olso, any chance for better resolution?

device111, 2020-03-02:
there is nothing optimized in this initial Source but you can not compare the BME280 with this cheap china sensor. This Sensor has answer time over 75ms and the documentaion is not realy complete. 
If i had time, will release a better Version in the next view days, with better timing and Dual address support.
At the moment ludwich66 is testing the new source.

crashdown79, 2020-03-15:
After trying the precompiled binary from @device111 i´ve tried now to compile it my self.
Unfortunately the result is that the sensor reading stays at Temp: 21.0°C and Humidty 29.0 for all time while it seems work normal with the precompiled bin. Could not find where could be the error. Do you have suggestions what/where i can check? 
On a WEMOS D1 mini with Arduino 1.8.12 and 8.1.0.11(tasmota)

Source changes:
Excluded:  //  #define USE_ADE7953  
Included:   #define USE_AHT1x  

{"StatusFWR":{"Version":"8.1.0.11(tasmota)","BuildDateTime":"2020-03-15T18:36:47","Boot":31,"Core":"2_5_2","SDK":"2.2.1(cfd48f3)","Hardware":"ESP8266EX","CR":"334/699"}}

{"StatusSNS":{"Time":"2020-03-15T18:44:25","AHT1x":{"Temperature":21.0,"Humidity":29.0},"TempUnit":"C"}}

device111, 2020-03-15:
Do you have other I2C Sensors wired or only AHT10?

crashdown79, 2020-03-15:
Only one AHT10 on D2 = SDA and D1 = SCL

device111, 2020-03-15:
ok, try to change the delay(80);  to  delay(100); in 
xsns_63_aht1x.ino (row 64)
and compile.

crashdown79, 2020-03-15:
seems to have other problems, neither with delay(80) nor delay(100) none of my 5 AHT10´s works now with the Wemos (not in parallel, one at a time, while a BME280 for comparison works directly)

device111, 2020-03-19:
Try the new code in the dev. Version.

and update your Core Version to a newer one,
2_5_2 ist old. We have 2.6.3 or 2.7 dev.:

{"StatusFWR":{"Version":"8.1.0.11(tasmota)","BuildDateTime":"2020-03-15T18:36:47","Boot":31,"**Core":"2_5_2"**,"SDK":"2.2.1(cfd48f3)","Hardware":"ESP8266EX","CR":"334/699"}}

crashdown79, 2020-03-21:
Now it works :-) 

12:02:27 MQT: stat/aht10_C88C20/STATUS = {"Status":{"Module":18,"FriendlyName":["aht10"],"Topic":"aht10_C88C20","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:02:27 MQT: stat/aht10_C88C20/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-21T12:00:23","Boot":31,"Core":"2_6_3","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"331/699"}}
12:02:27 MQT: stat/aht10_C88C20/STATUS10 = {"StatusSNS":{"Time":"2020-03-21T12:02:27","AHT1X-0x38":{"Temperature":25.8,"Humidity":37.0,"DewPoint":10.0},"TempUnit":"C"}}
12:02:27 MQT: stat/aht10_C88C20/STATUS11 = {"StatusSTS":{"Time":"2020-03-21T12:02:27","Uptime":"0T00:00:49","UptimeSec":49,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":32,"MqttCount":1,"Wifi":{"AP":1,"SSId":"cdciot","BSSId":"9C:C7:A6:42:C6:5A","Channel":11,"RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:08"}}}

I´ve updated the Core to 2.6.3 and used the new 8.2.0 tasmota. It works prety smoth with all of my sensors.
Thank you verry much guys. You really do a great job with tasmota. I love it.

alexandru2052, 2021-05-24:
you can share the tasmota compilation with other users???thanks!

crashdown79, 2021-06-19:
Hi, sorry wasn’t able to answer for a time.

I´ve created a new .bin today with AHT10 and MAX44009 which I use.
Would it be best to upload on my own webpage or can I link it direct in the forum?
Or where would be the usual/best place to share?

Regards Stefan

Tiny2910, 2022-01-08:
@crashdown79 Tried almost everything using your binary, but still unable to make AHT10 work. Can you help me locate the problem. I'm using wemos D1 mini.

crashdown79, 2022-01-13:
@Tiny2910 what does your setup look like? I´d described it in more details some time ago: https://crashdown.de/2020/05/01/wemos-d1-mini-on-tasmota-with-aht10-sensor/

crashdown79, 2024-07-08:
for the interested, i´ve updated to new tasmota 14.1.0.3 with active AHT10 
https://github.com/crashdown79/tasmota-aht10/releases/tag/tasmota