arendst/Tasmota #7599: MQTT Discovery: Publish Switches (just like buttons)

jensuffhaus, 2020-01-25:
**Have you looked for this feature in other issues and in the docs?**  Yes. I have tried countless configurations and cannot get the functionality as I would like it.

**Is your feature request related to a problem? Please describe.**  
I am trying to use a physical latching toggle switch on Tasmota as a input for HA to trigger events without triggering the local relay. Using Switch2 prevents the relay from triggering but the switch is not published to HA via discovery and HA does not know about it. Using Button2 solves the discovery but the button config does not seem to work well with latching switches.

**Describe the solution you'd like**  
Include switches (without relays) in the MQTT auto discovery as "read only" on/off sensors that cannot be controlled via MQTT. Toggeling the switch pushes the same events to HA as a switch with a relay. 

**Describe alternatives you've considered**  
I've tried using Button2 but buttons seem to be hard wired for momentary contact and not latching switches.
I've setup manual binary sensors in HA but this is cumbersome and prone to problems when everything else is using auto discovery.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

jensuffhaus, 2020-01-25:
Update:

After logging the above and the resetting one device I forgot to set "SwitchTopic 1" and noticed that the switch for that device is then published via auto discovery.

Unfortunately using "SwitchMode 0" disables MQTT messages and thus the hold or multitoggle feature.

Can the switch not continue to be published via auto discovery even when "switchtopic 1" is used. The the actual switch that is published could even receive the hold state, or even have the "Hold" state published as a separate binary sensor called Switch1Hold.

jensuffhaus, 2020-01-25:
And more information after more testing ...

It seems to get any switch configured in the gui I need to use "switchtopic 0", then restart and the switch will be discovered.

I can then set "switchtopic 1" and the switch state seems to continue being updated in HA, somehow, I don't quite know what happens in the background.

strangely once a switch was published in "switchtopic 0" then switchtopic was set to 1, if switchMODE 8 is used and the multi toggle is performed the console shows a HOLD message being sent but in HA the discovered switch toggles and stays, as if the switch was toggled normally.

The HOLD message should not affect the switch in HA at all and best would be if it was exposed as a button via auto discovery.

jensuffhaus, 2020-01-25:
Summary:

- If "SwitchTopic 0" is used we seem to get auto discovery of switches and they are tracked in HA. Unfortunately multipress/multitoggle modes are not supported in this setting.
- If "SwitchTopic 1" is used switches do not seem to be auto discovered and one must now rely purely on manual MQTT messaging. In HA this gets messy and devices are no longer a thing then.
- The HOLD feature is only supported under "SwitchTopic 1" at the cost of switch discovery.

Suggestion:

- Enable the HOLD, LongPress, MultiToggle feature even under "SwitchTopic 0" and instead of a "HOLD" message this will publish an additional virtual BUTTON called "Switch<x>Hold".
- Then if you have a switch with "switchTopic 0" and "SwitchMode 5/8" ... in HA you would end up with two discovered sensors, "switch1" (on/off) and "switch1hold" (on/instant-auto-off)

septillion-git, 2020-02-10:
The main problem is the weird implementation of the HOLD (switchmode 5) in Tasmota. It also sends a TOGGLE on release. Would be great if it would send a RELEASE on release after it send a HOLD.

This could be added as a separate switch or you just use it as a sensor. Auto discovery would be nice but might become a nightmare with the whole mixing of light and switch entities.

Btw, you can make the relay ignore the switch without setting a button topic. Just add the rule `ON Switch1#State DO  ENDON`

closed 2020-03-11 by stale[bot]