arendst/Tasmota #7600: Support for ESP32wroom32U

Palerm0, 2020-01-25:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I have an new ESP32wroom32u Flashed the new 8.1.0 (Doris) onto it. But I`dont see the Wifi
SSID. Flashing is done succesfully.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
Not supported, I also ask this in the discourt chat. there is was confirmed that is was not supported.
I believe its a brand new chip
 
**Additional context**  
_Add any other context or screenshots about the feature request here._
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None

**(Please, remember to close the issue when the problem has been addressed)**  

What do you need to possible support this device?

s-hadinger, 2020-01-25:
Esp32 is not supported and is unlikely to be in the near future (unless you want to port it). 

Esp32 architecture is very different from ESP8266.

Palerm0, 2020-01-25:
@s-hadinger 
I did indeed got some more reactions on the chat about that. 
Unfortunate. its a nice fast device.
I also read somewhere that the esp32 is the successor of the esp8266. So tasmota is supporting a older chip. Wonder what equal newer chips are supported

closed 2020-01-25 by Palerm0

meingraham, 2020-01-25:
So others that land here can follow the ultimate debate and counterpoints:

https://discordapp.com/channels/479389167382691863/479389167382691865/670641693493821442

ascillato, 2020-01-25:
Hi,

This is a duplicated issue. If you make a fast search in issues you will find that this was discussed several times before. As a summary, no one is against esp32. It is just a huge effort to adapt Tasmota to esp32. If you want to code it, please do it, we would love that. So far there isn't many commercial products with esp32 in order to make the esp32 support a big need.

ascillato2, 2020-01-25:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

Palerm0, 2020-01-25:
@ascillato and @meingraham
Sorry for duplicate post.
Thanks for your answers in discord. Did a search on google, not on this github page.

ascillato, 2020-01-25:
No problem at all. If you want to start working on a port to ESP32 for Tasmota, please, do it.

Palerm0, 2020-01-25:
If I had the knowledge, I`m a great fan of tasmota. Lots of my devices are running tasmota.