arendst/Tasmota #7606: Auto discovered switch ignores MQTT payload.

jensuffhaus, 2020-01-26:
### PROBLEM DESCRIPTION
A swich configured in Tasmota and pushed to HA via Auto discovery ignores the actual paload in the toggle mqtt message and it toggled for all messages on the topic.
This conflicts with SwitchModes 5/8 which send the HOLD payload on the same topic but this then also triggers the toggle in HA,

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
7:37:49 MQT: SM836101C514/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
07:37:49 MQT: SM836101C514/stat/RESULT = {"Module":{"1":"Sonoff Basic"}}
07:37:50 MQT: SM836101C514/stat/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"10":"Switch2"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"22":"Relay2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here: No rules configured.

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
07:38:55 MQT: SM836101C514/stat/STATUS = {"Status":{"Module":1,"FriendlyName":["SM836101C514","SM836101C514"],"Topic":"SM836101C514","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"0000","SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[8,8,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
07:38:55 MQT: SM836101C514/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:12:35","StartupUTC":"2020-01-26T06:26:20","Sleep":50,"CfgHolder":4617,"BootCount":42,"SaveCount":337,"SaveAddress":"FA000"}}
07:38:55 MQT: SM836101C514/stat/STATUS2 = {"StatusFWR":{"Version":"8.1.0.4(022a6a0-tasmota)","BuildDateTime":"2020-01-25T20:00:12","Boot":7,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"353/699"}}
07:38:55 MQT: SM836101C514/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["UFFHAUS",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["400A8009","2805C8000100060000005A00000000000000","00000200","00000000"]}}
07:38:55 MQT: SM836101C514/stat/STATUS4 = {"StatusMEM":{"ProgramSize":569,"Free":432,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
07:38:55 MQT: SM836101C514/stat/STATUS5 = {"StatusNET":{"Hostname":"SM836101C514-3698","IPAddress":"192.168.0.102","Gateway":"192.168.0.100","Subnetmask":"255.255.255.0","DNSServer":"1.1.1.1","Mac":"D8:F1:5B:DF:CE:72","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
07:38:55 MQT: SM836101C514/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.50","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DFCE72","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
07:38:55 MQT: SM836101C514/stat/STATUS7 = {"StatusTIM":{"UTC":"Sun Jan 26 06:38:55 2020","Local":"Sun Jan 26 07:38:55 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:28","Sunset":"17:36"}}
07:38:55 MQT: SM836101C514/stat/STATUS10 = {"StatusSNS":{"Time":"2020-01-26T07:38:55","Switch2":"OFF"}}
07:38:55 MQT: SM836101C514/stat/STATUS11 = {"StatusSTS":{"Time":"2020-01-26T07:38:55","Uptime":"0T00:12:35","UptimeSec":755,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"UFFHAUS","BSSId":"CC:2D:E0:62:47:0E","Channel":1,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here: During startup the following auto discovery command is issued ...

07:26:29 MQT: homeassistant/binary_sensor/DFCE72_SW_2/config = {"name":"SM836101C514 Switch2","stat_t":"~stat/SWITCH2","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"DFCE72_SW_2","device":{"identifiers":["DFCE72"],"connections":[["mac","D8:F1:5B:DF:CE:72"]]},"~":"SM836101C514/","value_template":"{%if is_state(entity_id,\"on\")-%}OFF{%-else-%}ON{%-endif%}"} (retained)

```

### TO REPRODUCE
SwitchTopic 0
SwitchMode 8
SetOption19 1
Restart 1

### EXPECTED BEHAVIOUR
I am not certain of the exact solution but the value_template should check the actual payload id "TOGGLE" if it is anything else nothing should change. 

My novice attempt:
{% if value_json.STATE == \"TOGGLE\"-%}
{%-if is_state(entity_id,\"on\")-%}OFF{%-else-%}ON{%-endif-%}
{%-else-%}
{{ states(entity_id) }}
{%-endif%}

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2020-01-26:
As this is a Home Assistant configuration issue, we do not have the expertise here to address your question. Please join a discussion on Discord server DigiblurDIY to get assistance.

jensuffhaus, 2020-01-26:
@meingraham This is not a HA issue, the reason the payload is ignored is due to the template that is configured from Tasmota not being correct. I even provide what I think is the correct template.

@ascillato2 could you please link to the duplicated issue.

jensuffhaus, 2020-01-26:
@meingraham here is a link to the code I believe may be where the change needs to happen. I am still too unfamiliar with the structure of Tasmota to begin tinkering with the code.

https://github.com/arendst/Tasmota/blob/9efe67127d796fadb1e7961c8898b93faf67c8c7/tasmota/xdrv_12_home_assistant.ino#L48

ascillato2, 2020-01-27:
Let's gather all equal issues in one for better follow-up.

#7599

closed 2020-01-27 by ascillato2

jensuffhaus, 2020-01-27:
> Let's gather all similar issues in one for better follow-up.
> 
> #7599

@ascillato2  Entirely up to you, but they are very separate issues. This one is a bug that should be fixed, the other is a feature that may or may not be implemented (would be nice if it is tho).

jensuffhaus, 2020-01-28:
> @jensuffhaus,
> at this moment HA discovery supports just `switchmode0` and `switchmode1` and there is no ETA to support other modes.
> `TOGGLE` payload will rise warnings on logs because is not considered a valid payload anymore, same thing for `HOLD`.
> Blame HA devs for that.
> The string you quoted on the .ino for value template is the one suggested on [HA docs](https://www.home-assistant.io/integrations/binary_sensor.mqtt/#toggle-the-binary-sensor-each-time-a-message-is-received-on-state_topic) and is used for wall switches.

Understood, that's not great ...

Then my other suggestion stands and hopefully works around the HA limitation ...

If SwitchMode has "HOLD" enabled (5 & 8) then Tasmota should announce another binary sensor to HA for the HOLD function with its own topic etc, that way HA can have two sensors, one for toggle and one for hold and the limitation you mentioned will not be a problem.

meingraham, 2020-01-28:
Over time, HA auto-discovery has become more complex/complicated and HA devs seem to be either putting up roadblocks or just dragging their feet toward working collaboratively with Tasmota. There comes a point where we need to assess the added program space, developers' time & effort, and contributors' time to assist users it is taking to maintain HA discovery in Tasmota. As it stands, it cannot discover all entities and seems to be unlikely to improve... or at least continue to function reliably.

At some point, the energy spent on this issue by all involved is probably better spent on other Tasmota enhancements and we should consider deprecating HA auto-discovery. We can instruct users how to manually add the entities in HA via additional documentation. This would be more comprehensive and can cover all types of entities (relays/power, switches, buttons, sensors, etc.). After all, other home automation hub users do this as a regular matter of deployment.

Respectfully.

Mike

ascillato, 2020-01-28:
I agree but I also like the autodiscovery of HA. So far I didn't have enough time to work on that to help effelle (Federico) that have been done a great job on that. I think that users just need to be patience and wait for us to have time or make a pull request with the new features.

So far HA autodiscovery is not so flash hungry, but if it becomes a big feature, we can always deprecate it and move to another approach like an add-on for HA.

So many options, so great things to do in Tasmota, so little time...

effelle, 2020-01-28:
Agree. The new discovery is still far away, mostly for the lack of time. Sorry folks, real life come first. 
If I get a good solution to play well with mDNS suffering with weird update issues (and can have a series of other problems depending by the router/daemon) then we can start to think about pushing it in the main `zeroconf` service already available under HA. In that case Setoption19 will be probably replaced by setoption55. Not in near future anyway.
For now HA Discovery is in maintenance mode and I really would like to avoid new stuff now and just concentrate on code cleaning and optimization to respect @arendst directives.