arendst/Tasmota #7607: Ability to set seconds timer.

49studebaker, 2020-01-26:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I would like to use the Yeti app or other interface to turn on a small 12v water irrigation pump at a certain time and then turn off the pump after 30 seconds.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Ability to set a 30 seconds timer.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
none

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

schrej-zz, 2020-01-26:
Did you really read the documentation?
I think, everything can be configured by timers and rules. See: https://github.com/arendst/Tasmota/wiki/Timers

meingraham, 2020-01-26:
... and `PulseTime`.

Please address any further questions to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://tasmota.github.io/docs/#/Home) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-01-26 by meingraham