arendst/Tasmota #7610: Connecting to Tasmota devices by hostname doesn't work (no name reolved)

EnidanBri, 2020-01-26:
Hi all, 

### PROBLEM DESCRIPTION
I cannot connect to my Tasmota devices by their hostname. I get ERR_NAME_NOT_RESOLVED when I try to access the Web UI. I also cannot ping the devices under their hostname. Accessing and pinging works correctly via the IP. I don't want to assign fixed IPs to them, I would prefer to use the hostname instead.
Please see below for what I have already tried.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Gosund SP1, Gosund SP111, Gosund SP112, AOFO 4 plug + USB
- [X] Tasmota binary firmware version number used: most recent, upgraded today
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: Tasmota Web UI
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
12:30:37 CMD: Backlog Template; Module; GPIO 255
12:30:37 SRC: WebConsole from 192.168.178.22
12:30:37 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
12:30:37 SRC: Backlog
12:30:37 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
12:30:37 MQT: stat/tasmota/sp111-01/RESULT = {"NAME":"Gosund SP111 V","GPIO":[57,255,56,255,132,134,0,0,131,17,0,21,0],"FLAG":0,"BASE":45}
12:30:37 SRC: Backlog
12:30:37 CMD: Group 0, Index 1, Command "MODULE", Data ""
12:30:37 MQT: stat/tasmota/sp111-01/RESULT = {"Module":{"0":"Gosund SP111 V"}}
12:30:38 SRC: Backlog
12:30:38 CMD: Group 0, Index 1, Command "GPIO", Data "255"
12:30:38 MQT: stat/tasmota/sp111-01/RESULT = {"GPIO0":{"57":"Led2i"},"GPIO1":{"0":"None"},"GPIO2":{"56":"Led1i"},"GPIO3":{"0":"None"},"GPIO4":{"132":"HLWBL CF1"},"GPIO5":{"134":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"131":"HLWBL SELi"},"GPIO13":{"17":"Button1"},"GPIO14":{"0":"None"},"GPIO15":{"21":"Relay1"},"GPIO16":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
12:31:23 CMD: Backlog Rule1; Rule2; Rule3
12:31:23 SRC: WebConsole from 192.168.178.22
12:31:23 CMD: Group 0, Index 1, Command "BACKLOG", Data "Rule1; Rule2; Rule3"
12:31:24 SRC: Backlog
12:31:24 CMD: Group 0, Index 1, Command "RULE", Data ""
12:31:24 MQT: stat/tasmota/sp111-01/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
12:31:24 SRC: Backlog
12:31:24 CMD: Group 0, Index 2, Command "RULE", Data ""
12:31:24 MQT: stat/tasmota/sp111-01/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
12:31:24 SRC: Backlog
12:31:24 CMD: Group 0, Index 3, Command "RULE", Data ""
12:31:24 MQT: stat/tasmota/sp111-01/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
12:39:45 CMD: status 0
12:39:45 SRC: WebConsole from 192.168.178.22
12:39:45 CMD: Group 0, Index 1, Command "STATUS", Data "0"
12:39:45 MQT: stat/tasmota/sp111-01/STATUS = {"Status":{"Module":0,"FriendlyName":["Gosund SP111 - 1"],"Topic":"tasmota/sp111-01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:39:45 MQT: stat/tasmota/sp111-01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota-minimal.bin","RestartReason":"Software/System restart","Uptime":"0T00:14:21","StartupUTC":"2020-01-26T11:25:24","Sleep":50,"CfgHolder":4617,"BootCount":28,"SaveCount":117,"SaveAddress":"F5000"}}
12:39:45 MQT: stat/tasmota/sp111-01/STATUS2 = {"StatusFWR":{"Version":"8.1.0.4(022a6a0-tasmota)","BuildDateTime":"2020-01-25T20:00:12","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"363/699"}}
12:39:45 MQT: stat/tasmota/sp111-01/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ANBKB-iot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000001","00000000"]}}
12:39:45 MQT: stat/tasmota/sp111-01/STATUS4 = {"StatusMEM":{"ProgramSize":569,"Free":432,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
12:39:45 MQT: stat/tasmota/sp111-01/STATUS5 = {"StatusNET":{"Hostname":"tasmota/sp111-01-2639","IPAddress":"192.168.179.84","Gateway":"192.168.178.1","Subnetmask":"255.255.254.0","DNSServer":"192.168.178.222","Mac":"2C:F4:32:DB:EA:4F","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
12:39:45 MQT: stat/tasmota/sp111-01/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.222","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DBEA4F","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
12:39:45 MQT: stat/tasmota/sp111-01/STATUS7 = {"StatusTIM":{"UTC":"Sun Jan 26 11:39:45 2020","Local":"Sun Jan 26 12:39:45 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:28","Sunset":"17:36"}}
12:39:45 MQT: stat/tasmota/sp111-01/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
12:39:45 MQT: stat/tasmota/sp111-01/STATUS10 = {"StatusSNS":{"Time":"2020-01-26T12:39:45","ENERGY":{"TotalStartTime":"2020-01-14T11:55:33","Total":0.795,"Yesterday":0.065,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
[... 48 lines cut ...]
_Steps to reproduce the behavior:_
Restart the device. No messages in the log that it tries to use DNS.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I would expect to be able to connect to the Web UI of the device by using its hostname.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
Not applicable

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Our DNS server is a Synology NAS with IP 192.168.178.222. Using hostnames to access devices works fine for other devices like Raspberry Pis etc. 
Our domain does not end with .local, so it should not be the mDNS issue. I tried to SetOption55 ON and OFF, but it doesn't make a difference.
What I find strange is that in the log, I never even see that it tries to connect to the DNS. Here in the issues, I found posts which contained messages like "DNS failed" or so, but I don't see anything like this, or about DNS at all, in the log. 
I checked the commands section for a command to enable DNS, but did not find one. 
I did find the command IPAddress4 there and set it to the IP of our NAS, but without any difference.

Does anybody have a suggestion what I could do?

Thanks & Regards,
Nadine

**(Please, remember to close the issue when the problem has been addressed)**

joba-1, 2020-01-27:
Unlikely this is a tasmota issue.
Name resolving is done by the dns server in cooperation with the dhcp server (based on hints from tasmota as the dhcp client). Both your synology NAS, I assume.

Anyways: this looks like the configured hostname to me: „tasmota/sp111-01-2639“. 
The dns server probably does not like “/“ in the name.

ascillato2, 2020-01-27:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

EnidanBri, 2020-01-27:
Yes, I noticed that "/" in the hostname also, and changed it in the meantime to "sp111-01", which should be a valid hostname. The other devices which I can also not access via hostname did not have a slash in their name. So that's not the answer.

I think that this must probably be a tasmota issue, because I do not have problems using the hostname with any other device in our network.

You closed this a bit too early. It is not solved. Can you please reopen the issue?

Thanks & Regards,
Nadine

ascillato, 2020-01-27:
your device has IP 192.168.179.84 and your DNS server has IP 192.168.178.222

Did you configure the correct mask for those subnets to connect each other?

EnidanBri, 2020-01-27:
Yes, that's true, and it we configured it correctly. It does work for all my Raspis and ESP32s, these are also in the 179 subnet. (This kind of stuff is my husbands job, so I assume he did this correctly.) He already looked at the issue, and concluded that it must be the tasmota devices, because the settings (inkl. subnet mask) are OK.

And I think he is right, because no other device has problems with the DNS. 
If you think it's not the tasmota devices, can you explain why don't I see anything in the log that the devices **try** to connect to the DNS although I entered the IP of the DNS server? 
I guess there must be _something_ which prevents them from doing so, but I can't find an option or so... Maybe something in the code which cannot handle subnets?

ascillato2, 2020-01-27:
Reopening as you requested.

So far I couldn't reproduce this issue.

Your subnet is configured as 255.255.254.0 in your Tasmota device.

Please, try for testing 255.255.0.0

ascillato2, 2020-01-27:
Also try in the console: `SetOption55 1`

joba-1, 2020-01-27:
The ip config is correct as it is. 255.255.0.0 should work as well to reach the gateway or dns, but would block traffic to other 192.168.0.0 subnets (if there are any).

Tasmota has no need to resolve its own hostname via dns. 
If you configure the hostname of the mqtt server, instead of the ip, it would need to contact dns to get the ip and you get errors if that doesn‘t work.

I am still convinced it is a dhcp/dns issue where the dns server does not accept the hostname from the dhcp server or the dhcp server does not try to register the name with the dns server at all when allocating the tasmota ip.

Have you looked at the NAS logs already during the time the tasmota device gets a new ip? What is used as nameserver and/or dhcp server? bind/iscp dhcpd? Dnsmasq? I try, but a Synology forum could probably help much better.

joba-1, 2020-01-27:
p.s.: maybe it is necessary to force the dhcp server to use a new ip to trigger the name registration with the fixed hostname

EnidanBri, 2020-01-27:
I mentioned SetOption 55 already in my first post. That did not make a difference. 

As joba-1 already pointed out, the subnet mask is correct. We wanted to be able to use 510 IP addresses instead of 255. Using subnet mask 255.255.0.0 would give us 65535 possible IP addresses, which is waaay too much. You can check that with an IP calculator like:
https://www.heise.de/netze/tools/netzwerkrechner/
Fill in 192.168.178.0 as IP and 23 in CIDR-Suffix. You will get this subnet mask and the possible IP addresses between 192.168.178.1 and 192.168.179.254.

All of your suggestions did not make a difference. 
When I used the hostname for the MQTT server instead of IP, the MQTT continued to work. Apparently it is resolving the name without an entry in the weblog (set to log level 4)?
But still, I cannot connect to the Tasmota devices by their hostname. 
I just don't get why it's working fine for all other devices except for the Tasmotas. 

However, I will follow a different route now: 
I will log the telemetry INFO2 MQTT messages. These contain the hostname and IP address. This way, I can create my own lookup table from hostname to IP address. Doing so, I won't have to assign static IP addresses, but can still access the WebUI by IP.

So you can close the issue, although it wasn't solved.

Thanks anyway for your help!

ascillato, 2020-01-27:
> I mentioned SetOption 55 already in my first post.

You mention` setoption55 ON` that is not a valid command. That is why I asked you to try `setoption55 1`

Anyway, good to know that you have found a workaround to your issue.

closed 2020-01-27 by ascillato2

barbudor, 2021-10-10:
@CoolID2 This issue is closed.  Whatever this issue was about, it was the original author use case. You case may be different.
If you feel that you have found a bug in Tasmota, please open a new issue and fill up all informations required.
If you feel that you need help & support in configuring Tasmota, open a Discussion or join the Discord server. Providing detailled information on your setup will be required too.