arendst/Tasmota #7613: MQTT answer for Timers command

nambuco, 2020-01-26:
### PROBLEM DESCRIPTION
I am trying to globally enable/disable tasmota Timers from Home Assistant. I am using a MQTT Switch entity with state_topic: `stat/sonoff_piscina/timers` .

The problem is that Timers command result is not only the global timer status (on / off) but also the status of all timers (5 messages in a sequence) remaining only the last one in the MQTT server. This last message does not have global timer information. See answer bellow:
```
12:29:49 CMD: timers
12:29:49 MQT: stat/sonoff_piscina/TIMERS = {"Timers":"ON"}
12:29:49 MQT: stat/sonoff_piscina/TIMERS = {"Timers1":{"Timer1":{"Arm":1,"Mode":0,"Time":"08:00","Window":0,"Days":"1111111","Repeat":1,"Output":1,"Action":1},"Timer2":{"Arm":1,"Mode":0,"Time":"11:00","Window":0,"Days":"1111111","Repeat":1,"Output":1,"Action":0},"Timer3":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer4":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0}}}
12:29:49 MQT: stat/sonoff_piscina/TIMERS = {"Timers2":{"Timer5":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer6":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer7":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer8":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0}}}
12:29:49 MQT: stat/sonoff_piscina/TIMERS = {"Timers3":{"Timer9":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer10":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer11":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer12":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0}}}
12:29:49 MQT: stat/sonoff_piscina/TIMERS = {"Timers4":{"Timer13":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer14":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer15":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer16":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0}}}
```
I am using `SetOption4 ON` !

Because of those multiple answers, I am not able to capture the timers state.

Is there a way to have only the global timer state (the first msg) as a answer or answer in multiple different topics like for `status 0`?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff POW R2
- [X] Tasmota binary firmware version number used: 8.1.0.4
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: VS / Platformio
- [X] Flashing tools used: OTA
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
12:48:10 CMD: Backlog Template; Module; GPIO 255
12:48:11 MQT: stat/sonoff_piscina/TEMPLATE = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
12:48:11 MQT: stat/sonoff_piscina/MODULE = {"Module":{"6":"Sonoff Pow"}}
12:48:11 MQT: stat/sonoff_piscina/GPIO = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"130":"HLWBL SEL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"132":"HLWBL CF1"},"GPIO14":{"133":"HLW8012 CF"},"GPIO15":{"52":"Led1"},"GPIO16":{"0":"None"}}
```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
12:49:57 CMD: Backlog Rule1; Rule2; Rule3
12:49:58 MQT: stat/sonoff_piscina/RULE = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
12:49:58 MQT: stat/sonoff_piscina/RULE = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
12:49:58 MQT: stat/sonoff_piscina/RULE = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
```
- [X] Provide the output of this command: `Status 0`:
```
12:50:48 CMD: status 0
12:50:48 MQT: stat/sonoff_piscina/STATUS = {"Status":{"Module":6,"FriendlyName":["Piscina"],"Topic":"sonoff_piscina","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
12:50:48 MQT: stat/sonoff_piscina/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T13:07:22","StartupUTC":"2020-01-26T02:43:26","Sleep":50,"CfgHolder":4617,"BootCount":52,"SaveCount":388,"SaveAddress":"FB000"}}
12:50:48 MQT: stat/sonoff_piscina/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2020-01-19T16:39:31","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"323/699"}}
12:50:48 MQT: stat/sonoff_piscina/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Time_Machine",""],"TelePeriod":60,"Resolution":"55C180C0","SetOption":["00008039","2805C8000100068000005AFF000000000000","00000001","00000000"]}}
12:50:48 MQT: stat/sonoff_piscina/STATUS4 = {"StatusMEM":{"ProgramSize":574,"Free":428,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","8FDAE397","043683A1","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
12:50:48 MQT: stat/sonoff_piscina/STATUS5 = {"StatusNET":{"Hostname":"sonoff_piscina","IPAddress":"10.0.1.212","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.1","Mac":"2C:3A:E8:07:1E:08","Webserver":2,"WifiConfig":4}}
12:50:48 MQT: stat/sonoff_piscina/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.209","MqttPort":1883,"MqttClientMask":"sonoff_piscina","MqttClient":"sonoff_piscina","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
12:50:48 MQT: stat/sonoff_piscina/STATUS7 = {"StatusTIM":{"UTC":"Sun Jan 26 15:50:48 2020","Local":"Sun Jan 26 12:50:48 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"-03:00","Sunrise":"04:28","Sunset":"13:36"}}
12:50:48 MQT: stat/sonoff_piscina/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
12:50:48 MQT: stat/sonoff_piscina/STATUS10 = {"StatusSNS":{"Time":"2020-01-26T12:50:48","ENERGY":{"TotalStartTime":"2019-11-10T17:01:58","Total":177.402,"Yesterday":2.071,"Today":3.179,"Power":661,"ApparentPower":698,"ReactivePower":226,"Factor":0.95,"Voltage":207,"Current":3.376}}}
12:50:48 MQT: stat/sonoff_piscina/STATUS11 = {"StatusSTS":{"Time":"2020-01-26T12:50:48","Uptime":"0T13:07:22","UptimeSec":47242,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Time_Machine","BSSId":"F0:D1:A9:0F:54:74","Channel":6,"RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:08"}}}
```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
12:52:03 MQT: stat/sonoff_piscina/TIMERS = {"Timers":"ON"}
12:52:03 MQT: stat/sonoff_piscina/TIMERS = {"Timers1":{"Timer1":{"Arm":1,"Mode":0,"Time":"08:00","Window":0,"Days":"1111111","Repeat":1,"Output":1,"Action":1},"Timer2":{"Arm":1,"Mode":0,"Time":"11:00","Window":0,"Days":"1111111","Repeat":1,"Output":1,"Action":0},"Timer3":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"00000 [... cut ...] tion":0},"Timer16":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0}}}
```

### TO REPRODUCE
Send `timers` command on console or mqtt

### EXPECTED BEHAVIOUR
multiple results as described before

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
not applicable

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
not applicable

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-01-26:
Which home automation software are you using? In home assistant is easy to capture that.
Tasmota respects json format, so a parsing in any home automation should be feasible.

mikroelektron, 2020-01-26:
i can easy capture the status of 16 times. I have build my own mobile app, and i can read the status of 16 timers and set any one.

nambuco, 2020-01-26:
> Which home automation software are you using? In home assistant is easy to capture that.
> Tasmota respects json format, so a parsing in any home automation should be feasible.

Hello Ascillato, I am using Home Assistant V 0.104.2.

Actually I just changed my mqtt topic from `stat/sonoff_piscina/timers` to `stat/sonoff_piscina/TIMERS` and it is working now, despite the additional answers.

If this multiple answers is not a issue for other applications, we can close this issue.

Thanks

ascillato, 2020-01-26:
Ok. Please close it

ascillato2, 2020-01-27:
Great to know that now it is working for you. Closing this issue. Thanks.

closed 2020-01-27 by ascillato2