arendst/Tasmota #7618: Blinds open position AND tilt position

bjschuit, 2020-01-27:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
There is only the possibility to set the position of blinds, which is limited for controlling venetian blinds.

**Describe the solution you'd like**  
To be able to set both the position of venetian blinds (up-down) as well as the tilt/orientation/angle of the slats using a single powered motor controlled by a Sonoff Dual/4CH Pro R2.

**Describe alternatives you've considered**  
Timing it myself and send timed Power commands. 

**Additional context**  
I want it to be discovered via MQTT in Home Assistant to have it look like this (example):
[image]
Where you can set both the tilt/orientation of the slats as well as the position. 

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-01-27:
See #7471

ascillato2, 2020-01-27:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

bjschuit, 2020-01-27:
@ascillato2 I do not want to be rude, but I think I don't understand what you mean with duplicated. I read the guidelines after I initially posted this request wrong, is that what you mean? 
I've been following that specific issue you are referring to. However, to my knowledge it does not address the functionality I describe above. Forgive if I'm mistaken.

ascillato2, 2020-01-27:
yes, you are right. my mistake.

stefanbode, 2020-02-24:
Can you give more information about how this is implemented in other software solutions. I would think about to define two shutters. One for up/down and one for the Venetian stuff. Can you please describe in a more detailed way what kind of functionality you expect

stefanbode, 2020-02-25:
As far as I understand the venetian blind they operate like this: if going down the tilt is left, if going up the tilt is right. If you want to go down and tilt right you go down and then for 1-2 seconds go up and change the tilt. This means there is no movement in the area where you change the tilt, but you have to capture the current tilt to ensure proper movement. All not very easy.

bjschuit, 2020-02-25:
Yes it is exactly like that. So the thing is that a user needs to 'time'/calibrate the up-down movement AND time/calibrate the tilt movement.

The venetian blinds have just one axis and motor. Moving up first tilts the slats in one direction, then starts moving up. Moving down first tilts the slats in the other direction, then moves down. 

 It has some challenges incorporated indeed, but mostly in timing/calibration which the user should perform. So you not only need to calibrate up- and down-movement, but also calibrate the tilt of the slats. 

How I would prefer to see it is that it uses the same shutter layout, but optionally (based upon setting a checkbox or general setting) adds a second slider for the slat tilt position. 
The calibration can be done quite the same way, however there should be two calibrations: one for tilting, one for up-down movement. They 'share' the mechanical and motor delay. As the tilting is always mechanically triggered first, it might make sense to calibrate that first. It can use the exact same calibration mechanism as currently implemented, it simply runs for a shorter time than usual. Then the second step for full up- and down needs to be implemented. The 'good' thing is that the fine grained calibration is most likely more interesting for the tilting part than for up/down. Because normally venetian blinds are always all the way down, except when you want to open or clean the window. So the tilting is the main function, up/down is secondary. 

I hope this clarifies my request.

stefanbode, 2020-02-26:
Ok, great understand the problem. Lets assume if going down tilt is down and if going up tilt is up. Having two sliders (2 Shutters defined, Both with Shutterrelay1) First long run time 15sek for up and down and second one short run time 2sec tilt down/up. If going down I first use shutter1 to go down and then tilt 0% down too with a rule. If going up I first use shutter1 to go up and then the tilt 100% up with a rule too. So after a movement the position should be ok. There a new rules possible with the latest commit from yesterday.
```
rule1 on Shutter1#Direction!=0 DO mem1 %value% ENDON
on Shutter1#Direction=0 DO IF mem1=1 THEN backlog mem1 0; shutterposition2 100 ENDIF ENDON
on Shutter1#Direction=0 DO IF mem1=-1 THEN backlog mem1 0; shutterposition2 0 ENDIF ENDON

```
I did not test this rule. Just an idea of how it could work. After a movement the tilt has to be set again, because this was changed by the movement.

megamarco833, 2020-03-18:
that's a great contribution to have also the venetian blinds that can be controlled by tasmota!
just one more thing, generally the venetian blinds has two pushbuttons: 1 x UP and 1 x DOWN
and generally they have two purpose: 1) tilting the venetians right or left  2)start the motors in contue to go up or down.

it's very usefull in this case implement short press and long press
with short press of push button generally it activate the motor for 1second to have the tilt
with long press of push button it activate continuosly the motors till reach total open or total close (dipends of course if you press pushbutton UP or Down)
of course if you longpress the push button UP it activate continuosly the motor and the venetian go up, but if you press the pushbutton UP again, or pushbutton Down, it will stop the moviment of venetian blinds

is it possible to implement also this functionality for pushbuttons related to venetian blind?
maybe is usefull to have inside the firmware one option for normal blind or venetian blind

what do you think?

> 
> 
> Ok, great understand the problem. Lets assume if going down tilt is down and if going up tilt is up. Having two sliders (2 Shutters defined, Both with Shutterrelay1) First long run time 15sek for up and down and second one short run time 2sec tilt down/up. If going down I first use shutter1 to go down and then tilt 0% down too with a rule. If going up I first use shutter1 to go up and then the tilt 100% up with a rule too. So after a movement the position should be ok. There a new rules possible with the latest commit from yesterday.
> 
> ```
> rule1 on Shutter1#Direction!=0 DO mem1 %value% ENDON
> on Shutter1#Direction=0 DO IF mem1=1 mem1 0; shutterposition2 100 ENDIF ENDON
> on Shutter1#Direction=0 DO IF mem1=-1 mem1 0; shutterposition2 0 ENDIF ENDON
> ```
> 
> I did not test this rule. Just an idea of how it could work. After a movement the tilt has to be set again, because this was changed by the movement.

that's a great contribution to have also the venetian blinds that can be controlled by tasmota!
just one more thing, generally the venetian blinds has two pushbuttons: 1 x UP and 1 x DOWN
and generally they have two purpose: 1) tilting the venetians right or left  2)start the motors in contue to go up or down.

it's very usefull in this case implement short press and long press
with short press of push button generally it activate the motor for 1second to have the tilt
with long press of push button it activate continuosly the motors till reach total open or total close (dipends of course if you press pushbutton UP or Down)
of course if you longpress the push button UP it activate continuosly the motor and the venetian go up, but if you press the pushbutton UP again, or pushbutton Down, it will stop the moviment of venetian blinds

is it possible to implement also this functionality for pushbuttons related to venetian blind?
maybe is usefull to have inside the firmware one option for normal blind or venetian blind

what do you think?

osr00, 2020-03-28:
cool idea, what i do not understand, how can i create a second shutter for the same shutterrelay1?

stefanbode, 2020-04-10:
Second question just easy to answer. Just set shutterrelay1 1 
And
SHUTTERRELAY2 1

Just double assign the same relay to two shutters.

megamarco833, 2020-04-10:
But what is the advantage to have shutter1 and shutter1 on same relay?
Relay1 should for down and relay2 for up (example)

stefanbode, 2020-04-10:
It is shutter 1 and 2. this is the magic

megamarco833, 2020-04-10:
... You mean control two different phisical blind using same relay? For example to close all blind of the house in one shot?

osr00, 2020-04-10:
The idea is to use one shutter control for up/down position and one for the position of the blind. Position of the blind means how much the blinds are open for example

- as 50%
/ 75%
\ 25%
| 0%

stefanbode, 2020-04-10:
You can control the blind and tilt. Justimpement the rules and check it out. Shutter 1 for the opening and shutter 2 for the tilt. Just some brain jogging to understand

megamarco833, 2020-04-10:
now understand, sorry :-)
but was not better/simple, my suggestion to use shorpress and longpress for switches?
shorpress activate relay only for 1seconds, so you have the tilt.
longpress activate relay continuosly so you trigger the motor for up/down

bjschuit, 2020-04-10:
Just got my blinds yesterday and getting the Sonoffs installed shortly. Gonna calibrate and try this out and gonna get back on this!

megamarco833, 2020-04-10:
But do you have "normal" blinds or venetian blinds?
The issue/improvements here is to menage the venetian blind that can be oriented, not only up and down, and so control the inclination tilt.

megamarco833, 2020-04-10:
i reported also here 
[VID_20200410_130609.zip](https://github.com/arendst/Tasmota/files/4461258/VID_20200410_130609.zip)
just look my video please.
inside the video you can see that:

i have 1 blind and 2 relay (up and down) and 2 push buttons (up and down)
if i shor press up button(time<2seconds) i tilt up => trigger relay1 for1second=> trigger motor up direction for 1 second
if i shor press down button (time<2seconds) i tilt down => trigger relay2 for 1second=> trigger motor for 1 second

if i longpress up button (time>=2seconds) i completly go up => trigger relay1 contiuosly=> trigger motor up direction contiusly to reach 100% open
if i longpress down button (time>=2seconds) i completly go down => trigger relay2 contiuosly=> trigger motor up direction contiusly to reach 0% close

if during a longress up or down i pres push button up or push button down i stop the continuos moviment of motor => stop activated relay1 or relay2 (dipend whitch one was actived at that time) and the blind stop.

stefanbode, 2020-04-10:
@megamarco833 for sure we talk about Venetian blind. Normal blind do not need two shutters.*bang my head*. Short and long press will not work, because a button is always only assigned to on shutter. Blocking issue.
I know what you want to do, but in this case the shutter button  must have the ability to trigger different shutters based on the command you send. This is not possible right now and the complete implementation of the button has no pah to enable this. A button is always hard coded related to one shutter and operates only this one.

stefanbode, 2020-04-10:
What about implementing rules? Short press down goto Shutterposition-5 and short press up to Shutterposition+5. Long press do shutteropen or Shutterclose .

effelle, 2020-04-10:
@stefanbode some time ago a couple of (strange) switchmodes were added to control dimmers, switchmode11 and 12. Perhaps can be used for shutters too? 
```
Tasmota will send a TOGGLE command (use Switch#state=2 in rules) when the button is pressed
 for a short time and is then released. When pressing the button (closing the circuit) for a 
long time (set in SetOption32) Tasmota sends repeated INC_DEC (increment or decrement
 the dimmer) commands (use Switch#state=4 in rules) as long as the button is pressed. 
Releasing the button starts a internal timer, the time is set in SetOption32. When released 
for the time set in SetOption32 Tasmota sends a CLEAR command (use Switch#state=6 in rules).
 If the button is pressed again before the timeout Tasmota sends a INV command (use 
Switch#state=5 in rules). The INV command is for the controlling software (home assistant)
 to switch between incrementing and decrementing the dimmer.
```

megamarco833, 2020-04-10:
> 
> 
> What about implementing rules? Short press down goto Shutterposition-5 and short press up to Shutterposition+5. Long press do shutteropen or Shutterclose .

i was testing someting like this:
```
SetOption80 1
ShutterRelay1 1
ShutterRelay2 3
Backlog Interlock 1,2 3,4; Interlock ON
switchmode1 5    // 5=to use long and shorpress
switchmode2 5
switchmode3 5
switchmode4 5
setoption32 20   //20 = secondi
```

rule to pass the position of the blind to domoticz using shutter<x>#target
```
Rule1 ON Shutter1#Target DO publish domoticz/in {"idx":169,"nvalue":2,"svalue":"%value%"} endon ON Shutter2#Target DO publish domoticz/in {"idx":170,"nvalue":2,"svalue":"%value%"} endon ON Shutter3#Target DO publish domoticz/in {"idx":171,"nvalue":2,"svalue":"%value%"} endon ON Shutter4#Target DO publish domoticz/in {"idx":172,"nvalue":2,"svalue":"%value%"} endon
rule1 1
```

rule to use short press and long press 
```
rule2 on switch1#state=2 do backlog power1 on;delay 10;power1 off endon on switch1#state=3 do power1 on endon
on switch2#state=2 do backlog power2 on;delay 10;power2 off endon on switch2#state=3 do power2 on endon
on switch3#state=2 do backlog power3 on;delay 10;power3 off endon on switch3#state=3 do power3 on endon
on switch4#state=2 do backlog power4 on;delay 10;power4 off endon on switch4#state=3 do power4 on endon
rule2 1
```

why for you this not could change?
thanks for the patience :-)

megamarco833, 2020-04-10:
> @stefanbode some time ago a couple of (strange) switchmodes were added to control dimmers, switchmode11 and 12. Perhaps can be used for shutters too?
>
> ```
> Tasmota will send a TOGGLE command (use [...]

what's suggested by effelle maybe could also make it easy...but i need to concentrate a bit more to  understand how work switchmode11 and switchmode12 becouse not cristal clear to m right now to try to implement it in a new rule, but if i understand the aim looks primising

metter, 2020-04-15:
Tilting the slats on your venetian blinds:

blink the relays on your shutter to open the slats.
**blinktime 8** (find out which number gets you the angle you need. It was 8 for me.)
**blinkcount 1**
**power1 3** (or **power2 3**, depending on your wiring)

cudos goes to [this guy ](https://groups.google.com/forum/#!topic/sonoffusers/EOXWIQx1iJo)who gave the tip.

megamarco833, 2020-04-15:
hi, thats's another great suggestion!
i used delay commands inside rule:

```
rule2 on switch1#state=2 do backlog power1 on;delay 10;power1 off endon on switch1#state=3 do power1 on endon
on switch2#state=2 do backlog power2 on;delay 10;power2 off endon on switch2#state=3 do power2 on endon
```

with your suggestion, suing `blinktime 10` and `blinkcount 1` could be:
```
rule2 on switch1#state=2 do power1 3 endon on switch1#state=3 do power1 on endon
on switch2#state=2 do power2 3 endon on switch2#state=3 do power2 on endon
```

i tested both rules, but it seams that thre first rule, using `delay 10` trigger for a bit more time than second rule using blinktime.
is it normal?
what's the best way for expert  to do this?
thanks

p.s.
i tested also switchmode11 as @effelle suggest some postabove, but i do not find a way to use it in a proper way for this scope :(

 @stefanbode do you have some "preference" or suggestion to choose the best rule? :-)
thanks again for big support to all team!

closed 2020-05-10 by ascillato2

andrejs2, 2021-12-17:
@stefanbode and guys in this topic!
Big thanks for all your effort on this project and time and brains you put in this, I really appreciate it!
I'm quite a noob, not an engineer or any near of knowledge of programming, but 1+ year of experiance in home automation:).
I would kindly ask for some help explaning some critical parts for understanding logic of tasmota firmware...

I read through and through the doc and wiki, got some really good explanations about principles but I'm missing the part **how to configure tilt with `shuttertilt` and not with `shutter2` (position)?**

Anyone has any suggestion how to configure this?

And second Q:
How are the angles of blinds used...in wiki we can read:

`The configuration need following parameters: angle of blinds during OPEN, angle of blinds during CLOSE. This are the max and the min values of the venetian blinds (e.g. -90° to 90°). Additionally the runtime is required from min to max and reverse. This is typically 1-2sec. The resolution of the time is 0.05sec. Duration in [sec] must be multiplied by 20. e.g. 1.2sec => 1.2 x 20 = 24. Two open and close the tilt you can define the angle for OPEN and the angle for CLOSE of the tilt. shuttertiltconfig1 -90 90 24 0 -90`

But where is -90, 90 0 angle...is it like N, S etc. and how it corresponds to blinds? Is it closed position with angle 0 or 180, copen position is 90 od -90? Any hints?

I have been digging this for quite some time, but ending with shutter opening and closing for indefinite time :) and have to restart the whole thing.

I would really appreciate for some guidance and hints how to configure to get is working also in HA.
Thanks!

stefanbode, 2021-12-17:
If you look from the side on your venetian blind the blades have an angle against the horizontal position. 0 = horizontal open. Then -90° and 90° is closed in one or the other direction. The min and max is never higher than 90. In the italien it is 0..85. because during opening the blind are horizontal.

andrejs2, 2021-12-18:
Thank you @stefanbode for this!

romi87, 2022-01-23:
@andrejs2 I have exactly the same problem as you. Based on the manual I configured everything correctly, but the blinds are opening and closing constantly (tilting) while staying in the same position vertically. When I set the tilt position with the command "shuttertilt" the blinds move to the vertical position which I described as the tilt position.

stefanbode, 2022-01-24:
I still do not understand the problem. The documentation should be quite clear about possible options and limitations.

stefanbode, 2022-01-24:
I assume there is some extra configuration for HA required

romi87, 2022-01-24:
@stefanbode my only guess is that when the tilting action happens the blinds are actually changing the up / down positions as well, so when I open the tilt on the blinds the bottom of the blinds are actually moving up and the opposite is true when I am closing the tilt. So my guess is that the blinds are trying to maintain the position after tilt but it is not possible because when you tilt them they have to move the position as well slightly. The blinds are trying to keep the position and that's why they start to close the tilt after they opened it halfway for example. I am a programmer myself, so I decided to look into the code to check what's going on. If I come to a conclusion will let you all know.

megamarco833, 2022-01-24:
@romi87 i have venetian blind ("italian" model where the inclination of wings is from -20 to +5
i'm using the tilt mode thanks to @stefanbode and it works.
without change the position of the blind i can change the inclination of the wings.
example: if i'm at 45% opening and tilt of wings at -10 i can issue position at 40% the blind will inclinate the wings at +5 (opening position) than move up to reach 40% and after that the blind restore the wing position at -10 (start position)

i can also still having 40% (so not change the position of the blind) but i can change the inclination of the wings from -10 (starting wing inclination) to +5 for example without change the blind position that will remain at 40%

i'm using the tilt option with my blind and it works very well

romi87, 2022-01-24:
@megamarco833 I assume the bottom of your blinds do not move up and down during the tilting process. Am I right? For me it is a synchronous movement.

megamarco833, 2022-01-24:
correct, as mention on my example i can change the wings tilt angle without change the positioning of the blind.
so in other words i can keep fixed the position of the blind but i can change the inlcination of the wings

stefanbode, 2022-01-24:
Why is it a problem it the shutter moves a little bit up/down if it changes the tilt. Should be that much. Normally, as long as the tilt is not in the endposition FIRST the tilt will change and THEN the position will change. Can you share a youtube video to visualize the effect? Anyow I assume we cannot do much against this behaviour.

stefanbode, 2022-01-24:
Maybe @romi87 you can send some logs when this "up down constantly" happen

romi87, 2022-01-24:
@stefanbode sure. I will make a video and send you some logs as well. It is not a problem, if the shutter moves up/down a little bit, but I think that is what is causing the issue. The tilting needs to be done almost always in the endposition though (for my case). The position of the blinds rarely changes, because the light coming through the blinds is regulated with the tilt, so the position remains closed almost always and the tilt changes more often.

stefanbode, 2022-01-27:
There is a known Limitation with the tiltduration. Changing the tilt e.g in 0.5 sec will never work with tasmota far to slow to set a specific angle. If the tiltprocess form one end to the other is 1second or better longer then all is fine

stefanbode, 2023-03-31:
There is a new enhanced ESP32 version you can enable with the following compiler option #define USE_SHUTTER_ESP32. Will be default in future versions for ESP32 after some testing. New features most likely only in ESP32 due to memory limitations in ESP8266. Current feature set:

up to 16 shutters,
up to 32 shutterbuttons,
new shutterposition pos,tilt
tilt definition with buttons.
Configuration saved to filesystem.

stefanbode, 2023-03-31:
I already updated the help pages. You can now define with `shutterbuttons` also the TILT you want to have at this position. Before only the position was updated and the tilt stay the same. regarding to this functionality i enhanced the "shutterposition" command to add with a "," the tiltposition. If this tiltposition is there any automatic calculation of the tilt will be disabled. You get what you want. e.g. `shutterposition1 50,0` moves the shutter to 50% and the tilt to 0°