arendst/Tasmota #7619: Support DiSEqC

cribskip, 2020-01-27:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Provide a DiSEqC module on a arbitrary output in order to control DiSEqC motors for e.g. heliostats or solar trackers

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
Putting an Arduino between the tasmota an the dish motor and using the mqtt serial bridge feature

**Additional context**  
_Add any other context or screenshots about the feature request here._
Over at ETH Zurich http://www.e-callisto.org/Hardware/Diseqc/Easy_Diseq/Easy_Diseq.ino is available to steer two motors.

**(Please, remember to close the issue when the problem has been addressed)**  

I'll take a look on how to integrate this into tasmota over the following months

ascillato2, 2020-01-27:
Added the label feature request to see if there is someone interested on donating her/his time to work on this.

Anyway, at this moment, you can use the _Easy_Diseq.ino_ file you provided in an arduino nano. Your sketch is completed as a SERIAL to DiSEqC converter, so then Tasmota can communicate by serial with your nano, and you could control your DiSEqC from MQTT.

The specific 22khz burst is not easy to implement in a driver due to adding many delays can degrate other Tasmota features, so seems that a pwm approach should be better (?)

ascillato2, 2020-02-10:
Closing as nobody is working on this. If anyone is interested on working on this, please provide a PR. Thanks.

closed 2020-02-10 by ascillato2