arendst/Tasmota #7620: Support for Sonoff Micro

saikek, 2020-01-27:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
This post is intended to store information about `Sonoff Micro` USB device.

**Describe the solution you'd like**  
It should be possible to use `Sonoff Micro` with Tasmota.

**Describe alternatives you've considered**  
Not Much.

**Additional context**  
There is a new chip CKW04 that was introduced by ITead guys. Support for Tasmota is unknown at this moment.

**(Please, remember to close the issue when the problem has been addressed)**

saikek, 2020-01-27:
# Dissasembly
I've just received mine - in order to dissasemble it i had to scratch case all over. 
It seems everything is much simpler - you need really thin screwdriver to be inserted exactly where red marks are. Than it will un-snap and you can use another scredriwer to un-snap other sides.

[image]
[image]

saikek, 2020-01-27:
# SPI connection
Board contains `V`, `G, `R`, `T` Pin group on one side 

and `LOG`, `R`, `T`, `RST` and button on the other. 

I've tried connecting my programmer to the device (3.3V), but default way of copying firmware does not work here  - not an ESP-compatible.
```
@esp_tool -pCOM12 -b115200 -a0x0000 -s0x100000 -or esp82XX.bin
```

saikek, 2020-01-27:
More images of device:
[image]
[image]

meingraham, 2020-01-27:
@saikek 

Tasmota is only compatible with the ESP8266/ESP8285 Wi-Fi SoC. esp_tool is provided by Espressif to work with their chips. The Micro uses a different chip. Neither esp_tool nor Tasmota are supported on this chip.

closed 2020-01-27 by meingraham

ascillato2, 2020-01-27:
Sorry, Tasmota only works on ESP8266 and ESP8285 devices.