arendst/Tasmota #7621: "Deep-Sleep-Wake" does not work reliably after PR #7602

Einstein67, 2020-01-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After PR #7602 changes "Deepsleep" ist not working reliable. Device Uptime and Power-Consumption are much higher as with Tasmota 8.1.0.4

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Wemos mini D1
- [ ] Tasmota binary firmware version number used: 8.1.0.5
  - [ ] Pre-compiled
  - [x ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
08:17:10 MQT: stat/wem4/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
08:17:10 MQT: stat/wem4/RESULT = {"Module":{"18":"Generic"}}
08:17:11 MQT: stat/wem4/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"5":"I2C SCL"},"GPIO2":{"0":"None"},"GPIO3":{"6":"I2C SDA"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"182":"DeepSleep"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
08:18:15 CMD: status 0
08:18:15 MQT: stat/wem4/STATUS = {"Status":{"Module":18,"FriendlyName":["KlimaMessung-mobil"],"Topic":"wem4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"0001","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:18:15 MQT: stat/wem4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"5N1","GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bix","RestartReason":"Software/System restart","Uptime":"0T00:02:01","StartupUTC":"2020-01-28T07:16:14","Sleep":50,"CfgHolder":4617,"BootCount":234,"SaveCount":2829,"SaveAddress":"3F8000"}}
08:18:15 MQT: stat/wem4/STATUS2 = {"StatusFWR":{"Version":"8.1.0.5(tasmota)","BuildDateTime":"2020.01.27 16:18:27","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"365/699"}}
08:18:15 MQT: stat/wem4/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Einstein-N450","Einstein-54G"],"TelePeriod":300,"Resolution":"958180C0","SetOption":["00008009","2805C8000100060000005AFF000000000000","00800048","00000000"]}}
08:18:15 MQT: stat/wem4/STATUS4 = {"StatusMEM":{"ProgramSize":475,"Free":2596,"Heap":27,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1625C2","FlashMode":3,"Features":["00000407","0F000286","04100000","020104C5","000000C0","00100800"],"Drivers":"1,2,3,4,9,10,20,21,29","Sensors":"1,2,3,5,6,9,13,22,58"}}
08:18:15 MQT: stat/wem4/STATUS5 = {"StatusNET":{"Hostname":"wemos4","IPAddress":"192.168.0.228","Gateway":"192.168.0.2","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.2","Mac":"B4:E6:2D:29:22:BC","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
08:18:15 MQT: stat/wem4/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.201","MqttPort":1883,"MqttClientMask":"Wem4","MqttClient":"Wem4","MqttUser":"Sonoff","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
08:18:15 MQT: stat/wem4/STATUS7 = {"StatusTIM":{"UTC":"Tue Jan 28 07:18:15 2020","Local":"Tue Jan 28 08:18:15 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":99,"Sunrise":"07:39","Sunset":"17:00"}}
08:18:15 MQT: stat/wem4/STATUS10 = {"StatusSNS":{"Time":"2020-01-28T08:18:15","ANALOG":{"A0":823},"BME280":{"Temperature":1.06,"Humidity":72.6,"Pressure":953.4},"PressureUnit":"hPa","TempUnit":"C"}}
08:18:15 MQT: stat/wem4/STATUS11 = {"StatusSTS":{"Time":"2020-01-28T08:18:15","Uptime":"0T00:02:01","UptimeSec":121,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Einstein-N450","BSSId":"78:D2:94:A8:25:88","Channel":11,"RSSI":48,"Signal":-76,"LinkCount":1,"Downtime":"0T00:00:09"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash latest dev version activate Deepsleep and check Device-Uptime
8.1.0.4 eg 0:15 min/cycle
8.1.0.5 eg 1:45 min/cycle

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Same behaviour as Tasmota 8.1.0.4

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

hoepfner, 2020-02-09:
I have a similar problem. commenting out (that is never used) the follwing line in support_wifi.ino in function WifiBegin solved the problem for me
//WiFi.setAutoReconnect(false);
Can you please try, if this will help to solve your problem too?
Maybe it can turn this on and off with a SetOptionXX in the future.

Einstein67, 2020-02-09:
Thank you for this info.

Have made a test build, and testing the now!

ascillato2, 2020-02-10:
Hi @Einstein67 @hoepfner 

In order to add this to Tasmota, can you test the following?

Instead of commenting out that line, please replace

https://github.com/arendst/Tasmota/blob/504570d5d5bd406ba0bd755e2e03f5bbddf83ede/tasmota/support_wifi.ino#L192

with :

```cpp
if (!(DeepSleepEnabled()) {
  WiFi.setAutoReconnect(false);
}
```

Thanks.

Einstein67, 2020-02-10:
Mhhhm, this code gives a compile error. Im testing with:

`#ifdef USE_DEEPSLEEP
  if (!DeepSleepEnabled()) {
#endif
    WiFi.setAutoReconnect(false);
#ifdef USE_DEEPSLEEP
  }
#endif`

and the result (after 5 DS-Wakes) is the same as commenting out line 192! 

Looks good!! :)

ascillato, 2020-02-10:
Thanks for testing.

Proposed PR https://github.com/arendst/Tasmota/pull/7713

hoepfner, 2020-02-10:
Hi ascillato, Hi Einstein67,

thank you for the fast help!
I believe, the problem not only occures when using deepsleep.
I've a simular problem since the patch 7602 to get a WLAN connection with tasmota-build.
with the tasmota-minimal and tasmota-lite it works.
May be one reason is also, that I have two AP with the same SSID (router + repeater).
When I comment out this line, all is working perfekt.

hoepfner, 2020-02-10:
in the moment I cannot make tests with the new version.
here is a test which made with the old one (8.1.0.5):

00:00:00 CFG: Loaded from flash at FA, Count 536
00:00:00 CFG: CR 331/699
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 8.1.0.5(tasmota)-2_6_1
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:01 WIF: Network (re)scan started...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Network 0, AP1, SSId FRITZ!Box Fon WLAN 7570 vDSL, Channel 11, BSSId 38:10:D5:E3:50:5F, RSSI -53, Encryption 1
00:00:04 WIF: Connecting to AP1 FRITZ!Box Fon WLAN 7570 vDSL Channel 11 BSSId 38:10:D5:E3:50:5F in mode 11N as tasmota-wemos-001-5910...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 CFG: Saved to flash at F9, Count 537, Bytes 4096
00:00:06 QPC: Reset
00:00:06 WIF: Checking connection...
00:00:06 WIF: Attempting connection...
00:00:08 APP: Boot Count 188
00:00:08 CFG: Saved to flash at F8, Count 538, Bytes 4096
00:00:08 WIF: Checking connection...
00:00:08 WIF: Attempting connection...
00:00:10 WIF: Checking connection...
00:00:10 WIF: Attempting connection...
00:00:12 WIF: Checking connection...
00:00:12 WIF: Attempting connection...
00:00:14 WIF: Checking connection...
00:00:14 WIF: Attempting connection...
00:00:16 WIF: Checking connection...
00:00:16 WIF: Attempting connection...
00:00:18 WIF: Checking connection...
00:00:18 WIF: Attempting connection...
00:00:20 WIF: Checking connection...
00:00:20 WIF: Attempting connection...
00:00:22 WIF: Checking connection...
00:00:22 WIF: Attempting connection...
00:00:24 WIF: Checking connection...
00:00:24 WIF: Attempting connection...
00:00:26 WIF: Checking connection...
00:00:26 WIF: Connect failed with AP timeout
00:00:27 WIF: Network (re)scan started...
00:00:28 WIF: Checking connection...
00:00:28 WIF: Attempting connection...
00:00:30 WIF: Network 0, AP1, SSId FRITZ!Box Fon WLAN 7570 vDSL, Channel 11, BSSId 38:10:D5:0F:67:DA, RSSI -84, Encryption 1
00:00:30 WIF: Network 1, AP1, SSId FRITZ!Box Fon WLAN 7570 vDSL, Channel 11, BSSId 38:10:D5:E3:50:5F, RSSI -55, Encryption 1
00:00:31 WIF: Connecting to AP1 FRITZ!Box Fon WLAN 7570 vDSL Channel 11 BSSId 38:10:D5:E3:50:5F in mode 11N as tasmota-wemos-001-5910...
00:00:31 WIF: Checking connection...
00:00:31 WIF: Attempting connection...
00:00:33 WIF: Checking connection...
00:00:33 WIF: Attempting connection...
00:00:35 WIF: Checking connection...
00:00:35 WIF: Attempting connection...
00:00:37 WIF: Checking connection...
00:00:37 WIF: Attempting connection...
00:00:39 WIF: Checking connection...
00:00:39 WIF: Attempting connection...
00:00:41 WIF: Checking connection...
00:00:41 WIF: Attempting connection...
00:00:43 WIF: Checking connection...
00:00:43 WIF: Attempting connection...
00:00:45 WIF: Checking connection...
00:00:45 WIF: Attempting connection...

ascillato, 2020-02-12:
Hi,

I have set 2 routers as you and indeed with the line WiFi.setAutoReconnect(false); it takes longer to jump from one to another and commenting it out makes Tasmota to work faster.

@arendst Theo, any idea about this? Can we comment out this line?

arendst, 2020-02-12:
Did you also set the channels equal? I have no idea why these channels are set equal but the first thing I would do is change one of them to something completely different from 11. Currently both AP's compete on the same channel. Not good.

ascillato, 2020-02-12:
I have channel 6 in one and channel 1 in the other.

arendst, 2020-02-12:
So let's remove the the AutoReconnect(false) and see what happens.

I'll make a PR.

arendst, 2020-02-12:
@s-hadinger this may solve your Ubiquiti issue too.

closed 2020-02-12 by ascillato2

s-hadinger, 2020-02-12:
@arendst Unfortunately it does not solve my Ubiquiti issue.

I tried a 'stock' Tasmota-8.1.0 and it works much better. Wifi connection takes 8 seconds instead of 5 but I can live with that. With the latest version connection fails the first time and it takes roughly 20 seconds to reconnect to Wifi after a reboot.

I didn't dive deeper in the details, but I suspect some change in #7602 (not sure though)

arendst, 2020-02-13:
Just did a new commit with more reverted changes from #7602 

@s-hadinger pls try again.

s-hadinger, 2020-02-13:
@arendst Unfortunately it makes no change. I will be out for a week, and I'll add lots of trace to see what is happening.

Wifi connection still takes 19 seconds most of the time with one failed connection, and sometimes connects in 8 seconds.

What is really strange is that after `Reset 3`, connection is always successful in 5 seconds - which should be the norm.