arendst/Tasmota #7622: Energy values are sometimes = 0

Eligio-Greco, 2020-01-28:
### PROBLEM DESCRIPTION
Ho un PZEM-16 con ESP-01 in funzione da Agosto 2019.
Da qualche mese i valori di: "Energia Oggi", "Energia Ieri" ed "Energia Totale" diventano "0" per qualche lettura.
Ho aggiornato il firmware all'ultima release (8.1.0).
Ho controllato collegamenti e elettronica alla ricerca di difetti evidenti (condensatori gonfi, falsi contatti, etc.).
Ho invertito i collegamenti della seriale: GPIO1 -> Tx e GPIO3 -> Rx.
Ho resettato valori di energia (EnergyReset 1, 2 e 3 a "0") e tasmota (Reset 6).
Il problema non si è risolto... Cosa posso fare per provare a risolvere ?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): PZEM-16 e ESP-01
- [X] Tasmota binary firmware version number used: 8.1.0
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: OTA
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:

10:51:08 MQT: pzem_16_200/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}

10:51:09 MQT: pzem_16_200/stat/RESULT = {"Module":{"18":"Generic"}}

10:51:09 MQT: pzem_16_200/stat/RESULT = {"GPIO0":{"0":"Nessuno"},"GPIO1":{"62":"PZEM0XX Tx"},"GPIO2":{"0":"Nessuno"},"GPIO3":{"98":"PZEM016 Rx"},"GPIO4":{"0":"Nessuno"},"GPIO5":{"0":"Nessuno"},"GPIO9":{"0":"Nessuno"},"GPIO10":{"0":"Nessuno"},"GPIO12":{"0":"Nessuno"},"GPIO13":{"0":"Nessuno"},"GPIO14":{"0":"Nessuno"},"GPIO15":{"0":"Nessuno"},"GPIO16":{"0":"Nessuno"}}

10:51:10 MQT: pzem_16_200/tele/STATE = {"Time":"2020-01-28T10:51:10","Uptime":"0T00:36:02","UptimeSec":2162,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":2,"SSId":"EGS-Smart","BSSId":"00:16:FB:B0:52:20","Channel":10,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:06"}}

10:51:10 MQT: pzem_16_200/tele/SENSOR = {"Time":"2020-01-28T10:51:10","ENERGY":{"TotalStartTime":"2020-01-28T00:00:00","Total":1274.4,"Yesterday":9.8,"Today":3.5,"Period":0.0,"Power":93.6,"ApparentPower":150.6,"ReactivePower":117.9,"Factor":0.62,"Frequency":50.0,"Voltage":245.2,"Current":0.614}}

10:51:10 MQT: pzem_16_200/tele/HASS_STATE = {"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:52:58","Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Module":"Generic","RestartReason":"Software/System restart","Uptime":"0T00:36:02","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:06","MqttCount":1,"BootCount":20,"SaveCount":98,"IPAddress":"192.168.1.200","RSSI":"78","LoadAvg":19}

  Configuration output here:

- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:

  Rules output here:

- [X] Provide the output of this command: `Status 0`:

10:53:28 MQT: pzem_16_200/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["PZEM_16_200"],"Topic":"pzem_16_200","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}

10:53:28 MQT: pzem_16_200/stat/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:38:20","StartupUTC":"2020-01-28T09:15:08","Sleep":50,"CfgHolder":4617,"BootCount":20,"SaveCount":101,"SaveAddress":"F7000"}}

10:53:28 MQT: pzem_16_200/stat/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:52:58","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"416/699"}}

10:53:28 MQT: pzem_16_200/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["EGS","EGS-Smart"],"TelePeriod":10,"Resolution":"548B88C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00400200","00000000"]}}

10:53:28 MQT: pzem_16_200/stat/STATUS4 = {"StatusMEM":{"ProgramSize":567,"Free":436,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashMode":3,"Features":["00000410","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}

10:53:28 MQT: pzem_16_200/stat/STATUS5 = {"StatusNET":{"Hostname":"pzem_16_200-2677","IPAddress":"192.168.1.200","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.10","Mac":"18:FE:34:F5:CA:75","Webserver":2,"WifiConfig":4}}

10:53:28 MQT: pzem_16_200/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.20","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F5CA75","MqttUser":"mosquitto","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}

10:53:28 MQT: pzem_16_200/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue Jan 28 09:53:28 2020","Local":"Tue Jan 28 10:53:28 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:26","Sunset":"17:40"}}

10:53:28 MQT: pzem_16_200/stat/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}

10:53:28 MQT: pzem_16_200/stat/STATUS10 = {"StatusSNS":{"Time":"2020-01-28T10:53:28","ENERGY":{"TotalStartTime":"2020-01-28T00:00:00","Total":1274.4,"Yesterday":9.8,"Today":3.5,"Power":175.5,"ApparentPower":244.1,"ReactivePower":169.7,"Factor":0.72,"Frequency":50.0,"Voltage":244.6,"Current":0.998}}}

10:53:28 MQT: pzem_16_200/stat/STATUS11 = {"StatusSTS":{"Time":"2020-01-28T10:53:28","Uptime":"0T00:38:20","UptimeSec":2300,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":1,"Wifi":{"AP":2,"SSId":"EGS-Smart","BSSId":"00:16:FB:B0:52:20","Channel":10,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:06"}}}

- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-01-28:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-01-28 by ascillato2

rubenzori86, 2020-07-14:
Have you fixed this?? I have the same issue

Eligio-Greco, 2020-07-15:
No...