arendst/Tasmota #7624: Allow Switches/Buttons to be Standalone

BBBits, 2020-01-30:
**Have you looked for this feature in other issues and in the docs?**  
Briefly

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I believe it is a problem when a switch/button isn't associated to a relay it defaults to relay1
Instead of a switch being standalone and using rules or KNX to give it function,
It will always control relay1 and do the rules or KNX events.
This makes no sense to me.
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Change this code in support_tasmota.ino
void ExecuteCommandPower(uint32_t device, uint32_t state, uint32_t source)

if ((device < 1) || (device > devices_present)) { 
device = 1;
}
TO:
  if (device < 1){
    AddLog_P2(LOG_LEVEL_INFO, PSTR("ERR: device < 1 !! device=%d"),device);
  \\ would have thought this was an error condition
    return;
  } 
  #ifdef DO_NOT_ALLOW_STANDALONE_SWITCHES
  if(device > devices_present))  {
    device = 1;
  }
  #endif
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
None
**Additional context**  
_Add any other context or screenshots about the feature request here._
The above Tests are done in a number of other places in the code base.
I don't use MQTT , so I have not made the mods anywhere else.
The result is Very Satisfying and Super Simple !!.
Using KNX & the extra input on Sonoff Mini's , I have  an All Off/On switch for the room . 
**(Please, remember to close the issue when the problem has been addressed)**

BBBits, 2020-01-30:
My Workaround for this problem is not correct.
a better solution would be to test if a switch or button has a corresponding relay and 
return from ExecuteCommandPower if it doesn't.

to-scho, 2020-01-30:
Have you tried "SwitchTopic 0" as described here: https://github.com/arendst/Tasmota/wiki/Buttons-and-switches

I am using switches just to trigger rules. No relay is toggled by 4 switches on a device having 1 relay.

ascillato2, 2020-01-30:
Closing this issue has been answered and as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-01-30 by ascillato2

BBBits, 2020-01-30:
A Tad hasty on closing My Issue ..

As I Mentioned , I am NOT using MQTT.
in support__switch.ino ::  void SwitchHandler(uint8_t mode)

we have this code
        if (!SendKey(KEY_SWITCH, i +1, switchflag)) {  // Execute command via MQTT
              ExecuteCommandPower(i +1, switchflag, SRC_SWITCH);  // Execute command internally (if i < devices_present) 
        }
The sendkey will likely return true if you are using MQTT , therefore ExecuteCommandPower won't be called.

BBBits, 2020-01-30:
Just to continue with My issue.
IMAO you are writing rules to overcome a problem that shouldn't be there in the first place.
The Default behavior of a switch that is not associated to a relay should be nothing.
If you want it to do something , write a rule.
currently it is back to front .

ascillato, 2020-01-30:
Sorry, rules are independent from mqtt. You don't need MQTT for using rules. Rules are a feature of Tasmota to config any behaviour you want.

The default of Tasmota is that buttons and switches are linked to relays.

If you need another behavior and unlink them, you just need to tell Tasmota what you want using rules. There is no need to modify the firmware. You need just to config Tasmota. That is why the support chat is aimed for that type of help and github issues are for bug reporting.

Please, search in the docs on how to use rules, search in issues for other examples (this issue has been addressed several times before) and find us in the support chat for a more interactive help.

ascillato, 2020-01-30:
> The Default behavior of a switch that is not associated to a relay should be nothing.

The default is the light stair case. If you need another behaviour, just config tasmota using rules

BBBits, 2020-01-30:
I get that. 
I am trying to make a point that the logic with choosing a function is wrong.
it should have no function

ascillato, 2020-01-30:
> it should have no function

That is an opinion. For example, Tasmota by default starts as it is a sonoff basic module (design decision). And for example ESPEasy starts as you said, no functions and no modules associated (design decision), But in both approaches, you are allowed to modify behaviours. Nothing wrong with those approaches. Just a different starting point.

BBBits, 2020-01-30:
so lets say that 1 out of 100 users use the  2way switch idea, the rest turn it off 
does that imply something needs to change ?
as far as using a 3v3 input of an ESP to traverse though the walls with other mains wiring is not something I would do anyway.

ascillato, 2020-01-30:
Thanks for your point of view, but due to backwards compatibility with other users setup, this is not going to be modified so as to not brake any actual setup. Tasmota has been downloaded more than 5 million times, so if it is not really necesary, we don't want to make breaking changes. Again, if you don't want the default, you can configure Tasmota to behave differently using rules, scripts, setoptions, timers, etc etc. Please check that this discussion has been done before in old issues. Thanks.

BBBits, 2020-01-30:
If this has been thrashed out before & the issue is backward compatibility , then clearly I'm not alone.
a Simple ifdef build option could tidy the issue up for ever.

ascillato, 2020-01-30:
You had another issue that is the same of this one #7590. Have you tried the chat?

If not, do you want any help on configuring Tasmota?

ascillato, 2020-01-30:
In your main issue you said:

It will always control relay1 and do the rules or KNX events.

Sorry, That is not correct. When you definde rules for all states of the switch/button, it will not control the relay.

If you do the following:

```yaml
switchtopic 0
switchmode2 1
rule1 1
rule1 on switch2#state=0 do delay endon on switch2#state=1 do KnxTx_Cmnd1 1 endon
```

when the switch is in OFF position, your device will do nothing
when the switch is moved to the ON position, your device will execute command `KnxTx_Cmnd1 1`
No relay is going to be toggled. No mqtt needed.

BBBits, 2020-01-30:
yes , got shut down with a chap saying I wouldn't understand stuff.
had to leave the chat in frustration.
been messing with stuff for over 40yrs .
guess i am just outdated in my thinking .
-------------------
I see your edit regarding method.
although you say No MQTT needed , I'd be  guessing that switchtopic 0 is actually MQTT ?
then the sendkey returns true and executecommandpower  isn't called

ascillato, 2020-01-30:
the config `switchtopic 0` is the default for `switchtopic` but I added in the list of commands as I did for `rule1 1` in order to set all the required commands that will make the rules to execute.

`Switchtopic` needs to be 0 in order to let rules to execute. It was just that. No MQTT needed for rules to execute.

ascillato, 2020-01-30:
Please, test the commands I have posted.

BBBits, 2020-01-30:
I'll do some tests & figure out what actually lets sendkey return true.

many thanks for the feedback

ascillato, 2020-01-30:
Sendkey function is not used in the rules routine. For rules code, see the rules driver instead _( xdrv_10_rules.ino )_.

But please try the commands I have posted. They work as you need.

BBBits, 2020-01-30:
I'm referring to the place where my problem actually lies.
In the SwitchHandler routine.
        if (!SendKey(KEY_SWITCH, i +1, switchflag)) {  // Execute command via MQTT
              ExecuteCommandPower(i +1, switchflag, SRC_SWITCH);  // Execute command internally (if i < devices_present) 
        }

ascillato, 2020-01-30:
rules routine overrides that.

Please, try the default Tasmota codebase with the commands I posted.

BBBits, 2020-01-30:
ok , I'll give it  a go  .. ( he said begrudgingly )
thanks

BBBits, 2020-01-31:
as expected , using rules etc gets it working the way I want .
The elefant is still in the corner of the room tho .
Perhaps in the future there might be an extension to the config page that has things like that switchable via a checkbox.
In the meantime , I'll go and sit with the elefant and keep my mouth shut .
Thanks for your patience

ascillato, 2020-01-31:
> Perhaps in the future there might be an extension to the config page that has things like that switchable via a checkbox.

Today is your lucky day. 

Check the docs for TasUI

ascillato, 2020-01-31:
Direct link https://github.com/IOTReboot/TasUI

ascillato, 2020-01-31:
If you are using Tasmota 8 then you must enable CORS under wifi settings adding
http://tasui.shantur.com as domain.

Then go to http://tasui.shantur.com and that UI is going to be downloaded to your browser and then you can control and config your Tasmotas from there. The old UI is going to be there too if you need.

BBBits, 2020-01-31:
Thanks for the Heads UP
Will definitely give it go !
I did say I would be quite .... Dang
I presume it would be writing rules while the elefant & I , grind our ever  diminishing teeth .

BBBits, 2020-01-31:
I'm Back.
In the interest of correctness , I should mention that the Rules that I used to stop the switch from controlling relay1 Does Influence the return Value of SendKey .
It also produces Incorrect Log  entries .
{"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":444,"Rules":"on switch2#state=1 do delay endon on switch2#state=0 do delay endon"}
Added a little log entry
`      if (switchflag <= POWER_TOGGLE) {
        if (!SendKey(KEY_SWITCH, i +1, switchflag)) {  // Execute command via MQTT
              ExecuteCommandPower(i +1, switchflag, SRC_SWITCH);  // Execute command internally (if i < devices_present) 
        }else{
          AddLog_P2(LOG_LEVEL_INFO, PSTR("BAZ: SendKey returned True"));
        }
      }`
Log entry's aftter hitting switch2 
10:56:54 RUL: SWITCH2#STATE=1 performs "delay"
10:56:54 RSL: RESULT = {"Delay":0}
10:56:54 KNX: Button 2 = 1 sent to 3.3.1
10:56:54 BAZ: SendKey returned True
10:56:55 RUL: SWITCH2#STATE=0 performs "delay"
10:56:55 RSL: RESULT = {"Delay":0}
10:56:55 KNX: Button 2 = 0 sent to 3.3.1
10:56:55 BAZ: SendKey returned True
10:56:56 RSL: RESULT = {"POWER":"ON"}
10:56:56 RSL: POWER = ON
10:56:57 RSL: RESULT = {"POWER":"OFF"}
10:56:57 RSL: POWER = OFF

The POWER  Entries are Bogus