arendst/Tasmota #7625: Support for mystrom Wifi buttons

ouinouin, 2020-01-30:
**Have you looked for this feature in other issues and in the docs?**  
yes mystom buttons doesn seem to be supported in any esp8266 projects
**Is your feature request related to a problem? Please describe.**  
adding support for the great mystrom button and button + wich are the only commercial esp buttons battery powered. they actually send http payload only, but have support plugins for many home automation platform + nodered.
https://mystrom.ch/wifi-button/

**Describe the solution you'd like**  
the easyest button to be supported is the simple one (mystrom button) , but the mystrom button + is equipped with a touch sensitive surface and some gestures can be performed on it to add functionalities.
I have many of those and can give or buy some to devs for device exploration.

**Describe alternatives you've considered**  
N/A

**Additional context**  
mystrom button : [button](https://mystrom.ch/wifi-button/)
mystrom button+ : [button+](https://mystrom.ch/wifi-button-plus/)
released sourcecode : [https://github.com/myStrom/mystrom-button](https://github.com/myStrom/mystrom-button)

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-01-30:
Hi,

Tasmota can be flashed on those devices. You just need to find out which GPIOs of the ESP8266 is connected to and to configure those as a template using Tasmota webUI and should work. For more information on templates, please check the docs and if you need help, please address that to the Tasmota support chat.

Anyway, as this is a battery-powered device, it requires a battery optimized configuration in order to extend the battery charge. Tasmota uses much more energy than the default firmware because it has much more features. So you will need to test if deepsleep feature in Tasmota (check related article in the docs) is suitable for you.

closed 2020-01-30 by ascillato2

ouinouin, 2020-02-06:
Hi, 
thanks for reply I got further on this small board, I managed to flash tasmota, but there are some very strange behaviors, while I power the board trhough the 3.3 volts pins, it boots most of the time but uptime never exceed 10 to 15 minutes.
when i power it through battery, it boots and reboots randomly, sometimes touching the back of the board makes it booting.
I suspect some instabilities due to the status of the pin but i dont have a methodology to troubleshoot that (but I have a logic analyzer available.
so basicly, I Have the pinout for usb to serial connection.
I have some pictures of the boards
I have partially reversed the schematic and the components of the boards (buttton, battery charger etc etc.)
I d like to share this , is there a place where I can contribute in the tasmota project (i first thought about the template pages, but for now i d like to post pictures and have advices to gues the function of some components on the board.
Thanks.

elnkosc, 2020-06-06:
I have been playing around with tasmota on this button as well. I manage to control the LEDs but I cannot get the button to work (if I look at the original firmware it seems to be on GPIO4. When pushing the button, the device most of the time hangs. I'm trying to get an MQTT update when the button is pressed. NO luck so far....