arendst/Tasmota #7627: Temperature Display and Console missing for Sonoff TH(4) with DS18x20(4)

amouses, 2020-01-30:
### PROBLEM DESCRIPTION
I configure a DS18x20 Temperature sensor, but nothing is displayed on the Tasmota TH Module Main Menu or shown as reported in the log

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [yes ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ yes] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _TH16____
- [ yes] Tasmota binary firmware version number used: _8.1.0 UK language ____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
 No rules?

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

00:00:29 CMD: status 0
00:00:29 SRC: WebConsole from 192.168.1.11
00:00:29 CMD: Group 0, Index 1, Command "STATUS", Data "0"
00:00:29 MQT: stat/sonoff/STATUS = {"Status":{"Module":4,"FriendlyName":["Tasmota"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:00:29 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:30","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":53,"SaveCount":118,"SaveAddress":"F6000"}}
00:00:29 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":7,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"330/699"}}
00:00:29 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":2,"SysLog":0,"LogHost":"192.168.1.199","LogPort":514,"SSId":["byre",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
00:00:29 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1420C2","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
00:00:29 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"immersion","IPAddress":"192.168.1.72","Gateway":"192.168.1.99","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.222","Mac":"80:7D:3A:31:55:DD","Webserver":2,"WifiConfig":4}}
00:00:29 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.207","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3155DD","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
00:00:29 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:29 1970","Local":"Thu Jan 01 00:00:29 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:03"}}
00:00:29 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:29"}}
00:00:29 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:29","Uptime":"0T00:00:30","UptimeSec":30,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"byre","BSSId":"B8:69:F4:B5:14:8C","Channel":8,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:06"}}}
00:00:45 WIF: Checking connection...
00:00:45 WIF: Connected
00:01:05 WIF: Checking connection...
00:01:05 WIF: Connected

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Connect   a DS18b20 temperature probe to the TH16 and configure GPIO14 as DS18x20
There should be Temp information sent in the console log.  And that temp should be displayed on main menu.

Actually I assume that even without any temp sensor connected there should be a temp display notice on main screen.  (There is not)

### EXPECTED BEHAVIOUR
When I specify module type TH (4) and then on GPIO14 select DS18x20(4) I expect to see the Termperature display on the top of the main screen.   Just like it is for the 2 other termp sensors i.e. AM2301 and SI7021

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
I added screenshots

[image]

[image]

[image]

[image]

[image]

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-01-30:
Hi,

Have you checked old issues? This troubleshooting has been addressed several times before. Have you tried the solutions proposed in old issues? Have you tried the support chat?

amouses, 2020-01-30:
> Hi,
> 
> Have you checked old issues? This troubleshooting has been addressed several times before. Have you tried the solutions proposed in old issues? Have you tried the support chat?

I have indeed looked at the old issues that seem to have specified the same problem.   One of them was not using a TH16

Another person had the same issue, then he said 'pluggin it in ' completely helped.  This does not explain why on the menu there is no temperature to be displayed as in the case of the other two sensors.

I have not tried the support chat but will do so presently.

amouses, 2020-01-30:
This report looked promising.  But the user did not explain why there was no temp display when the DS18x20 is not plugged in.  In the cases as shown above for the other 2 sensors there most definitely is a display of the Temperature on the man page
https://github.com/arendst/Tasmota/issues/6139

Okay I found the Chat link here
https://discord.gg/Ks2Kzd4

Noob comment.  This is a discord channel.

amouses, 2020-01-30:
Obviously I have tried the "recommended solution !" which is to PRESS HARD the plug into the socket.   My problem with this idea  (yes I tried) is that this does not explain why the TEMP is not displayed on menu or in log.  Nobody seems to have answered that question.

ALSO: I've just ordered another TH16 and DS18B20 sensor and I will check both my current sensor and new sensor arriving with the TH16 in EWElink.  This would establish that the sensors themselves are without error.  (but still does not answer question why nothing shown in console or on menu).

ascillato, 2020-01-30:
If you want more debug info use command `weblog 4` in the console.

ascillato, 2020-01-30:
Have you tried the firmware type **Tasmota_sensors.bin** ?

ascillato, 2020-01-30:
Please, try http://thehackbox.org/tasmota/tasmota-sensors.bin Thanks

amouses, 2020-01-30:
There are comments on the chat about needing a pullup resistor but I understand that this was thought to be unecessary  https://github.com/arendst/Tasmota/issues/2756   (says not necessary to install pullup) and it is definitely NOT in the tasmota guide for TH10/TH16

ascillato, 2020-01-30:
https://tasmota.github.io/docs/#/peripherals/DS18x20

amouses, 2020-01-30:
I tried  http://thehackbox.org/tasmota/tasmota-sensors.bin   which at time of installing is 8.1.0.5 tasmota and **it does not improve matters.**   

I am purchasing an additional sensor just in case the one I am using is dead.

However can anybody explain, as per original pictures why when you configure the other 2 temp sensors they show up on the main menu, EVEN IF not connected.  and the DS18x20 when selected shows nothing.   Can somebody confirm this is working as designed?

ascillato, 2020-01-30:
Yes, it is like that.

Have you find the pull up resistor? Can you tried the pull up as the docs article?

amouses, 2020-01-30:
Okay thanks for the confirmation that for some reason this temp sensor (unlike the others) is not displayed on the menu.

No I've not tried putting a pullup resistor.   It seems unbelievable that nobody has definitively mentioned this in the guide?    And why don't the other sensors need  a pullup?  In the chat the pullup is a generic recommendation for an ESP8266, and since it's not easy for me to try and solder one in, no I have not done that yet.

If anybody who has a DS18x20 can confirm that they needed to add the resistor then of course I would have to add one.   But seems strange that nobody with a Sonoff TH16 or TH10 has mentioned this right?

My next debug step is to wait 2 days for the 2nd temp sensor to arrive, test it with a second TH16 with ewelink , this will prove this probe works fine.   Then using a proven good probe I'll plug that/ those into the tasmota 8.1.0.5 flashed TH16 and report back.

ascillato, 2020-01-31:
For this sensor you need the pull up resistor. If not it don't work.

If your sensor is the sensor alone, you have to add that resistor. If your sensor comes with the wire and the plug, it is most likely that it has the resistor already.

So, if you have a multimeter, it would be great if you can measure it to see if it is there and not broken. Use the diagram of the docs. Link in previous comments.

bkmit, 2020-01-31:
You can try to use internal pullup by "SetOption74 1" too

amouses, 2020-01-31:
First everybody, thanks so much for this great support.   I'm learning a lot.

As bkmit said code was changed to allow the internal pullup to be used
https://github.com/arendst/Tasmota/pull/6795/commits/393f56827696297da734b01ffd37439c3306467c

However I tried it and this did not work

----

Next I measured the resistance between GPIO14 and the 3.3V rail on the DS18x20 plug.  According to my diagram this is between the tip of the plug and the base of the plug. **There is NO 4K7 resistor in there.**

----

Next step is to wait for the new sensor arrive,  (in 24 hours),  test it using the also arriving unflashed TH16, when I know it is good under ewelink, then I will plug into the existing and flashed TH16,  and then the dance can continue.

amouses, 2020-02-01:
IT IS ALL WORKING.   He is the debrief and learning points

Today I received a new TH16 sensor and a second  DS18B20

With the new sensor plugged into existing TH16 running Tasmota 8.1.0.5 (sensors bin version) the temperature is displayed on the main menu

[image]

The temperature is also of course sent as a message and we can see this is the console log

[image]

Other points to note

- It IS NOT necessary to change any hardware on the TH16.  In particular no resistor needs to be added to the unit

- I checked the resistance values of the new and working temp sensor, they DONT have any 4K7 or equivalent resistor between vcc 3.3V and GPIO14 pins.   So to me the pulup resistance MUST be inside the TH16

- You will see from the console log picture above that changing SetOption74 from 0 to 1 has no effect.  (I rebooted twice between changes).   The temperature is always displayed regardless of SetOption74 setting  when a working probe is plugged in.

- If the Config module for GPIO14 has the DS18x20(4) set , then if there is no working sensor plugged in, there is not Temperature and Null value displayed.  This is unlike the 2 other temperature sensor settings.   For them if you set GPIO14 to them and no equivalent sensor is plugged in the Main Menu displays a temperature heading with NULL value

**- So the issue was faulty temp sensor,  combined with an unclear understanding of TH16 sensor for me and others, now fixed!**

--

ascillato, 2020-02-01:
Great to know that now it is working for you, please close this issue.

closed 2020-02-01 by ascillato2