arendst/Tasmota #7633: Output of the SDS0X1 disappears and  reappears (Tasmota 8.1.0)

Palerm0, 2020-01-31:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have an ESP8266 with Tasmota installd on it (Version Tasmota 8.1.0)
Connected to it i have the SDS0X1 sensor. It works i believe, But on the menu the complete sensor disappears for a second and then reappears.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [V ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [V ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [V ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [V ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [V ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Sonoff-Sensor.bin
- [ ] Tasmota binary firmware version number used: _____
  - [V ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [V] Flashing tools used: NodeMCU-PyFlasher-4.0-x64
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
12:46:21 MQT: sonoff/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
12:46:21 MQT: sonoff/stat/RESULT = {"Module":{"18":"Generic"}}
12:46:21 MQT: sonoff/stat/RESULT = {"GPIO0":{"60":"MHZ Tx"},"GPIO1":{"101":"SDS0X1 Tx"},"GPIO2":{"5":"I2C SCL"},"GPIO3":{"70":"SDS0X1 Rx"},"GPIO4":{"61":"MHZ Rx"},"GPIO5":{"6":"I2C SDA"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
  Configuration output here:
The only adittional configuration are:
SetOption19 1
TelePeriod 30
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
12:49:04 MQT: sonoff/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["WeerstationCloudNL"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:49:04 MQT: sonoff/stat/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-NL.bin","RestartReason":"External System","Uptime":"0T00:17:35","StartupUTC":"2020-01-31T11:31:29","Sleep":50,"CfgHolder":4617,"BootCount":19,"SaveCount":38,"SaveAddress":"F9000"}}
12:49:04 MQT: sonoff/stat/STATUS2 = {"StatusFWR":{"Version":"8.1.0(sensors)","BuildDateTime":"2019-12-25T12:38:43","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"364/699"}}
12:49:04 MQT: sonoff/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Legendary",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005AFF000000000000","00000200","00000000"]}}
12:49:04 MQT: sonoff/stat/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FDAE387","0437A3A4","B7FFBFCD","01DA9BC0","0016FCC7"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,18,19,20,21,22,24,26,29,30","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,34,37,39,40,42,43,51,52,55,56,58"}}
12:49:04 MQT: sonoff/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff-7129","IPAddress":"192.168.25.21","Gateway":"192.168.25.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.224","Mac":"2C:3A:E8:39:1B:D9","Webserver":2,"WifiConfig":4}}
12:49:04 MQT: sonoff/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.100.231","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_391BD9","MqttUser":"a131774","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
12:49:04 MQT: sonoff/stat/STATUS7 = {"StatusTIM":{"UTC":"Fri Jan 31 11:49:04 2020","Local":"Fri Jan 31 12:49:04 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:22","Sunset":"17:45"}}
12:49:04 MQT: sonoff/stat/STATUS10 = {"StatusSNS":{"Time":"2020-01-31T12:49:04","BME280":{"Temperature":22.9,"Humidity":40.9,"Pressure":1003.5},"MHZ19B":{"Model":"B","CarbonDioxide":812,"Temperature":30.0},"TSL2561":{"Illuminance":30.170},"SDS0X1":{"PM2.5":9.2,"PM10":1.6},"PressureUnit":"hPa","TempUnit":"C"}}
12:49:04 MQT: sonoff/stat/STATUS11 = {"StatusSTS":{"Time":"2020-01-31T12:49:04","Uptime":"0T00:17:35","UptimeSec":1055,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":34,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Legendary","BSSId":"3C:0E:23:C8:8E:40","Channel":1,"RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:05"}}}
12:49:12 MQT: sonoff/tele/STATE = {"Time":"2020-01-31T12:49:12","Uptime":"0T00:17:43","UptimeSec":1063,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":51,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Legendary","BSSId":"3C:0E:23:C8:8E:40","Channel":1,"RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:05"}}
12:49:12 MQT: sonoff/tele/SENSOR = {"Time":"2020-01-31T12:49:12","BME280":{"Temperature":22.9,"Humidity":40.8,"Pressure":1003.5},"MHZ19B":{"Model":"B","CarbonDioxide":812,"Temperature":30.0},"TSL2561":{"Illuminance":30.160},"SDS0X1":{"PM2.5":9.8,"PM10":2.8},"PressureUnit":"hPa","TempUnit":"C"}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
No need to

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
Have an video:
[IMG_0779.zip](https://github.com/arendst/Tasmota/files/4139131/IMG_0779.zip)

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

If you need more information, please let me know

ascillato, 2020-01-31:
There is an specific time where this show and hide occurs. It happens all the time with the same interval or it is random interval?

Have you checked if you don't have any loose wire or cold soldered contact?

Jason2866, 2020-01-31:
How is all the stuff powered? Which device do you use?

Palerm0, 2020-01-31:
@ascillato 
I use a stopwatch o time:
1: 27 Sec
2: 30 Sec
3: 1.0 Min
4: 30 Sec
5: 30 Sec
6: 1.0 Min
Interesting numbers here. :-)
Quadruple check the wires. They all sturdy connected to the board.

@Jason2866 
This should not be a problem according many (ask on some chats). I`m using the ESP8266mod (nodeMCU v3)
It has one 5v output, Here is the MHZ19A and the SDS0X1 connected to. Checkt the joints already.
If orderd a few new ESP8266`s, so I cannot test it on a other board.  
Could it be a power problem? I hope not.

Jason2866, 2020-01-31:
Which 5V output from the NodeMCU do you use?
AND what power supply do you have for delivering the 5 Volts for the NodeMCU?
The power supply can be your problem!

ascillato, 2020-01-31:
Please, as Jason has stated, try powering your sensors with an extra power source to test if this is the source of your issue. Remember to connect the gnd of the extra source with the one of the nodemcu. Thanks.

Palerm0, 2020-01-31:
There is a (VU) port witch supplys 5v directely from the USB port. 
I need to look for a stronger USB power supply. 
BRB
@ascillato Indeed. All the 5v sensors (In this case only 2) are grounded on one of the GND ports
 
[image]

Palerm0, 2020-01-31:
Im using @fist a power suply of 5v output 1000mA
Now i have an Apple 10W 5.1V 2.1A

still having the same problem

sfromis, 2020-01-31:
You could also debug by running with only the PM sensor connected. If this works, it might be power regulation issues, or just devices not living peacefully on the bus.

Palerm0, 2020-01-31:
Removed all the other sensors. 
Connected the PM sensor. But still the readings disappear. I used the 5v power source of a ESP32 witch I had laying around. So I connected the + and GRN on the ESP32. And the TX & RX on the ESP8266, Still same problem. The power consumption is "Maximum Current: 100mA" under load.
Is there a debug to be set on High!?

ascillato2, 2020-02-01:
Please, post a photo of your setup. Thanks.

Palerm0, 2020-02-02:
For the SDS0X1:
Rated current: 70mA
and for the MHZ19A
Average current < 60mA (@5V)
Peak current 150mA (@5V)
That is in total 310 ma give or take a few,
I have a power supply with 2A.  That should be enough.

Here are some pictures of the setup. IF you need more please let me know. 
 
[image]
[image]

Jason2866, 2020-02-02:
Do you see at startup in the log something `...uses hardware serial`
If not please change your setup for gpio1 and 3

Palerm0, 2020-02-02:
Changes the gpio TX->D3 and RX->D4
Still jumping away. But also I dont see that look like 
> ...uses hardware serial
...

Jason2866, 2020-02-02:
You have to use gpio1 and gpio3 for hardware serial.
You need to swap tx and rx assigning to the gpios. Doing this activates hardware serial
Wiring needs to be swaped!

TrDA-hab, 2020-02-04:
@Palerm0
Perhaps my scheme will help you ?!
https://github.com/TrDA-hab/blinds/blob/master/images/Multi-sensor/Multi-sensor-1.jpg

... it worked fine ... until the kids broke (((

Palerm0, 2020-02-05:
@Jason2866 
It took me some time, sorry,
I Changed everything back and swap the wires. Unfortunately no cigar.

@TrDA-hab 
That is a nice setup. So you have extra power and the USB connected!?
What i dont understand tha the red wire on the extra DC power 5V is connected to the VUSB.
That VUSB is also giving 5V. I need to order some parts. This will take a while. Il close the ticket and when I receive all the parts and set it up i hope the problem is solved. 

Thanks for all you replies, it gave me a more understanding of how this works.  
Greetings
palermo

Palerm0, 2020-02-05:
@TrDA-hab
... everything worked fine ... until the children destroyed it (((

Are you gonna rebuild it? such a nice setup.

closed 2020-02-05 by Palerm0