arendst/Tasmota #7635: Sonoff Dual R2 Shutter - relay2 ramdom powered position remains 100 but NO ghost switch

arko36, 2020-01-31:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Hi,
I'm using switches instead of buttons and I did a mod on the board to use GPIO9 and GPIO10 for the switches to avoid entering to prog mode if some of the switches is left to GND while a power cut.
The circuit mode I'm using is relay1 off/up relay2 off/down with the interlock commands applied from the wiki.

The problem is that I do not have any log (syslog or console) showing a ghost switch but the relay2 gets randomly powered with two consecutive pulses doing a "tac-tac" moving the blinds down a little. However, the position remains stored as 100. LEt's say after a couple of hours the blind are half way, I can by any means close them but they will not open fully again unless I close them completely, wait for relay1 to stop and then open completely.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Sonoff Duar R2
- [ ] Tasmota binary firmware version number used: 6.7.1(sonoff)
  - [ ] Pre-compiled
  - [ x] Self-compiled
    - [ ] IDE / Compiler used: Arduino
- [x ] Flashing tools used: Arduino
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
13:23:29 RSL: RESULT = {"NAME":"ARKO Shutter","GPIO":[255,255,0,255,0,22,10,9,21,56,0,0,0],"FLAG":0,"BASE":39}
13:23:30 RSL: RESULT = {"Module":{"0":"ARKO Shutter"}}
13:23:30 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"22":"Relay2"},"GPIO9":{"10":"Switch2"},"GPIO10":{"9":"Switch1"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
13:24:19 RSL: RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
13:24:19 RSL: RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
13:24:20 RSL: RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
13:24:40 RSL: STATUS = {"Status":{"Module":0,"FriendlyName":["room1-shutter","room1-persiana"],"Topic":"room1-shutter","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[2,2,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:24:40 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Hardware Watchdog","Uptime":"0T12:27:29","StartupUTC":"2020-01-30T23:57:11","Sleep":0,"CfgHolder":4617,"BootCount":144,"SaveCount":349,"SaveAddress":"F6000"}}
13:24:40 RSL: STATUS2 = {"StatusFWR":{"Version":"6.7.1(sonoff)","BuildDateTime":"2019-11-19T09:50:15","Boot":7,"Core":"2_5_2","SDK":"2.2.1(cfd48f3)"}}
13:24:40 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":4,"LogHost":"192.168.1.48","LogPort":514,"SSId":["My_Router","My_House"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000E001","2805C8000100060000005A00000000000000","40000000"]}}
13:24:40 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":600,"Free":400,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","003683A1","22B617CD","01001BC0","0000F181"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,30","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
13:24:40 RSL: STATUS5 = {"StatusNET":{"Hostname":"room1-shutter","IPAddress":"192.168.1.102","Gateway":"192.168.1.2","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"DC:4F:22:9D:9C:F9","Webserver":2,"WifiConfig":2}}
13:24:40 RSL: STATUS7 = {"StatusTIM":{"UTC":"Fri Jan 31 12:24:40 2020","Local":"Fri Jan 31 13:24:40 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:22","Sunset":"17:45"}}
13:24:40 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-01-31T13:24:40","Switch1":"ON","Switch2":"OFF","Shutter1":{"Position":100,"direction":0}}}
13:24:40 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-01-31T13:24:40","Uptime":"0T12:27:29","UptimeSec":44849,"Heap":25,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"My_Router","BSSId":"00:18:F8:46:43:77","Channel":5,"RSSI":72,"LinkCount":69,"Downtime":"0T00:03:56"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
as I explained, when this happens no log, nothing, web or syslog in '4'

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
I have the device working on a table for weeks before installing it, as this clicking happens randomly and could take hours I didn't noticed. Now connected I can see the blinds half way when I return home and sometimes I heard the motor tac-tac closing it a little by little.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expect the blind to remain fully open when position shows 100.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
N/A

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
The tasmota went to factory defaul long time ago by itself, happened only once, I configure the wifi and uploaded the backup config and up again.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-01-31:
If there is no log, seems to be a hardware issue. May be the power regulator is failing to keep the relay coils powered on the 5v rail.

There isn't much we can do from software side.

arko36, 2020-01-31:
Hi,

Thanks. But the relays are OFF once the time shutteropenduration. Sounds like the oposite but only on relay2. could be the interlock control ? is always 2 short pulses after long periods of time randomly . However, but position remains always 100.
Can we discard completely discard a software problem on relay2 ?
I will buy an flash another one to check if I have same problem.

Thanks
ARKO

ascillato, 2020-01-31:
> could be the interlock control ?

If so, you would have a log. Any operation performed by the software on relays has an output on the console log. In my tests I never saw that behavior.

So, seems indeed that it is hardware and nlt software. Sorry.

Jason2866, 2020-01-31:
You use a very very old outdated version (v.6.7.1). Update to latest Tasmota version

arko36, 2020-01-31:
hi,
Thanks for the tips, I will change the module and also the version. let's see. Any chance to get the schematic of the Dual R2 ? Thanks again.

ascillato2, 2020-02-01:
> I will change the module and also the version.

Great. Please, tell us if that solves this problem. Thanks.

closed 2020-02-01 by ascillato2