arendst/Tasmota #7645: Setoption13 not reset to 0 when device can't connect to wifi

SylvainGa, 2020-02-01:
### PROBLEM DESCRIPTION
On tasmota 8.1.0, when my device loses wifi (wificonfig 5) and with setoption13 set to 1 and setption1 to 0 (not 1), it never resets to 0, no matter long I hold the button pressed. If the device is wifi connected, pressing it for 40+ seconds (setoption32 is at 40), it does reset setoption13 to 0. I want to set it back to 0 so I can press the button 6 times to reboot the device so it can reconnect to the wifi.

Yes, I know wificonfig 2 will do that but I don't want all my devices rebooting if for some reason I have to reboot my router! Anybody else seeing this? Not setting option13 to 1 is annoying for devices where a single button press is enough so hopefully there is a better option that keeping it at 0!

Looks like the device restarts itself before the counter reaches its threshold so it never manage to setoption13 to 0. That 4xsetoption32 (which would make it 16 seconds instead of 40) ain't happening either since it took over 40 seconds before the wifi connected device would toggle setoption13 (as seeing from the console screen). Looking at the LED on the device, it looks like it reboots itself every 8 seconds.

The only option left is to power cycle the devices through a breaker off/on. Once done, doubled checked setoption1 and 13 and they were set to 0 and 1 respectively so it should have worked. Rules 1 to 3 are also off so it's not a rule kicking in at button press.

I can't reproduce on my NodeMCU when I set a bad SSID name or bad SSID password. Holding the button for 40+ seconds does set option13 to 0 (reading from serial port) . Don't know what the wifi status of those unconnected switches are, but there is something not working. 

Since I can't reproduce, device becomes unresponsive and it happens randomly, it's impossible for me to be at the latest development binaries unfortunately but I am at a very recent release, 8.1.0.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Generic 18
- [X] Tasmota binary firmware version number used: 8.1.0
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: Tasmota firmware upgrade
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
LumiereCaveFluorescent/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
09:29:06 MQT: LumiereCaveFluorescent/stat/RESULT = {"Module":{"18":"Generic"}}
09:29:06 MQT: LumiereCaveFluorescent/stat/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"52":"Led1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
09:30:27 MQT: LumiereCaveFluorescent/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["LumiereCaveFluorescent"],"Topic":"LumiereCaveFluorescent","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:30:27 MQT: LumiereCaveFluorescent/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:24:40","StartupUTC":"2020-02-01T14:05:47","Sleep":50,"CfgHolder":4617,"BootCount":11,"SaveCount":180,"SaveAddress":"FA000"}}
09:30:27 MQT: LumiereCaveFluorescent/stat/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"360/699"}}
09:30:27 MQT: LumiereCaveFluorescent/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":1,"LogHost":"192.168.2.84","LogPort":514,"SSId":["BELL776","Test"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["400AA009","2805C8000100060000005A00000000000000","00000200","00000000"]}}
09:30:27 MQT: LumiereCaveFluorescent/stat/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":0,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"!1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
09:30:27 MQT: LumiereCaveFluorescent/stat/STATUS5 = {"StatusNET":{"Hostname":"LumiereCaveFluorescent-6643","IPAddress":"192.168.2.178","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"B4:E6:2D:18:39:F3","Webserver":2,"WifiConfig":2}}
09:30:27 MQT: LumiereCaveFluorescent/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.84","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1839F3","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
09:30:27 MQT: LumiereCaveFluorescent/stat/STATUS7 = {"StatusTIM":{"UTC":"Sat Feb 01 14:30:27 2020","Local":"Sat Feb 01 09:30:27 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Nov 01 02:00:00 2020","Timezone":99,"Sunrise":"02:21","Sunset":"11:46"}}
09:30:27 MQT: LumiereCaveFluorescent/stat/STATUS10 = {"StatusSNS":{"Time":"2020-02-01T09:30:27"}}
09:30:27 MQT: LumiereCaveFluorescent/stat/STATUS11 = {"StatusSTS":{"Time":"2020-02-01T09:30:27","Uptime":"0T00:24:40","UptimeSec":1480,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"BELL776","BSSId":"A8:9A:93:FD:CC:2E","Channel":11,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Unable to reproduce using my NodeMCU running 8.1.0 or 8.1.0.5

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Holding the button press for 10x setoption32 should reset setoption13 to 0. This is not happening

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-02-01:
From your status 0, your device has "WifiConfig":2, so that restarting behavior is expected with your actual config.

Please, configure it as wificonfig 4, that is the default, and try again. Thanks

SylvainGa, 2020-02-01:
You're seeing wificonfig 2 because I had rebooted the device and set it to
2 already. It was at 5 when it was misbehaving.

ascillato, 2020-02-01:
As you said, I can't reproduce it on my devices neither, so seems that your misbehaving device had another config like wificonfig 2 or wificonfig 0

You can try the following:

+ do a reset 5 to set everything to default and then config it again 

Then try again.

SylvainGa, 2020-02-01:
Before setting wificonfig to 2, I did check what value it was at as well as setoption1 and setoption13. If it was set at wificonfig 2, I would have seen an AP show up but none did (I checked for that). I usually do a reset 5 right after tuya converting the device but who knows, I might have forgotten. This device behaved like that twice since last week. I'll reset it and reprogrammed it once I'm back home.

ascillato, 2020-02-01:
Great, after reset 5, please update to latest Tasmota version too. Thanks.

ascillato2, 2020-02-03:
Please, when you have time, tell us if this solves your issue. Thanks.

closed 2020-02-03 by ascillato2