arendst/Tasmota #7646: TuyaMCU support for multiple enum dpIds

dgwynne, 2020-02-02:
**Have you looked for this feature in other issues and in the docs?** 

It is similar to #7586, but is less specific

**Is your feature request related to a problem? Please describe.**  

I have a couple of [Anko HEGSM40](https://www.kmart.com.au/wcsstore/Kmart/pdfs/42779896_Manual.pdf) wifi enabled fans. They have a lot of controls which I would like to expose over MQTT without having to write a lot of boilerplate Rules for. The specific controls are:

- Power

Turning the motor on and off is controlled via Enums on DpId 1. Value 0 is off, 1 is on. This is supported by default and is visible as POWER1.

- Oscillation

Turning oscillation on and off is controlled via Enums on DpId 4. Value 0 turns oscillation off, 1 turns it on. This is already supported by using `TuyaMCU 12,4` to expose it as POWER2.

- Fan speed

The fan has 8 speeds exposed as an Enum on DpId 3 with values ranging from 0 to 7. Maybe this could be handled as a dimmer with `TuyaMCU 21,3`?

- Mode

There are 3 operating modes exposed as an Enum on DpId 2. The "Normal" fan mode is value 0, "Nature" mode is 1, and the "Sleep" mode is 2. See 8.2 in the manual linked above if you want to know what they do.

- Timer

This tells the fan how long to run for in hours before it turns the motor off. It's exposed as an Enum on DpId 6, with values ranging from 0 to 9.

These last three controls are not well supported using the current set of FunctionIds available with the TuyaMCU command. The current best practice is to write rules.

**Describe the solution you'd like**  

I'd like to have Tasmota offer another type of function for use with the TuyaMCU command to support publishing and controlling the extra functionality described above. I think the simplest would be providing an "Enum" type, or "Dial" or something, which just reports and caches the values directly via MQTT when the MCU sends a state update, and sends an equivalent of `TuyaSend4` to the configured DpId when an update command is issued.

As well as there being a new type of function, I'd like a bunch of this type so I can easily publish and control multiple DpIds independently.

**Describe alternatives you've considered**  

I've considered enabling `SetOption66` so the TuyaReceived messages end up being sent via MQTT, but I don't want to parse the raw messages myself when tasmota already does a good of that. I also don't want to see the MCU heartbeat messages, or the messages already handled by POWER1 and POWER2.

I've also considered extending SetOption66 so it becomes a bitfield where you can request filtering of the heartbeat messages, filtering of messages where the checksum has failed, filtering of messages that are already handled by TuyaMCU filters, and enabling parsing of the messages by Tasmota so the MQTT messages have the DpId, DpIdType, and DpIdData like the output of Weblog 4.

**Additional context**  

I'm happy to try and write code to help in this area, but I'd like to discuss how it should work before I spend time doing it.

dgwynne, 2020-02-02:
Using `TuyaMcU 21,3` to control the fan speed doesn't work too well. I don't think it gets the message type right when sending commands to the MCU, and it enables the Speed and Fade controls in tasmota, which feels a bit awkward for a fan.

Jason2866, 2020-02-02:
Supported since https://github.com/arendst/Tasmota/pull/7535
Use latest development version to use the feature

dgwynne, 2020-02-03:
> Supported since #7535
> Use latest development version to use the feature

I think #7535 and this issue are talking about different things, but unfortunately use a lot of the same words. Commit https://github.com/arendst/Tasmota/commit/920501fef99d38c3886b26acc6db1bcb6a5fabcf implements parsing of messages that contain data from multiple DpIds. I'm asking for the creation of a new function type (ie, something new should go into the `TuyaSupportedFunctions` enum) to better support DpIds that act like a mode selector or option dial on the MCU. On top of that I'd like to be able to configure multiple functions of that type, like you can for Buttons or Relays.

dgwynne, 2020-02-03:
A quick and dirty diff to try and make it easier to publish the state of the device to MQTT is at https://github.com/dgwynne/Tasmota/commit/cf8cd47126cd051aa834abc1e316168d8f50dd43

With the following rule, I can see the speed, mode, and timer values change as the kids press stuff on the remote:

```
Rule1
  on TuyaReceived#Cmnd7DpId3DpType4 do Publish stat/%topic%/SPEED %value% endon
  on TuyaReceived#Cmnd7DpId2DpType4 do Publish stat/%topic%/FANMODE %value% endon
  on TuyaReceived#Cmnd7DpId6DpType4 do Publish stat/%topic%/FANTIMER %value% endon
```

It would still be nice to have DpIds mapped to functions so their state is visible in STATE updates too

closed 2020-02-20 by ascillato2