arendst/Tasmota #7655: Option to decouple switches/buttons from relays

micw, 2020-02-03:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
I want to automate different lights with non-trivial logic, so I'd like to use rules for it. Unfortunately, buttons/switches are automatically connected to relays or PWMs. So it's not possible to do the handling in rules only.
Example:
In my bathroom I have a light behind a glass door and a door switch at this door. The light is dimmed externally depending on day time/sun position (day=off, evening/morning=bright, night=15%). When the dor is opened, the light is dimmed to 100%, when it's closed, the light is dimmed to it's previous value.

**Describe the solution you'd like**  
Create an option to disable connection between switches/buttons and "power".

**Describe alternatives you've considered**  
Setting SwitchTopic does not help because it needs to be set to 0 to make rules work

Adding additional relays on non-used GPIOs is a workaround but dirty hack (requires available GPIOs, messes up the GUI with ON/OFF buttons without functionality)

micw, 2020-02-03:
I solved it, after adding the rule, the binding was removed. So it just works as I wish ;-)

ascillato2, 2020-02-03:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-02-03 by ascillato2

micw, 2020-02-03:
@ascillato2 Can you mention the duplicate? I did not find it when I opened this one.

ascillato, 2020-02-03:
On a quick search, there are more than 10 duplicated issues on this subject. The last duplicated issue is from 4 days ago: https://github.com/arendst/Tasmota/issues/7624 . This issue has been addressed so many times that now it should have a super-duplicated label. In the docs is also explained this subject, but don't worry because obviously we need to improve the docs in order to make this to be found more easily. Please, can you help us with that?

micw, 2020-02-03:
Sure, I can create a PR to the docs. I did not find any hint about that (I searched with search function in docs and issues but obviously used the wrong terms). What I found where some hints which after reading a lot pointed me in the right direction.

Example: https://tasmota.github.io/docs/#/Buttons-and-Switches
It mentions "corresponding relays" a lot of times but I did not found a definition for "corresponding" and how it's setup. This is done automagic and the doc assumes this knowledge.

Example: https://tasmota.github.io/docs/#/Rule-Cookbook?id=thermostat-example
> NOTE: Until rules are created the pushbutton will still control the relay!
This is the _only_ hint I found that adding a rule for a state suppresses the implicit connection between button and the relay.

To improve docs, I need a bit more information how things work:
- in which order are buttons assigned to PWMs and relays? In https://tasmota.github.io/docs/#/Commands?id=light I found `When SetOption68 is set to 1 Channel<x> will follow Power<x> numbering with Relays first then PWM` - does this mean with SetOption68 to 0, Power1 is the Light and relays follow?
- does adding a rule for switch<N>#state influence the order of that binding?
- does a rule for switch1#state prevents only the binding to be suppressed or does it also supress all other rules for switch1#state?

Henkonoly, 2023-11-13:
Hi,

I have I many sonoff THR316d tasmota flashed.
Before flashing it would also work without a  network to switch its internal relais on say 40 degrees

is there a possibility to regain this option?
Everything works fine in automation now via mqtt, but 1 trigger is very important to switch always at 40 degrees. 
Otherwise I might have a fire going ;)

Any sugestions are more the welkom

thx in advance

barbudor, 2023-11-13:
@Henkonoly your question seems totally unrelated to the topic of this discussion (setoption68) 

Please check as usage of Tasmota without network as been discussed many time but not here
Alternatively open a new discussion