arendst/Tasmota #7656: Support of MCP2515 Stand-alone SPI CAN Bus Controller

Jojo-A, 2020-02-03:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  
Hi,
first of all I like to thank you sooo much for this project/firmware and the effort you've all put in it!

I like to contribute in making this project even more flexible and versatile by requesting support of the stand-alone CAN bus controller MCP2515 (or equivalent).
The MCP2515 is a very broad available and relatively cheap CAN bus controller with SPI interface. It is available on tiny and cheap ready-to-use "evaluation boards" and also as shields for the Arduino platform

### Usecase
A system of several CAN bus devices which I like to control/monitor remotely.

### What I already have
I already created a very basic software (based on Arduino IDE), which runs on a Wemos ESP8266 board. It basically consists of the libraries:
- WifiManager
- PubSubClient
- CAN-BUS Shield

This software does (basically) nothing else but converting CAN to MQTT and vice versa, there the CAN bus ID is part of the topic, and the CAN bus data field is the payload (divided by ";" or "," or "." or " " into single bytes). There is also a sign for data or RTR frames.
So an example MQTT-to-CAN publish message to _send_ data into the CAN bus could look like this:
`cmnd/MQTTtoCAN/98765/DAT/100;200;250;150;66;3;21;10`
An example MQTT-to-CAN publish message to _request_ data from a CAN bus device could look like this;
`cmnd/MQTTtoCAN/98765/RTR/` (empty payload)

The other direction (CAN-to-MQTT) works the same. So if the ESP _receives_ a CAN message from a CAN bus device, it will be published to the MQTT broker as
`stat/MQTTtoCAN/98765/DAT/100;200;250;150;66;3;21;10`

This software runs quite reliable. Remarkable limitation: it only supports extended identifiers (not so bad from my point of view) and it is just not as flexible and mature as Tasmota.

### Idea regarding integration
I know, that this is not ideal for the data management used by Tasmota. So my idea is to handle it a little bit different:
the CAN controller could appear as a "sensor", with three values:
1. CAN ID
2. frame format (data or RTR)
3. data field

Every new incoming CAN message would trigger the "sensor" to be updated.
So the telemetry data from Tasmota could look like:
`tele/MQTTtoCAN/SENSOR = {"Time":"yyyy-mm-dd","MCP2515":{"ID":"98765","Format":"Dat","Data":"100;200;250;150;66;3;21;10"}}`

### What we would need
- the SPI CAN library
- a configurable chip select pin
- (a configurable interrupt pin?)
- a sensor component
- parsing/conversion from CAN message elements to MQTT (topic and payload)

### What I can provide
- "my" source code
- time and hardware for testing (ESP8266 boards and MCP2515 boards)

For me it looks as if many things are already there and need "only" to be merged. Would it be worth it?
What do you think?

Greetings,
Jojo

closed 2020-03-05 by stale[bot]

pkkrusty, 2022-09-26:
@Jojo-A any further interest in this? @kawaitek was working on a similar project, but hasn't PRd anything yet.

Jojo-A, 2022-09-29:
Hi,
well basically yes. The work of @kwiatek6324 seems indeed similar to what I have thought about. But I personally think, that there is not so much general interest in that. So I understand why developers put not much effort in this...

kwiatek6324, 2022-09-29:
Sorry guys for delay, a lot of other thinkgs were going on.
Created PR for this:
https://github.com/arendst/Tasmota/pull/16684

Regards
AK

arendst, 2022-09-29:
PR now implemented in such a way that you have to select either ``define USE_CANSNIFFER`` or define ``USE_MCP2515``.

Jojo-A, 2022-09-30:
> PR now implemented in such a way that you have to select either `define USE_CANSNIFFER` or define `USE_MCP2515`.

Okay, cool! I will give it a try when I have time!
If it works as I hope, my own work regarding the [MQTT-to-CAN](https://github.com/Jojo-A/MQTT-CAN-Gateway) gateway will be mostly obsolete ;)