arendst/Tasmota #7660: Wrong IR decoding (COOLIX) Midea HVAC

thewhitegoatkf, 2020-02-03:
### PROBLEM DESCRIPTION
Im having issue in decoding IR COOLIX protocol using tasmota 8.1.0 (IR/IRCUSTOM)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic
- [x] Tasmota binary firmware version number used:8.1.0
  - [x] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: PlatformIO
- [x] Flashing tools used: PlatformIO
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
17:47:26 MQT: sonoff/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
17:47:26 MQT: sonoff/stat/RESULT = {"Module":{"1":"Sonoff Basic"}}
17:47:27 MQT: sonoff/stat/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"51":"IRrecv"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
17:43:28 MQT: sonoff/stat/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:43:28 MQT: sonoff/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:50:51","StartupUTC":"2020-02-03T15:52:37","Sleep":50,"CfgHolder":4617,"BootCount":4,"SaveCount":12,"SaveAddress":"FA000"}}
17:43:28 MQT: sonoff/stat/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2020-02-03T17:47:29","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"370/699"}}
17:43:28 MQT: sonoff/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["REDACTED","REDACTED"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["400A8009","2805C8000100060000005AFF000000000000","00000300","00000000"]}}
17:43:28 MQT: sonoff/stat/STATUS4 = {"StatusMEM":{"ProgramSize":662,"Free":340,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
17:43:28 MQT: sonoff/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3858","IPAddress":"192.168.1.23","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"BC:DD:C2:EB:0F:12","Webserver":2,"WifiConfig":4}}
17:43:28 MQT: sonoff/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.22","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EB0F12","MqttUser":"sonoff","MqttCount":1,"MAX_PACKET_SIZE":2000,"KEEPALIVE":30}}
17:43:28 MQT: sonoff/stat/STATUS7 = {"StatusTIM":{"UTC":"Mon Feb 03 16:43:28 2020","Local":"Mon Feb 03 17:43:28 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:18","Sunset":"17:49"}}
17:43:28 MQT: sonoff/stat/STATUS10 = {"StatusSNS":{"Time":"2020-02-03T17:43:28"}}
17:43:28 MQT: sonoff/stat/STATUS11 = {"StatusSTS":{"Time":"2020-02-03T17:43:28","Uptime":"0T00:50:51","UptimeSec":3051,"Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"REDACTED":"REDACTED","BSSId":"78:44:76:XXXX:60","Channel":1,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:08"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  TASMOTA:

[17:55:17] sonoff/tele/RESULT {"IrReceived":{"Protocol":"COOLIX","Bits":24,"Data":"0xB2BFCC","DataLSB":"0x4DFD33","Repeat":0,"IRHVAC":{"Vendor":"COOLIX","Model":-1,"Power":"Off","Mode":"Off","Celsius":"On","Temp":0.0,"FanSpeed":"UNKNOWN","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},
"RawData":[4324,4356,510,1654,510,574,510,1654,510,1654,510,574,510,576,510,1654,510,578,510,572,510,1652,510,576,510,574,512,1654,510,1650,512,576,510,1656,510,1650,510,576,510,1654,510,1652,510,1654,510,1654,510,1652,512,1656,508,574,510,1654,510,576,510,576,510,574,510,576,510,574,510,578,510,1652,510,1654,510,576,510,574,510,1652,512,1652,512,576,510,578,510,572,510,574,512,1652,510,1654,510,574,510,576,510,1654,510,1654,536,5174,4352,4338,510,1652,510,576,510,1652,512,1654,510,576,510,574,510,1654,510,576,510,574,510,1654,510,576,510,574,510,1654,510,1652,512,576,510,1656,510,1652,510,574,516,1648,510,1654,510,1654,510,1654,510,1654,510,1656,510,574,510,1654,510,574,510,576,510,576,510,576,510,576,510,576,510,1652,510,1654,510,576,510,576,510,1654,510,1654,510,574,510,578,510,572,510,576,510,1654,510,1654,510,576,510,574,512,1654,510,1656,510],"RawDataInfo":[199,199,0]}}

[17:56:09] sonoff/tele/RESULT {"IrReceived":{"Protocol":"COOLIX","Bits":24,"Data":"0xB27BE0","DataLSB":"0x4DDE07","Repeat":0,"IRHVAC":{"Vendor":"COOLIX","Model":-1,"Power":"Off","Mode":"Off","Celsius":"On","Temp":0.0,"FanSpeed":"UNKNOWN","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1},
"RawData":[4352,4332,510,1654,510,576,510,1654,510,1652,512,576,510,576,510,1654,510,578,510,574,510,1654,510,576,510,576,510,1654,510,1654,510,578,508,1656,510,572,510,1654,510,1654,510,1654,510,1654,508,576,510,1654,510,1656,510,1652,510,574,510,576,510,578,508,576,510,1656,508,576,510,578,508,1650,512,1654,510,1654,510,576,510,576,510,576,510,576,508,578,510,574,510,576,510,576,510,1654,510,1622,542,1654,510,1654,510,1652,538,5174,4352,4338,510,1654,510,576,510,1654,510,1654,510,576,510,576,510,1654,510,578,510,574,512,1650,510,576,508,576,510,1654,510,1652,510,576,510,1656,510,574,510,1654,510,1654,510,1654,510,1604,560,574,510,1654,510,1656,508,1652,510,576,510,576,510,576,510,574,510,1654,510,576,508,578,510,1652,510,1654,510,1654,508,576,510,576,510,576,510,576,510,576,510,574,510,576,510,576,510,1654,510,1654,510,1654,510,1654,510,1654,510],"RawDataInfo":[199,199,0]}}

```

### TO REPRODUCE
Use a COOLIX protocol IR remote 

### EXPECTED BEHAVIOUR
Similar output as IRecvDumpv2 does on the same board with the correct values:
```
IRecvDumpv2:
Timestamp : 000128.058
Library   : v2.7.3

Protocol  : COOLIX
Code      : 0xB2BFCC (24 Bits)
Mesg Desc.: Power: On, Mode: 3 (Heat), Fan: 5 (Auto), Temp: 25C, Zone Follow: Off, Sensor Temp: Off
uint16_t rawData[199] = {4350, 4336,  510, 1652,  512, 574,  510, 1652,  510, 1652,  510, 576,  508, 576,  510, 1652,  510, 578,  508, 574,  510, 1652,  510, 574,  510, 574,  510, 1652,  510, 1654,  510, 574,  510, 1656,  510, 1650,  510, 576,  510, 1654,  510, 1652,  510, 1652,  510, 1652,  510, 1654,  510, 1658,  508, 572,  510, 1654,  510, 574,  510, 574,  510, 576,  510, 574,  510, 574,  510, 576,  512, 1650,  510, 1654,  516, 568,  508, 576,  510, 1654,  508, 1654,  510, 576,  510, 576,  510, 572,  512, 574,  510, 1654,  510, 1654,  508, 576,  508, 576,  510, 1652,  510, 1652,  534, 5174,  4350, 4336,  510, 1652,  510, 576,  510, 1654,  510, 1652,  510, 574,  510, 576,  510, 1654,  510, 576,  510, 574,  510, 1654,  508, 576,  510, 574,  510, 1654,  510, 1652,  510, 576,  510, 1654,  510, 1650,  510, 574,  510, 1654,  510, 1654,  510, 1652,  510, 1654,  510, 1654,  510, 1654,  512, 572,  510, 1652,  510, 574,  510, 574,  510, 576,  510, 574,  510, [... cut ...]  576,  510, 574,  512, 574,  510, 578,  510, 572,  512, 574,  510, 576,  508, 1654,  510, 1654,  510, 1652,  510, 1652,  512, 1654,  534};  // COOLIX B27BE0
uint64_t data = 0xB27BE0;

```

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
PubSubClient.h and tasmota_post.h
MQTT_MAX_PACKET_SIZE 2000

Tried the precompiled binaries (IR/IRCUSTOM) with the same behaviour

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-02-03:
It is a error in the used ir library version 2.7.2.
In your test you used the latest (just a few days released) version 2.7.3
See https://github.com/crankyoldgit/IRremoteESP8266/commit/0c6ba2d0f708744fc7ec1c09a96be8f60bd00daf

thewhitegoatkf, 2020-02-03:
Yes but I have compiled tasmota under 2.7.3 IRremoteESP8266 with the same result as the provided 2.7.1, while the IRecvDumpV2 of 2.7.3 decodes just fine.

Jason2866, 2020-02-03:
Only using the new 2.7.3 and compiling will not work, file IRremoteESP8266.h needs to be changed.

thewhitegoatkf, 2020-02-03:
I have noticed just now that I pulled a slightly older source with the 2.7.1 instead of the 2.7.2 will test later and update if it fixes the issue

Jason2866, 2020-02-03:
Or just wait. See #7661

s-hadinger, 2020-02-03:
I will try to look at it today or tomorrow. I doubt that 2.7.3 changes anything, the code for Tasmota-IR or Coolix didn't change in the last 3 months.

thewhitegoatkf, 2020-02-03:
Wow that was quick, tested the 2.7.2 doens't fix the issue with the dev build ,I think also increasing the IR_RCV_BUFFER_SIZE slightly might benefit lots more HVACs

thewhitegoatkf, 2020-02-03:
Just tested your merge with ircustom and larger buffer and still no go :/

s-hadinger, 2020-02-03:
It seems there is a bug in `ir_Coolix.cpp` that causes `Power` to be always `Off`, hence all other parameters are ignored. Still investigating.

s-hadinger, 2020-02-03:
Found it!

There is a special COOLIX handling in `resultAcToString()`

```
#if DECODE_COOLIX
      case decode_type_t::COOLIX: {
        IRCoolixAC ac(0);
        ac.on();
        ac.setRaw(result->value);  // Coolix uses value instead of state.
        return ac.toString();
      }
#endif  // DECODE_COOLIX
```

Specifically the `ac.on()` that defaults the Power to on and switches it off depending on state.

I need to open an issue in David's repo.

If you want a quick fix, in `lib/IRremoteESP8266-2.7.<x>/src/IRac.cpp`, line 2235, add `ac.on();`

You should have the following:

```

#if DECODE_COOLIX
      case decode_type_t::COOLIX: {
        IRCoolixAC ac(kGpioUnused);
        ac.on();
        ac.setRaw(decode->value);  // Uses value instead of state.
        *result = ac.toCommon(prev);
        break;
      }
#endif  // DECODE_COOLIX
```

I pushed the quick fix to the open PR #7661

thewhitegoatkf, 2020-02-03:
What a champ!
Fixed and working

closed 2020-02-03 by thewhitegoatkf