arendst/Tasmota #7667: MQTT Count rises every 5 min.

kugelkopf123, 2020-02-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Multiple reconnects from 2 of 5 devices.
I have MQTT Reconnects every 5 or 10 mins. It is pretty accurate. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Shelly1 & Shelly 2.5
- [ x] Tasmota binary firmware version number used: Tasmota 8.1.0.5
  - [ ] Pre-compiled
  - [ x] Self-compiled
    - [x ] IDE / Compiler used: PlatformIO on VS Base.
- [ x] Flashing tools used: PlatformIO>esptool.py
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
Shelly1:
15:01:39 CMD: Backlog Template; Module; GPIO 255
15:01:40 MQT: stat/Shelly1_1/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
15:01:40 MQT: stat/Shelly1_1/RESULT = {"Module":{"1":"Sonoff Basic"}}
15:01:40 MQT: stat/Shelly1_1/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

Shelly2.5:
15:17:00 MQT: stat/Shelly25_3/RESULT = {"NAME":"Shelly 2.5 (bu","GPIO":[56,0,19,0,21,127,0,0,6,126,5,22,156],"FLAG":2,"BASE":18}
15:17:00 MQT: stat/Shelly25_3/RESULT = {"Module":{"0":"Shelly 2.5 (bu"}}
15:17:00 MQT: stat/Shelly25_3/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"0":"None"},"GPIO2":{"19":"Button3"},"GPIO3":{"0":"None"},"GPIO4":{"21":"Relay1"},"GPIO5":{"127":"Button2in"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"6":"I2C SDA"},"GPIO13":{"126":"Button1in"},"GPIO14":{"5":"I2C SCL"},"GPIO15":{"22":"Relay2"},"GPIO16":{"156":"ADE7953 IRQ"}}
```
- [x ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
Shelly1: No Rules set.

Shelly2.5:
15:14:25 MQT: stat/Shelly25_3/RESULT = {"Rule1":"ON","Once":"ON","StopOnError":"OFF","Free":247,"Rules":"on energy#current[2]>0.800 do backlog var1=%value%;power1 0;power2 0; publish stat/%topic%/RESULT {"BLACKOUT1“:“%var1%“} endon on energy#current[1]>0.800 do backlog var2=%value%;power1 0;power2 0;publish stat/%topic%/RESULT {"BLACKOUT2“:“%var2%“} endon"}
15:14:26 MQT: stat/Shelly25_3/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Free":417,"Rules":"on SHUTTER1#Position do backlog publish stat/%topic%/RESULT {"ShutterPosition1":%value%} endon"}
15:14:26 MQT: stat/Shelly25_3/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [ x] Provide the output of this command: `Status 0`:
```
Shelly1:

15:02:34 MQT: stat/Shelly1_1/STATUS = {"Status":{"Module":1,"FriendlyName":["Shelly1_1"],"Topic":"Shelly1_1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:02:34 MQT: stat/Shelly1_1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power on","Uptime":"0T02:51:02","StartupUTC":"2020-02-04T11:11:32","Sleep":50,"CfgHolder":4617,"BootCount":2,"SaveCount":6,"SaveAddress":"F6000"}}
15:02:34 MQT: stat/Shelly1_1/STATUS2 = {"StatusFWR":{"Version":"8.1.0.5(tasmota)","BuildDateTime":"2020.02.03 00:07:59","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"340/699"}}
15:02:34 MQT: stat/Shelly1_1/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Eiswuerstchen",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
15:02:34 MQT: stat/Shelly1_1/STATUS4 = {"StatusMEM":{"ProgramSize":429,"Free":572,"Heap":27,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000407","8E002186","040003A1","00000004","000000C0","00000000"],"Drivers":"1,2,3,4,10","Sensors":"2,3"}}
15:02:34 MQT: stat/Shelly1_1/STATUS5 = {"StatusNET":{"Hostname":"Shelly1_1-7502","IPAddress":"192.168.1.201","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"A4:CF:12:B9:DD:4E","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:02:34 MQT: stat/Shelly1_1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.73","MqttPort":1883,"MqttClientMask":"Shelly1_1","MqttClient":"Shelly1_1","MqttUser":"tasmota","MqttCount":7,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
15:02:34 MQT: stat/Shelly1_1/STATUS7 = {"StatusTIM":{"UTC":"Tue Feb 04 14:02:34 2020","Local":"Tue Feb 04 15:02:34 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":99}}
15:02:34 MQT: stat/Shelly1_1/STATUS10 = {"StatusSNS":{"Time":"2020-02-04T15:02:34"}}
15:02:34 MQT: stat/Shelly1_1/STATUS11 = {"StatusSTS":{"Time":"2020-02-04T15:02:34","Uptime":"0T02:51:02","UptimeSec":10262,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":7,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Eiswuerstchen","BSSId":"E0:28:6D:22:3F:BC","Channel":1,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:06"}}}

Shelly2.5:
15:17:46 MQT: stat/Shelly25_3/STATUS = {"Status":{"Module":0,"FriendlyName":["Shelly25_3","Shelly25_32"],"Topic":"Shelly25_3","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:17:46 MQT: stat/Shelly25_3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T14:00:07","StartupUTC":"2020-02-04T00:17:39","Sleep":50,"CfgHolder":4617,"BootCount":7,"SaveCount":70,"SaveAddress":"FB000"}}
15:17:46 MQT: stat/Shelly25_3/STATUS2 = {"StatusFWR":{"Version":"8.1.0.5(tasmota)","BuildDateTime":"2020.02.02 18:46:02","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"393/699"}}
15:17:47 MQT: stat/Shelly25_3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Eiswuerstchen",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C80B","1405C8000100060000005A00000000000000","40000020","00000000"]}}
15:17:47 MQT: stat/Shelly25_3/STATUS4 = {"StatusMEM":{"ProgramSize":444,"Free":556,"Heap":25,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000407","8E002186","040003A1","00000004","010000C0","00000100"],"Drivers":"1,2,3,4,10,27","Sensors":"2,3"}}
15:17:47 MQT: stat/Shelly25_3/STATUS5 = {"StatusNET":{"Hostname":"Shelly25-3","IPAddress":"192.168.1.205","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"98:F4:AB:F3:4D:85","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:17:47 MQT: stat/Shelly25_3/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.73","MqttPort":1883,"MqttClientMask":"Shelly25_3","MqttClient":"Shelly25_3","MqttUser":"tasmota","MqttCount":61,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
15:17:47 MQT: stat/Shelly25_3/STATUS7 = {"StatusTIM":{"UTC":"Tue Feb 04 14:17:47 2020","Local":"Tue Feb 04 15:17:47 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":99}}
15:17:47 MQT: stat/Shelly25_3/STATUS10 = {"StatusSNS":{"Time":"2020-02-04T15:17:47","ANALOG":{"Temperature":46.5},"ENERGY":{"TotalStartTime":"2020-02-02T18:43:08","Total":0.009,"Yesterday":0.003,"Today":0.002,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":0,"Voltage":0,"Current":[0.000,0.000]},"Shutter1":{"Position":100,"Direction":0},"TempUnit":"C"}}
[... cut ...]
I was hoping that a complete reset via "reset 2" and switching off the power might help. However, this did not solve the Problem. With the Shelly25_3, the one with the problem, I did "erase flash" via PlatformIO at the beginning of last week and flashed it again via cable. With the Shelly1_1 it was 3-4 weeks ago that I removed it from its place behind the switch, deleted it and flashed it.

Last night I removed the Shelly1_1 and changed it for the Shelly1_2, which runs smoothly. The Shelly1_1 is currently running on a USB2Serial adapter via 3.3V on my desk to be able to reproduce the error. See image.

**(Please, remember to close the issue when the problem has been addressed)**
ok.

ascillato, 2020-02-04:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-02-04 by ascillato2

kugelkopf123, 2020-02-04:
I dont get it, why you think its not a bug?
Ive searched the issues and looked at the FAQ + Forum and was not able to find a solution that fix this Problem. Nearly everything that was mentioned in the FAQ or other Issues had i tried but nothing fix this. And thats defitly no Wifi Problem. The Shelly1_1 is just a half meter away from the Router. Perhaps it has something to do with: #7602

ascillato, 2020-02-04:
> Perhaps it has something to do with: #7602

Ok, please, try 8.1.0.4 that is before the referenced PR. You can downgrade to it without issues. The precompiled bin from the archive for 8.1.0.4 before this PR is:
http://thehackbox.org/tasmota/archive/20200125-200026-022a6a0-tasmota.bin

ascillato, 2020-02-04:
or if you like in German: http://thehackbox.org/tasmota/archive/20200125-200337-022a6a0-tasmota-DE.bin

kugelkopf123, 2020-02-05:
So.. New Informations. 
I wasnt able to flash the tasmota-DE.bin via WebGui. Got an Error. 
So i flashed it via Cable without erasing the Flash. 
Done Reset 2. Restart 1.
Problem was still present. 
Erased Flash via Cable and Uploaded your Tasmota-DE.bin
No Problem anymore. 
Rebuild my Version and uploaded it. Problem is back. 

user_config_override.h:

```

#ifndef _USER_CONFIG_OVERRIDE_H_
#define _USER_CONFIG_OVERRIDE_H_

#warning **** user_config_override.h: Using Settings from this File ****

#define MY_LANGUAGE            de-DE           // German in Germany

#undef USE_DOMOTICZ
#undef USE_HOME_ASSISTANT 
#undef USE_EMULATION_HUE
#undef USE_EMULATION_WEMO  
#undef USE_TIMERS 
#undef USE_SONOFF_RF 
#undef USE_SONOFF_SC 
#undef USE_TUYA_MCU 
#undef USE_ARMTRONIX_DIMMERS
#undef USE_PS_16_DZ 
#undef USE_SONOFF_IFAN 
#undef USE_BUZZER  
#undef USE_ARILUX_RF
#undef USE_DEEPSLEEP  
#undef USE_WS2812
#undef USE_MY92X1
#undef USE_SM16716 
#undef USE_SM2135
#undef USE_SONOFF_L1
#undef USE_COUNTER   
#undef USE_DS18x20
#undef USE_SHT
#undef USE_HTU
#undef USE_BH1750
#undef USE_SHT3X 
#undef USE_BMP
#undef USE_LM75AD  
#undef USE_ADE7953
#undef USE_MHZ19 
#undef USE_SENSEAIR
#undef USE_PMS5003 
#undef USE_NOVA_SDS
#undef USE_SERIAL_BRIDGE
#undef USE_ENERGY_MARGIN_DETECTION
#undef USE_PZEM004T
#undef USE_PZEM_AC 
#undef USE_PZEM_DC 
#undef USE_MCP39F501
#undef USE_DHT
#undef USE_IR_REMOTE 
#undef USE_IR_RECEIVE  
#undef USE_SR04 
#undef USE_HX711 
#undef USE_BMP
#undef USE_DISCOVERY          

#define APP_TIMEZONE 99
#undef NTP_SERVER1 
#undef NTP_SERVER2
#define NTP_SERVER1            "fritz.box"       // [NtpServer1] Select first NTP server by name or IP address (129.250.35.250)
#define NTP_SERVER2            "de.pool.ntp.org"    // [NtpServer2] Select second NTP server by name or IP address (5.39.184.5)

#undef LATITUDE
#define LATITUDE 52.123456

#undef LONGITUDE
#define LONGITUDE 7.123456

#undef ALTITUDE
#define ALTITUDE 33

#define Shelly1

//#define SonoffDual
//#define Shelly_25

#undef HOSTNAME
#undef FRIENDLY_NAME

#define USE_JAVASCRIPT_ES6 

#ifdef SonoffDual
#define WIFI_IP_ADDRESS "192.168.1.203" 
#define FRIENDLY_NAME "SonoffDual_R2"
    #define HOSTNAME "SonoffDual_R2"
    #undef MQTT_CLIENT_ID
    #define MQTT_CLIENT_ID "SonoffDual_R2"
    #undef MQTT_TOPIC
    #define MQTT_TOPIC "SonoffDual_R2"
    //#define MQTT_USE 0 
    #define TELE_PERIOD 300
    //#define USE_TIMERS     
//#undef USE_ADC_VCC
//#define USE_ADE7953
#define USE_SHUTTER           // Add Shutter support (+6k code)

#define USE_EXPRESSION                         // Add support for expression evaluation in rules (+3k2 code, +64 bytes mem)
#define SUPPORT_IF_STATEMENT                 // Add support for IF statement in rules (+4k2 code, -332 bytes mem)

#endif

#ifdef Shelly_25
#define WIFI_IP_ADDRESS "192.168.1.205" 
  #define FRIENDLY_NAME "Shelly25_3"
    #define HOSTNAME "Shelly25_3"
    #undef MQTT_CLIENT_ID
    #define MQTT_CLIENT_ID "Shelly25_3"
    #undef MQTT_TOPIC
    #define MQTT_TOPIC "Shelly25_3"
    //#define MQTT_USE 0 
    #define TELE_PERIOD 300
    //#define USE_TIMERS 
#undef USE_ADC_VCC
#define USE_ADE7953
#define USE_SHUTTER           // Add Shutter support (+6k code)

#define USE_EXPRESSION                         // Add support for expression evaluation in rules (+3k2 code, +64 bytes mem)
#define SUPPORT_IF_STATEMENT                 // Add support for IF statement in rules (+4k2 code, -332 bytes mem)
#endif

#ifdef fourCh
#define WIFI_IP_ADDRESS "192.168.1.200" 
#define FRIENDLY_NAME "4ch_OutdoorLights"
#define HOSTNAME "4ch-OutdoorLights"
    #undef MQTT_CLIENT_ID
    #define MQTT_CLIENT_ID "4ch_OutdoorLights"
    #undef MQTT_TOPIC
    #define MQTT_TOPIC "4ch_OutdoorLights"
    #define MQTT_USE 0 // MQTT Zuanfang ansgeschaltet
    #define TELE_PERIOD 300
   // #define USE_TIMERS 
#ifndef USE_RULES
#define USE_RULES
#endif  

#endif

#ifdef Shelly1
#define WIFI_IP_ADDRESS "192.168.1.201" 
 #define FRIENDLY_NAME "Shelly1_1"
    #define HOSTNAME "Shelly1-1"
    #undef MQTT_CLIENT_ID
    #define MQTT_CLIENT_ID "Shelly1_1"
    #undef MQTT_TOPIC
    #define MQTT_TOPIC "Shelly1_1"
    //#define MQTT_USE 0 
    //#define TELE_PERIOD 300
    //#define USE_TIMERS 
#endif

#undef  STA_SSID1
#define STA_SSID1         "My WifiSSID"             // [Ssid1] Wifi SSID

#undef  STA_PASS1
#define STA_PASS1         "My Wifi PW"     // [Password1] Wifi password

/*
#undef  STA_SSID2
#define STA_SSID2         ""             // [Ssid1] Wifi SSID

#undef  STA_PASS2
#define STA_PASS2         ""     // [Password1] Wifi password
*/

#undef  MQTT_HOST
#define MQTT_HOST         "192.168.1.123" // [MqttHost]

#undef  MQTT_PORT
#define MQTT_PORT         1883                   // [MqttPort] MQTT port (10123 on CloudMQTT)

#undef  MQTT_USER
#define MQTT_USER         "myMQTTUSER"         // [MqttUser] Optional user

#undef  MQTT_PASS
#define MQTT_PASS         "myMQTTPW"         // [MqttPassword] Optional password

#endif  // _USER_CONFIG_OVERRIDE_H_
```

Have no idea what the problem could be...

ascillato, 2020-02-05:
> Rebuild my Version and uploaded it. Problem is back.

under which Tasmota version? 8.1.0.4 or 8.1.0.6?

kugelkopf123, 2020-02-10:
So. I continued to research and found that the error occurs whenever my Mi Vaccuum vacuum cleaner drives. Now the question is, is it because of the ESP built in by me that also sends / receives data via MQTT, but on a different port (1884) with a separate iobroker MQTT broker. Or is it due to the Mi Vacuum ioBroker adapter that works on port: 53421

Is there anything known about these ports?
There are other MQTT applications running on the MQTT server used by the vacuum cleaner, which obviously do not cause any errors.

kugelkopf123, 2020-02-13:
Found the problem and wanted to report, so if someone has similar problems, knows what it could be. It was the static IP of the ESP07 that was installed in the vacuum cleaner. This ESP had the same IP as the Shelly25_3. Oddly enough, this caused all other Shelly and Sonoff devices to go crazy. The error has not occurred since the IP from the ESP07 changed to DHCP.

lobocobra, 2020-03-11:
Are you sure that this was the solution?
=> It is clear that it has an impact on the device with the duplicate IP, but it must not have any impact on others.

I experience as well regular d/c of MQTT on some devices, it can not be in my case that this is due to a fix IP. So could you please confirm that the problem is still gone after a month?

kugelkopf123, 2020-03-11:
I'll check that out. What I've found so far is that a Mqtt reconnect is also related to the renewal of the lease time of my router.
Yes, I was actually surprised that so many devices went crazy at once. I only got it out because the number of cases increased significantly while the robot vacuum was running.
I then found that the IP of the ESP I installed in the robot was static and the same as the static ip of one of my tasmota devices.
When I have checked it, I will contact you again.

kugelkopf123, 2020-03-14:
Just like i said. The Leasetime from my Fritzbox router is at the moment the reason that mqtt is counting up. but it rises just by one every three days.

kugelkopf123, 2020-03-18:
Just for info.
From time to time it can happen that such problems occur due to updates, especially updates of the core. Furthermore, there are also such problems if you change the flash memory size without first deleting the entire device via cable.

You can try "reset 3" in the web console and then completely remove the voltage. e.g. by turning off the circuit breaker.
Sometimes that helps. If not, only deleting and re-uploading the FW helps.