arendst/Tasmota #7668: Blitzwolf SHP-7 support

blueboi69, 2020-02-04:
**Have you looked for this feature in other issues and in the docs?**  
YES

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
YES. The power meter work incorrectly. It only displays 34W power. I tried the calibration and I asked for help in the support chat. The template that is shared is not working, it displays only 34W or below, if anything is turned on.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-02-04:
This device is already supported. Seems that you have a clone or a new version of this device. As you are the one that have this device, only you can figure out which gpio are assigned to.

If you need help on how to make a new template, you need to ask for help on the chat.

closed 2020-02-04 by ascillato2

barbudor, 2022-11-22:
It happens that manufacturers change their design and totally forget to let us know... 😂

You can suggest a edit on the template site to give your alternate templates