arendst/Tasmota #7672: can not calibrate power

stackner, 2020-02-05:
### PROBLEM DESCRIPTION
i am using the dev build 8.1.0.5 and a ct clamp on A0 when i try to calibrate the power and send POWERSET i just get unknown command.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ yes] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [yes] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [yes] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [yes] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [yes] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ *] Device used (e.g., Sonoff Basic): d1 mini and ct clamp
- [* ] Tasmota binary firmware version number used: 8.1.0.5
  - [yes] Pre-compiled
  - [no] Self-compiled
    - [* ] IDE / Compiler used: _____
- [* ] Flashing tools used: nodemcu pyflasher
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
05:58:06 MQT: stat/solar_meter/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
05:58:06 MQT: stat/solar_meter/RESULT = {"Module":{"18":"Generic"}}
05:58:06 MQT: stat/solar_meter/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
05:58:46 MQT: stat/solar_meter/STATUS = {"Status":{"Module":18,"FriendlyName":["Solar Meter"],"Topic":"solar_meter","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
05:58:46 MQT: stat/solar_meter/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"2T21:08:13","StartupUTC":"2020-02-02T07:50:33","Sleep":50,"CfgHolder":4617,"BootCount":7,"SaveCount":33,"SaveAddress":"FB000"}}
05:58:46 MQT: stat/solar_meter/STATUS2 = {"StatusFWR":{"Version":"8.1.0.5(26cee00-tasmota)","BuildDateTime":"2020-02-01T21:00:20","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"376/699"}}
05:58:46 MQT: stat/solar_meter/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["LFDD4x4","Recovery"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060100005A00000000000000","00000000","00000000"]}}
05:58:46 MQT: stat/solar_meter/STATUS4 = {"StatusMEM":{"ProgramSize":569,"Free":432,"Heap":23,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
05:58:46 MQT: stat/solar_meter/STATUS5 = {"StatusNET":{"Hostname":"Solar Meter","IPAddress":"192.168.1.173","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"B4:E6:2D:53:D8:4C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
05:58:46 MQT: stat/solar_meter/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.111","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_53D84C","MqttUser":"stackner","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
05:58:46 MQT: stat/solar_meter/STATUS7 = {"StatusTIM":{"UTC":"Wed Feb 05 04:58:46 2020","Local":"Wed Feb 05 05:58:46 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:15","Sunset":"17:53"}}
05:58:46 MQT: stat/solar_meter/STATUS10 = {"StatusSNS":{"Time":"2020-02-05T05:58:46","ANALOG":{"Energy":46.487,"Power":1989,"Voltage":230,"Current":8.648}}}
05:58:46 MQT: stat/solar_meter/STATUS11 = {"StatusSTS":{"Time":"2020-02-05T05:58:46","Uptime":"2T21:08:13","UptimeSec":248893,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"LFDD4x4","BSSId":"FC:EC:DA:38:E8:EB","Channel":8,"RSSI":100,"Signal":-23,"LinkCount":1,"Downtime":"0T00:00:57"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
05:59:35 CMD: powerset
05:59:35 MQT: stat/solar_meter/RESULT = {"Command":"Unknown"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
type POWERSET in the console or any power calibration command

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
will just get a unknown command responce

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-02-05:
Power calibaration is NOT supported when using a CT connected to ADC.
Please search in issues and pull request how it works.

ascillato2, 2020-02-07:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-02-07 by ascillato2

domdfcoding, 2022-08-03:
I'd be helpful if the page on power calibration said that calibration for the ADC isn't supported (or if it already does, it's not obvious )

barbudor, 2022-08-03:
It cannot be calibrated using the standard calibration commands (PowerCal/POwerSet, ...) but through `AdcParam 4,...` as documented here: https://tasmota.github.io/docs/ADC/#commands
The docs being open-source you can propose a change by clicking on the pencil button