arendst/Tasmota #7673: BL0937 - JulesV Power Calibration

Lu-Fi, 2020-02-05:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Device: "Jules V (Upgrade Version) Power Monitoring Plug" 
Template: https://templates.blakadder.com/jules_v.html
While Calibrating this Device the following Problem Occurs. Device is Calibrated by using a 60Watts Bulb, Multimeter shows 60.0Watt and 231Volt 0,260Amp.

VoltageSet 231 =  {"VoltageSetCal":1843}
PowerSet 60.0 = {"PowerSetCal":12911}
CurrentSet = 260.0 = {"CurrentSetCal":51}

Now when i pull off the Bulb, Voltage and Power are still 0 but Current remains 0,260Amp
If I now e.g. connect a clothes dryer. The Power Consumption is showing a maximum of 60Watts.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_
Any known Issues for other Expirience or this problem ?

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Jules V
- [x] Tasmota binary firmware version number used: latest
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tuya Convert / OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:

  Configuration output here:
09:50:17 CMD: Backlog Template; Module; GPIO 255
09:50:17 MQT: stat/JulesV_737B35/RESULT = {"NAME":"Jules-V_UV","GPIO":[56,0,0,0,0,134,0,0,21,17,132,57,131],"FLAG":0,"BASE":18}
09:50:17 MQT: stat/JulesV_737B35/RESULT = {"Module":{"0":"Jules-V_UV"}}
09:50:18 MQT: stat/JulesV_737B35/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"134":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"17":"Button1"},"GPIO14":{"132":"HLWBL CF1"},"GPIO15":{"57":"Led2i"},"GPIO16":{"131":"HLWBL SELi"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
09:51:47 CMD: Backlog Rule1; Rule2; Rule3
09:51:47 MQT: stat/JulesV_737B35/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
09:51:47 MQT: stat/JulesV_737B35/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
09:51:47 MQT: stat/JulesV_737B35/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
09:52:16 MQT: stat/JulesV_737B35/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"JulesV_737B35","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:52:16 MQT: stat/JulesV_737B35/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power on","Uptime":"1T23:38:04","StartupUTC":"2020-02-03T09:14:12","Sleep":50,"CfgHolder":4617,"BootCount":8,"SaveCount":58,"SaveAddress":"FA000"}}
09:52:16 MQT: stat/JulesV_737B35/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"380/699"}}
09:52:16 MQT: stat/JulesV_737B35/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FB_IOT","FB"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060100005A00000000000000","00000000","00000000"]}}
09:52:16 MQT: stat/JulesV_737B35/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
09:52:16 MQT: stat/JulesV_737B35/STATUS5 = {"StatusNET":{"Hostname":"JulesV_737B35-6965","IPAddress":"192.168.10.70","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.1","Mac":"80:7D:3A:73:7B:35","Webserver":2,"WifiConfig":4}}
09:52:16 MQT: stat/JulesV_737B35/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.45","MqttPort":7625,"MqttClientMask":"JulesV_%06X","MqttClient":"JulesV_737B35","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
09:52:16 MQT: stat/JulesV_737B35/STATUS7 = {"StatusTIM":{"UTC":"Wed Feb 05 08:52:16 2020","Local":"Wed Feb 05 09:52:16 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:15","Sunset":"17:53"}}
09:52:16 MQT: stat/JulesV_737B35/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
09:52:16 MQT: stat/JulesV_737B35/STATUS10 = {"StatusSNS":{"Time":"2020-02-05T09:52:16","ENERGY":{"TotalStartTime":"2020-02-03T09:46:44","Total":3.108,"Yesterday":1.224,"Today":0.006,"Power":0,"ApparentPower":59,"ReactivePower":59,"Factor":0.00,"Voltage":231,"Current":0.256}}}
09:52:16 MQT: stat/JulesV_737B35/STATUS11 = {"StatusSTS":{"Time":"2020-02-05T09:52:16","Uptime":"1T23:38:04","UptimeSec":171484,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"FB_IOT","BSSId":"3A:B5:C2:75:78:76","Channel":9,"RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
![image](https://user-images.githubusercontent.com/1177251/73826038-5c5b3280-47fd-11ea-8855-59ce77e22223.png)

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-02-05:
It looks like that the config of the power measurement GPIOs is not correct.
Sorry we cant help because we dont have the device

Lu-Fi, 2020-02-05:
Let me know, where to Ship.
[image]
[image]

Jason2866, 2020-02-05:
You have to find out which pin(s) go from the BL0937 to which pin of the ESP82xx

ascillato2, 2020-02-07:
Hi,

Any news on this?

ascillato2, 2020-02-08:
When you have time, please report back if you could find the correct gpio assignment for your device. Thanks

closed 2020-02-08 by ascillato2