arendst/Tasmota #7674: Today Energy unexpected increase

andker87, 2020-02-05:
### PROBLEM DESCRIPTION
Tasmota 8.1 on Wemos d1mini connected to a Pzem-016 (v 2.0).
While the real time readings are working well, it happens that suddenly values appear definitely out of scale in the "Energy Today" (and consequently in "Energy Total"). I consume about 6000kWh per day, but yesterday twice has been "added" 16000kWh, and today only once (for now) 12000kWh

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
{"Status":{"Module":18,"FriendlyName":["Tasmota"],"Topic":"pzem016","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0},"StatusPRM":{"Baudrate":9600,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Hardware Watchdog","Uptime":"0T00:18:35","StartupUTC":"2020-02-05T08:57:07","Sleep":50,"CfgHolder":4617,"BootCount":95,"SaveCount":123,"SaveAddress":"F9000"},"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T21:28:56","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"333/699"},"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WIFI-famcrestani",""],"TelePeriod":15,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]},"StatusMEM":{"ProgramSize":567,"Free":436,"Heap":25,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"},"StatusNET":{"Hostname":"pzem016-1963","IPAddress":"192.168.1.210","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"84:0D:8E:AA:87:AB","Webserver":2,"WifiConfig":4},"StatusMQT":{"MqttHost":"192.168.1.251","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AA87AB","MqttUser":"user_mqtt","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30},"StatusTIM":{"UTC":"Wed Feb 05 09:15:42 2020","Local":"Wed Feb 05 10:15:42 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:15","Sunset":"17:53"},"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0},"StatusSNS":{"Time":"2020-02-05T10:15:42","ENERGY":{"TotalStartTime":"2020-02-01T13:33:31","Total":13.176,"Yesterday":6.064,"Today":1.58,"Power":205,"ApparentPower":335,"ReactivePower":266,"Factor":0.61,"Frequency":50,"Voltage":232,"Current":1.443}},"StatusSTS":{"Time":"2020-02-05T10:15:42","Uptime":"0T00:18:35","UptimeSec":1115,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"WIFI-myownwifi","BSSId":"08:6A:0A:E7:FA:CD","Channel":6,"RSSI":44,"Signal":-78,"LinkCount":1,"Downtime":"0T00:00:44"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-02-05:
Hi, according with your status 0, you are having hardware watchdogs. Check if your power supply is ok.

andker87, 2020-02-05:
I will try to disconnect the wemos and redo all the connections. I followed [this](https://hassiohelp.eu/2019/03/27/pzem-016/) guide and put two 2k resistors.
can it be that this value so greatly wrong is due to the power supply? or is it more likely a bug in the tasmota calculation?

ascillato, 2020-02-05:
As your device is having watchdog restarts, it is likely that your issue comes from a bad power supply or a bad flashing. Please, try instead the information in http://Tasmota.com

If you didn't erase the flash before flashing tasmota, you need to do a reset 5, reconfigure and then do a reset 3 and after a reboot do a power cycle of your device.

If you need help on that, remember that it is available the Tasmota Support Chat.

Jason2866, 2020-02-05:
Sorry cant reproduce, i have a Pzem-016 too. Please check what already suggested from @ascillato

andker87, 2020-02-05:
> As your device is having watchdog restarts, it is likely that your issue comes from a bad power supply or a bad flashing. Please, try instead the information in http://Tasmota.com
>
> If you didn't [...]

I erased the memory before flashing tasmota.
I think the best thing to do is to reopen the pzem, remove everything and redo the welds on the wemos.
in your opinion is therefore a problem of the 5v/gnd that goes from pzem to nodemcu and you can exclude that the problem is related to the connections/welding of the RX/TX? Because of my lack of welding expertise imho those welds were quite complex...

ascillato, 2020-02-05:
You need to solve your hardware watchdog restarts. Having that abnormal behaviour may result in your issue, and if it is not a software problem, we can't do anything from our side to solve it.

Most of the times, hardware watchdog is insufficient power or bad flashing.

ascillato2, 2020-02-08:
When you have time, please, report back if you could solve your hardware watchdogs issue changing power supply and re flashing.

Thanks

closed 2020-02-08 by ascillato2

andker87, 2020-02-09:
I redid everything from the beginning. I erased the memory, flashed Tasmota and redid all the connections and weldings. After hours of testing, everything was fine.  So I put the pzem in the system and after a few hours I noticed that the energy today values sometimes go down by themselves (I enclose the image extracted from home assistant).
[url=https://ibb.co/4VBm8T3][img]https://i.ibb.co/4VBm8T3/IMG-4718.jpg[/img][/url]
Here below the status 0

> 21:07:10 CMD: Status 0
21:07:10 MQT: stat/pzem016/STATUS = {"Status":{"Module":18,"FriendlyName":["Tasmota"],"Topic":"pzem016","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:07:10 MQT: stat/pzem016/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power on","Uptime":"0T04:33:07","StartupUTC":"2020-02-09T15:34:03","Sleep":0,"CfgHolder":4617,"BootCount":7,"SaveCount":21,"SaveAddress":"F7000"}}
21:07:10 MQT: stat/pzem016/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T21:28:56","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"333/699"}}
21:07:10 MQT: stat/pzem016/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WIFI-famcrestani",""],"TelePeriod":15,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
21:07:10 MQT: stat/pzem016/STATUS4 = {"StatusMEM":{"ProgramSize":567,"Free":436,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
21:07:10 MQT: stat/pzem016/STATUS5 = {"StatusNET":{"Hostname":"pzem016-1963","IPAddress":"192.168.1.220","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"84:0D:8E:AA:87:AB","Webserver":2,"WifiConfig":4}}
21:07:10 MQT: stat/pzem016/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.251","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AA87AB","MqttUser":"mqttuser","MqttCount":13,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
21:07:10 MQT: stat/pzem016/STATUS7 = {"StatusTIM":{"UTC":"Sun Feb 09 20:07:10 2020","Local":"Sun Feb 09 21:07:10 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:09","Sunset":"17:59"}}
21:07:10 MQT: stat/pzem016/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
21:07:10 MQT: stat/pzem016/STATUS10 = {"StatusSNS":{"Time":"2020-02-09T21:07:10","ENERGY":{"TotalStartTime":"2020-02-09T16:42:58","Total":0.805,"Yesterday":0.000,"Today":0.805,"Power":501,"ApparentPower":566,"ReactivePower":263,"Factor":0.89,"Frequency":50,"Voltage":231,"Current":2.445}}}
21:07:10 MQT: stat/pzem016/STATUS11 = {"StatusSTS":{"Time":"2020-02-09T21:07:10","Uptime":"0T04:33:07","UptimeSec":16387,"Heap":24,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":13,"Wifi":{"AP":1,"SSId":"WIFI-famcrestani","BSSId":"08:6A:0A:E7:FA:CD","Channel":6,"RSSI":70,"Signal":-65,"LinkCount":10,"Downtime":"0T00:09:36"}}}
21:07:17 MQT: tele/pzem016/STATE = {"Time":"2020-02-09T21:07:16","Uptime":"0T04:33:13","UptimeSec":16393,"Heap":26,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":13,"Wifi":{"AP":1,"SSId":"WIFI-famcrestani","BSSId":"08:6A:0A:E7:FA:CD","Channel":6,"RSSI":48,"Signal":-76,"LinkCount":10,"Downtime":"0T00:09:36"}}
21:07:17 MQT: tele/pzem016/SENSOR = {"Time":"2020-02-09T21:07:17","ENERGY":{"TotalStartTime":"2020-02-09T16:42:58","Total":0.806,"Yesterday":0.000,"Today":0.806,"Period":2,"Power":491,"ApparentPower":555,"ReactivePower":257,"Factor":0.89,"Frequency":50,"Voltage":231,"Current":2.405}}

andker87, 2020-02-10:
watchdogs issues seem to be solved. anyway, i'm finding abnormal value resets (something that happened in the last hours when watchdogs issue was present). the setoption72 is set to 0, so if i understand correctly the counter of the values is in tasmota and not get by the pzem. resets don't happen periodically, but randomly.
as I already said, I cleaned the memory, flashed tasmota from scratch (my own compil) and redo all the connections and soldering.

[image]

[image]

ascillato, 2020-02-10:
That means that you are still having resets. Have you change the power source to a more powerful one?

andker87, 2020-02-10:
yes I did. The only thing (I think) that I can test is to enable the SetOption72 to get Enrgy data from the pzem. if, in that case, everything goes well, I presume the problem is related to Tasmota. Is there any other test that I can do?

ascillato, 2020-02-10:
After you see the reset to 0 of the energy today measurement, please, do `status 0` on the console of that device to see the restart reason. Please, post the output here. Thanks.

andker87, 2020-02-10:
I restarted the wemos d1 mini and reset the energy values (today, yesterday and total) to 0 and cleaned the db (to have a clean graph). I have the Sleep value to 0 and the SetOption72 set to 0 (as default). now it's all working, as soon as I notice a reset of the energy value I query the status 0 and post the output here.

andker87, 2020-02-10:
It just happened. After about 16 min, Today Energy value has been reset, while the d1 mini is still running (no reset).

Here the Status 0 and below the graph
`{"Status":{"Module":18,"FriendlyName":["Tasmota"],"Topic":"pzem016","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0},"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:16:58","StartupUTC":"2020-02-10T13:45:28","Sleep":0,"CfgHolder":4617,"BootCount":10,"SaveCount":36,"SaveAddress":"F8000"},"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T21:28:56","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"333/699"},"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WIFI-myownwifi",""],"TelePeriod":15,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]},"StatusMEM":{"ProgramSize":567,"Free":436,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"},"StatusNET":{"Hostname":"pzem016","IPAddress":"192.168.1.171","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"84:0D:8E:AA:87:AB","Webserver":2,"WifiConfig":4},"StatusMQT":{"MqttHost":"192.168.1.251","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AA87AB","user_mqtt":"user_mqtt","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30},"StatusTIM":{"UTC":"Mon Feb 10 14:02:26 2020","Local":"Mon Feb 10 15:02:26 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:07","Sunset":"18:01"},"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0},"StatusSNS":{"Time":"2020-02-10T15:02:26","ENERGY":{"TotalStartTime":"2020-02-10T14:45:14","Total":0.006,"Yesterday":0,"Today":0.006,"Power":341,"ApparentPower":388,"ReactivePower":185,"Factor":0.88,"Frequency":50,"Voltage":227,"Current":1.707}},"StatusSTS":{"Time":"2020-02-10T15:02:26","Uptime":"0T00:16:58","UptimeSec":1018,"Heap":25,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"Wifi":{"AP":1,"SSId":"WIFI-myownwifi","BSSId":"08:6A:0A:E7:FA:CD","Channel":6,"RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:12"}}}`

[image]

arendst, 2020-02-10:
Then first upgrade to the leatest dev version 8.1.0.7

ascillato, 2020-02-10:
And after updating, please set weblog to 4 and leave the console open. Then, when the reset to 0 happens, please share the output of the console. Thanks

andker87, 2020-02-10:
got it under 8.1.0 right now!
`19:31:31 MQT: tele/pzem016/STATE = {"Time":"2020-02-10T19:31:31","Uptime":"0T04:46:03","UptimeSec":17163,"Heap":26,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"Wifi":{"AP":1,"SSId":"WIFI-famcrestani","BSSId":"08:6A:0A:E7:FA:CD","Channel":6,"RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:12"}}
19:31:31 MQT: tele/pzem016/SENSOR = {"Time":"2020-02-10T19:31:31","ENERGY":{"TotalStartTime":"2020-02-10T14:45:14","Total":1.687,"Yesterday":0.000,"Today":1.687,"Period":4,"Power":797,"ApparentPower":836,"ReactivePower":253,"Factor":0.95,"Frequency":50,"Voltage":225,"Current":3.721}}
19:31:32 DMP: 01 04 14 08 C7 0E 9F 00 00 1F 4C 00 00 EF 44 00 00 01 F4 00 5F 00 00 80 38
19:31:33 DMP: 01 04 14 08 C6 0E 9B 00 00 1F 42 00 00 EF 44 00 00 01 F4 00 5F 00 00 E9 8B
19:31:34 DMP: 01 04 14 08 C6 0E 9B 00 00 1F 42 00 00 EF 44 00 00 01 F4 00 5F 00 00 E9 8B
19:31:35 DMP: 01 04 14 08 C8 0E 8C 00 00 1F 2A 00 00 EF 45 00 00 01 F4 00 5F 00 00 AF EE
19:31:36 DMP: 01 04 14 08 C7 0E 96 00 00 1F 38 00 00 EF 45 00 00 01 F3 00 5F 00 00 5F 8E
19:31:37 DMP: 01 04 14 08 BC 0E A8 00 00 1F 39 00 00 EF 45 00 00 01 F4 00 5F 00 00 64 4D
19:31:38 DMP: 01 04 14 08 B6 0E 99 00 00 1F 00 00 00 EF 45 00 00 01 F4 00 5F 00 00 E6 32
19:31:39 DMP: 01 04 14 08 B6 0E 99 00 00 1F 00 00 00 EF 45 00 00 01 F3 00 5F 00 00 53 F2
19:31:40 DMP: 01 04 14 08 B8 0E 9C 00 00 1F 0D 00 00 EF 46 00 00 01 F3 00 5F 00 00 2D F7
19:31:41 DMP: 01 04 14 08 B9 0E 9B 00 00 1F 18 00 00 EF 46 00 00 01 F4 00 5F 00 00 25 6F
19:31:42 DMP: 01 04 14 08 B9 0E A5 00 00 1F 27 00 00 EF 46 00 00 01 F4 00 5F 00 00 00 FC
19:31:43 DMP: 01 04 14 08 B9 0E A5 00 00 1F 27 00 00 EF 46 00 00 01 F4 00 5F 00 00 00 FC
19:31:44 DMP: 01 04 14 08 B5 0E A6 00 00 1F 1D 00 00 EF 47 00 00 01 F4 00 5F 00 00 B7 37
19:31:45 WIF: Checking connection...
19:31:45 WIF: Connected
19:31:45 DMP: 01 04 14 08 B3 0E A7 00 00 1F 13 00 00 EF 47 00 00 01 F4 00 5F 00 00 78 A1
19:31:46 DMP: 01 04 14 08 B4 0E 8C 00 00 1E D7 00 00 EF 47 00 00 01 F4 00 5F 00 00 AC 69
19:38:45 WIF: Connected
19:38:45 DMP: 01 04 14 08 B5 0E B8 00 00 1E BD 00 00 EF 9C 00 00 01 F4 00 5E 00 00 E1 C4
19:38:46 DMP: 01 04 14 08 B5 0E C2 00 00 1E D8 00 00 EF 9C 00 00 01 F4 00 5E 00 00 6F 18
19:38:46 MQT: tele/pzem016/STATE = {"Time":"2020-02-10T19:38:46","Uptime":"0T04:53:18","UptimeSec":17598,"Heap":27,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"Wifi":{"AP":1,"SSId":"WIFI-famcrestani","BSSId":"08:6A:0A:E7:FA:CD","Channel":6,"RSSI":48,"Signal":-76,"LinkCount":1,"Downtime":"0T00:00:12"}}
19:38:46 MQT: tele/pzem016/SENSOR = {"Time":"2020-02-10T19:38:46","ENERGY":{"TotalStartTime":"2020-02-10T14:45:14","Total":0.003,"Yesterday":0.000,"Today":0.003,"Period":42947904,"Power":790,"ApparentPower":842,"ReactivePower":293,"Factor":0.94,"Frequency":50,"Voltage":223,"Current":3.778}}
19:38:47 DMP: 01 04 14 08 BB 0E BF 00 00 1E E5 00 00 EF 9D 00 00 01 F4 00 5E 00 00 EB 94
19:38:48 DMP: 01 04 14 08 B4 0E C5 00 00 1E D9 00 00 EF 9D 00 00 01 F4 00 5E 00 00 EF C4
19:38:49 DMP: 01 04 14 08 B4 0E C5 00 00 1E D9 00 00 EF 9D 00 00 01 F4 00 5E 00 00 EF C4
19:38:50 DMP: 01 04 14 08 B0 0E DD 00 00 1E F5 00 00 EF 9D 00 00 01 F3 00 5E 00 00 4F 9F
19:38:51 DMP: 01 04 14 08 BB 0E C9 00 00 1E F7 00 00 EF 9D 00 00 01 F4 00 5E 00 00 14 3C
19:38:52 DMP: 01 04 14 08 D0 0E A9 00 00 1F 03 00 00 EF 9E 00 00 01 F4 00 5E 00 00 5E 4B
19:38:53 DMP: 01 04 14 08 D0 0E A9 00 00 1F 03 00 00 EF 9E 00 00 01 F4 00 5E 00 00 5E 4B
19:38:54 DMP: 01 04 14 08 C3 0E AE 00 00 1E D8 00 00 EF 9E 00 00 01 F4 00 5E 00 00 49 26
19:38:55 DMP: 01 04 14 08 CA 0E B2 00 00 1F 00 00 00 EF 9E 00 00 01 F4 00 5E 00 00 7E 65
19:38:56 DMP: 01 04 14 08 CA 0E A8 00 00 1E E8 00 00 EF 9F 00 00 01 F4 00 5E 00 00 39 7A
19:38:57 DMP: 01 04 14 08 CD 0E AA 00 00 1E F2 00 00 EF 9F 00 00 01 F4 00 5E 00 00 0B F4
19:38:58 DMP: 01 04 14 08 CD 0E AA 00 00 1E F2 00 00 EF 9F 00 00 01 F4 00 5E 00 00 0B F4
19:38:59 DMP: 01 04 14 08 D1 0E AA 00 00 1F 0A 00 00 EF 9F 00 00 01 F4 00 5E 00 00 83 1E
19:39:00 DMP: 01 04 14 08 CF 0E B3 00 00 1F 0F 00 00 EF 9F 00 00 01 F4 00 5E 00 00 FA A6
19:39:01 DMP: 01 04 14 08 B7 0E C1 00 00 1E DE 00 00 EF A0 00 00 01 F4 00 5E 00 00 EE 97
19:39:01 MQT: tele/pzem016/STATE = {"Time":"2020-02-10T19:39:01","Uptime":"0T04:53:33","UptimeSec":17613,"Heap":26,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"Wifi":{"AP":1,"SSId":"WIFI-famcrestani","BSSId":"08:6A:0A:E7:FA:CD","Channel":6,"RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:12"}}
19:39:01 MQT: tele/pzem016/SENSOR = {"Time":"2020-02-10T19:39:01","ENERGY":{"TotalStartTime":"2020-02-10T14:45:14","Total":0.007,"Yesterday":0.000,"Today":0.007,"Period":4,"Power":790,"ApparentPower":843,"ReactivePower":293,"Factor":0.94,"Frequency":50,"Voltage":223,"Current":3.777}}
19:39:02 DMP: 01 04 14 08 B7 0E C1 00 00 1E DE 00 00 EF A0 00 00 01 F4 00 5E 00 00 EE 97
19:39:03 DMP: 01 04 14 08 BF 0E B7 00 00 1E E2 00 00 EF A0 00 00 01 F4 00 5E 00 00 9A B0
19:39:04 DMP: 01 04 14 08 BC 0E B6 00 00 1E D6 00 00 EF A0 00 00 01 F4 00 5E 00 00 C0 10
19:39:05 WIF: Checking connection...
19:39:05 WIF: Connected
19:39:05 DMP: 01 04 14 08 BA 0E D5 00 00 1F 13 00 00 EF A0 00 00 01 F4 00 5E 00 00 9D B3
19:39:06 DMP: 01 04 14 08 B7 0E BA 00 00 1E CD 00 00 EF A1 00 00 01 F4 00 5E 00 00 8A BD
19:39:07 DMP: 01 04 14 08 B7 0E BA 00 00 1E CD 00 00 EF A1 00 00 01 F4 00 5E 00 00 8A BD
19:39:08 DMP: 01 04 14 08 B3 0E BA 00 00 1E B7 00 00 EF A1 00 00 01 F4 00 5E 00 00 DC B7
[... cut ...]
19:39:16 MQT: tele/pzem016/STATE = {"Time":"2020-02-10T19:39:16","Uptime":"0T04:53:48","UptimeSec":17628,"Heap":26,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"Wifi":{"AP":1,"SSId":"WIFI-famcrestani","BSSId":"08:6A:0A:E7:FA:CD","Channel":6,"RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:12"}}
19:39:16 MQT: tele/pzem016/SENSOR = {"Time":"2020-02-10T19:39:16","ENERGY":{"TotalStartTime":"2020-02-10T14:45:14","Total":0.010,"Yesterday":0.000,"Today":0.010,"Period":3,"Power":805,"ApparentPower":855,"ReactivePower":290,"Factor":0.94,"Frequency":50,"Voltage":226,"Current":3.791}}`

now I try to update...

ascillato, 2020-02-10:
Ok, so there is no restart. It just jump from `"Today":1.687` to `"Today":0.003`. Weird.

Please, can you to latest Tasmota?

andker87, 2020-02-11:
I upgraded to 8.1.0.7, that's what happened:
1) it happened again that the today energy value dropped (unfortunately I didn't have access to the console so I don't have the values)
2) tonight at 2 a.m. (so after starting the count from 0 for the new day), the value suddenly increased to 42949.672. this morning when I woke up I reset the value manually with EnergyReset10, and since then the realtime detection works, but the Today Energy is fixed at 0 and does not increase

> 11T06:46:14","Uptime":"0T10:00:59","UptimeSec":36059,"Heap":26,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"Wifi":{"AP":1,"SSId":"WIFI-myownwifi","BSSId":"08:6A:0A:E7:FA:CD","Channel":6,"RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:08"}}
06:46:14 MQT: tele/pzem016/SENSOR = {"Time":"2020-02-11T06:46:14","ENERGY":{"TotalStartTime":"2020-02-10T14:45:14","Total":1.858,"Yesterday":1.858,"Today":0.000,"Period":0,"Power":149,"ApparentPower":307,"ReactivePower":269,"Factor":0.48,"Frequency":50,"Voltage":231,"Current":1.328}}
06:46:15 DMP: 01 04 14 09 08 05 36 00 00 05 DB 00 00 FE 73 00 00 01 F4 00 31 00 00 98 CB
06:46:16 DMP: 01 04 14 09 07 05 33 00 00 05 D4 00 00 FE 73 00 00 01 F4 00 31 00 00 C8 80
06:46:17 DMP: 01 04 14 09 07 05 35 00 00 05 DC 00 00 FE 73 00 00 01 F4 00 31 00 00 34 69
06:46:17 WIF: Checking connection...
06:46:18 DMP: 01 04 14 09 07 05 35 00 00 05 DC 00 00 FE 73 00 00 01 F3 00 31 00 00 81 A9
06:46:19 DMP: 01 04 14 09 06 05 35 00 00 05 D8 00 00 FE 73 00 00 01 F3 00 31 00 00 B2 D1
06:46:20 DMP: 01 04 14 09 06 05 38 00 00 05 DE 00 00 FE 73 00 00 01 F3 00 31 00 00 2F 84
06:46:21 DMP: 01 04 14 09 03 05 2D 00 00 05 C6 00 00 FE 74 00 00 01 F4 00 30 00 00 1E 34
06:46:22 DMP: 01 04 14 09 03 05 33 00 00 05 D4 00 00 FE 74 00 00 01 F4 00 31 00 00 1F 80
06:46:22 WIF: Checking connection...
06:46:23 DMP: 01 04 14 09 03 05 33 00 00 05 D4 00 00 FE 74 00 00 01 F4 00 31 00 00 1F 80
06:46:24 DMP: 01 04 14 09 08 05 37 00 00 05 DA 00 00 FE 74 00 00 01 F4 00 31 00 00 ED EA
06:46:25 DMP: 01 04 14 09 08 05 39 00 00 05 E0 00 00 FE 74 00 00 01 F4 00 31 00 00 D0 73
06:46:26 DMP: 01 04 14 09 06 05 37 00 00 05 DD 00 00 FE 74 00 00 01 F4 00 31 00 00 8C 44
06:46:27 DMP: 01 04 14 09 06 05 37 00 00 05 DD 00 00 FE 74 00 00 01 F4 00 31 00 00 8C 44
06:46:27 WIF: Checking connection...
06:46:28 DMP: 01 04 14 09 04 05 34 00 00 05 D6 00 00 FE 74 00 00 01 F4 00 31 00 00 1F 87
06:46:29 DMP: 01 04 14 09 04 05 41 00 00 05 F5 00 00 FE 74 00 00 01 F4 00 31 00 00 53 AE
06:46:29 MQT: tele/pzem016/STATE = {"Time":"2020-02-11T06:46:29","Uptime":"0T10:01:14","UptimeSec":36074,"Heap":26,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"Wifi":{"AP":1,"SSId":"WIFI-myownwifi","BSSId":"08:6A:0A:E7:FA:CD","Channel":6,"RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:08"}}
06:46:29 MQT: tele/pzem016/SENSOR = {"Time":"2020-02-11T06:46:29","ENERGY":{"TotalStartTime":"2020-02-10T14:45:14","Total":1.858,"Yesterday":1.858,"Today":0.000,"Period":0,"Power":153,"ApparentPower":310,"ReactivePower":270,"Factor":0.49,"Frequency":50,"Voltage":231,"Current":1.345}}
06:46:30 DMP: 01 04 14 09 0B 05 33 00 00 05 D1 00 00 FE 74 00 00 01 F4 00 30 00 00 A0 24
06:46:31 DMP: 01 04 14 09 0A 05 37 00 00 05 DE 00 00 FE 74 00 00 01 F4 00 31 00 00 9A D6
06:46:32 DMP: 01 04 14 09 0A 05 37 00 00 05 DE 00 00 FE 74 00 00 01 F3 00 31 00 00 2F 16
06:46:32 WIF: Checking connection...
06:46:33 DMP: 01 04 14 09 0A 05 3A 00 00 05 E5 00 00 FE 74 00 00 01 F4 00 31 00 00 55 3E
06:46:34 DMP: 01 04 14 09 0A 05 30 00 00 05 CF 00 00 FE 74 00 00 01 F4 00 30 00 00 45 36
06:46:35 DMP: 01 04 14 09 08 05 33 00 00 05 D4 00 00 FE 74 00 00 01 F4 00 30 00 00 E8 65
06:46:36 DMP: 01 04 14 09 08 05 33 00 00 05 D4 00 00 FE 74 00 00 01 F4 00 30 00 00 E8 65
06:46:37 NTP: Drift 0, (UTC) Tue Feb 11 05:46:36 2020, (DST) Sun Mar 29 02:00:00 2020, (STD) Sun Oct 25 03:00:00 2020
06:46:37 DMP: 01 04 14 09 05 05 2B 00 00 05 BD 00 00 FE 74 00 00 01 F4 00 30 00 00 D0 A6
06:46:37 HTP: Console
06:46:37 WIF: Checking connection...
06:46:38 DMP: 01 04 14 09 09 05 37 00 00 05 D7 00 00 FE 74 00 00 01 F3 00 30 00 00 27 5B
06:46:39 DMP: 01 04 14 09 0C 05 3E 00 00 05 EA 00 00 FE 74 00 00 01 F4 00 31 00 00 8B 78
06:46:40 DMP: 01 04 14 09 0B 05 3A 00 00 05 E8 00 00 FE 74 00 00 01 F4 00 31 00 00 7B 8F
06:46:41 DMP: 01 04 14 09 0B 05 3A 00 00 05 E8 00 00 FE 74 00 00 01 F4 00 31 00 00 7B 8F
06:46:42 DMP: 01 04 14 09 0B 05 37 00 00 05 DE 00 00 FE 74 00 00 01 F3 00 31 00 00 12 EA
06:46:42 WIF: Checking connection...
06:46:43 DMP: 01 04 14 09 0B 05 32 00 00 05 D0 00 00 FE 75 00 00 01 F4 00 30 00 00 FE A5
06:46:44 DMP: 01 04 14 09 0C 05 34 00 00 05 D9 00 00 FE 75 00 00 01 F4 00 31 00 00 E4 79
DMP: 01 04 14 09 08 05 39 00 00 05 E5 00 00 FE 76 00 00 01 F4 00 31 00 00 C5 16

so I had to restart the d1 mini...Then the value reset happened again:

> 07:37:15 MQT: tele/pzem016/STATE = {"Time":"2020-02-11T07:37:15","Uptime":"0T00:43:27","UptimeSec":2607,"Heap":26,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"Wifi":{"AP":1,"SSId":"WIFI-myownwifi","BSSId":"08:6A:0A:E7:FA:CD","Channel":6,"RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:06"}}
07:37:15 MQT: tele/pzem016/SENSOR = {"Time":"2020-02-11T07:37:15","ENERGY":{"TotalStartTime":"2020-02-10T14:45:14","Total":2.048,"Yesterday":1.858,"Today":0.190,"Period":0,"Power":244,"ApparentPower":326,"ReactivePower":217,"Factor":0.75,"Frequency":50,"Voltage":232,"Current":1.406}}
[... cut ...]
07:38:47 DMP: 01 04 14 09 15 05 7F 00 00 09 78 00 00 FF 4F 00 00 01 F3 00 4A 00 00 89 50
07:38:48 DMP: 01 04 14 09 1D 05 81 00 00 09 84 00 00 FF 4F 00 00 01 F4 00 4A 00 00 AF 9C
07:38:49 DMP: 01 04 14 09 1D 05 81 00 00 09 84 00 00 FF 4F 00 00 01 F3 00 4A 00 00 1A 5C
07:38:50 DMP: 01 04 14 09 1E 05 86 00 00 09 8E 00 00 FF 50 00 00 01 F4 00 4A 00 00 CD D3
07:38:51 DMP: 01 04 14 09 1E 05 83 00 00 09 8C 00 00 FF 50 00 00 01 F3 00 4A 00 00 6C 00
07:38:51 WIF: Checking connection...
07:38:52 DMP: 01 04 14 09 1B 05 75 00 00 09 73 00 00 FF 50 00 00 01 F4 00 4A 00 00 E6 60

andker87, 2020-02-11:
I'm really sorry about bothering you, but this is very strange to me... Today Energy and Total Energy are still stuck even after `Restart 1`, and this is what is happening in the console:

> 21:25:30 MQT: tele/pzem016/STATE = {"Time":"2020-02-11T21:25:30","Uptime":"0T00:00:42","UptimeSec":42,"Heap":26,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":271,"MqttCount":1,"Wifi":{"AP":1,"SSId":"WIFI-myownwifi","BSSId":"08:6A:0A:E7:FA:CD","Channel":6,"RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:06"}}
21:25:30 MQT: tele/pzem016/SENSOR = {"Time":"2020-02-11T21:25:30","ENERGY":{"TotalStartTime":"2020-02-10T14:45:14","Total":1.858,"Yesterday":1.858,"Today":42949.672,"Period":0,"Power":413,"ApparentPower":466,"ReactivePower":215,"Factor":0.89,"Frequency":50,"Voltage":232,"Current":2.010}}
21:25:31 WIF: Checking connection...
21:25:31 DMP: 01 04 14 09 0C 07 E1 00 00 10 2C 00 00 0F B8 00 01 01 F4 00 59 00 00 9E C7
21:25:32 DMP: 01 04 14 09 0B 07 DC 00 00 10 21 00 00 0F B8 00 01 01 F4 00 59 00 00 B8 62
21:25:33 DMP: 01 04 14 09 09 07 E3 00 00 10 33 00 00 0F B8 00 01 01 F4 00 59 00 00 D9 F4
21:25:34 DMP: 01 04 14 09 09 07 E3 00 00 10 33 00 00 0F B9 00 01 01 F4 00 59 00 00 D4 64
21:25:35 DMP: 01 04 14 09 0B 07 E7 00 00 10 3C 00 00 0F B9 00 01 01 F4 00 59 00 00 FB 12
21:25:36 WIF: Checking connection...
21:25:36 DMP: 01 04 14 09 0B 07 EC 00 00 10 45 00 00 0F B9 00 01 01 F4 00 59 00 00 2E D9
21:25:37 DMP: 01 04 14 09 0C 07 DE 00 00 10 2E 00 00 0F B9 00 01 01 F4 00 59 00 00 B5 E9
21:25:38 DMP: 01 04 14 09 0C 07 D3 00 00 10 11 00 00 0F B9 00 01 01 F4 00 59 00 00 74 85
21:25:39 DMP: 01 04 14 09 0C 07 D3 00 00 10 11 00 00 0F B9 00 01 01 F4 00 59 00 00 74 85
21:25:40 DMP: 01 04 14 09 0B 07 DE 00 00 10 2B 00 00 0F B9 00 01 01 F4 00 59 00 00 0C 98
21:25:41 WIF: Checking connection...
21:25:41 DMP: 01 04 14 09 0C 07 DC 00 00 10 25 00 00 0F B9 00 01 01 F4 00 59 00 00 0E 02
21:25:42 DMP: 01 04 14 09 0D 07 CA 00 00 10 01 00 00 0F BA 00 01 01 F4 00 59 00 00 B8 8E
21:25:43 DMP: 01 04 14 09 0D 07 CA 00 00 10 01 00 00 0F BA 00 01 01 F4 00 59 00 00 B8 8E
21:25:44 DMP: 01 04 14 09 0E 07 D5 00 00 10 1B 00 00 0F BA 00 01 01 F4 00 59 00 00 E2 66
21:25:45 DMP: 01 04 14 09 0E 07 D4 00 00 10 1A 00 00 0F BA 00 01 01 F4 00 59 00 00 B1 77
21:25:45 MQT: tele/pzem016/STATE = {"Time":"2020-02-11T21:25:45","Uptime":"0T00:00:57","UptimeSec":57,"Heap":26,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":295,"MqttCount":1,"Wifi":{"AP":1,"SSId":"WIFI-myownwifi","BSSId":"08:6A:0A:E7:FA:CD","Channel":6,"RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:06"}}
21:25:45 MQT: tele/pzem016/SENSOR = {"Time":"2020-02-11T21:25:45","ENERGY":{"TotalStartTime":"2020-02-10T14:45:14","Total":1.860,"Yesterday":1.858,"Today":0.002,"Period":2,"Power":412,"ApparentPower":465,"ReactivePower":214,"Factor":0.89,"Frequency":50,"Voltage":232,"Current":2.004}}
21:25:46 WIF: Checking connection...
21:25:46 DMP: 01 04 14 09 0D 07 D7 00 00 10 1E 00 00 0F BA 00 01 01 F4 00 59 00 00 0B 47
21:25:47 DMP: 01 04 14 09 0E 07 D2 00 00 10 17 00 00 0F BA 00 01 01 F4 00 59 00 00 41 9B
21:25:48 DMP: 01 04 14 09 0E 07 D2 00 00 10 17 00 00 0F BA 00 01 01 F4 00 59 00 00 41 9B
21:25:49 DMP: 01 04 14 09 0D 07 D7 00 00 10 20 00 00 0F BA 00 01 01 F4 00 59 00 00 5C 39
21:25:50 DMP: 01 04 14 09 0B 07 D9 00 00 10 23 00 00 0F BA 00 01 01 F4 00 59 00 00 B5 11
21:25:51 WIF: Checking connection...
21:25:51 DMP: 01 04 14 09 0B 07 DA 00 00 10 21 00 00 0F BB 00 01 01 F4 00 59 00 00 4F 33
21:25:52 DMP: 01 04 14 09 0B 07 DA 00 00 10 21 00 00 0F BB 00 01 01 F4 00 59 00 00 4F 33
21:25:53 DMP: 01 04 14 09 0C 07 D4 00 00 10 17 00 00 0F BB 00 01 01 F4 00 59 00 00 D6 12
21:25:54 DMP: 01 04 14 09 0A 07 D6 00 00 10 16 00 00 0F BB 00 01 01 F4 00 59 00 00 FD 7B
21:25:55 DMP: 01 04 14 09 0A 07 CE 00 00 10 08 00 00 0F BB 00 01 01 F4 00 59 00 00 5F 62
21:25:56 WIF: Checking connection...
21:25:56 DMP: 01 04 14 09 0A 07 DB 00 00 10 23 00 00 0F BB 00 01 01 F3 00 59 00 00 91 DD
21:25:57 DMP: 01 04 14 09 0A 07 DB 00 00 10 23 00 00 0F BB 00 01 01 F4 00 59 00 00 24 1D
21:25:58 DMP: 01 04 14 09 09 07 CD 00 00 10 07 00 00 0F BB 00 01 01 F4 00 59 00 00 FF D9
21:25:59 DMP: 01 04 14 09 09 07 D4 00 00 10 14 00 00 0F BB 00 01 01 F4 00 59 00 00 1F 1D
21:26:00 DMP: 01 04 14 09 09 07 CE 00 00 10 07 00 00 0F BB 00 01 01 F4 00 59 00 00 0F 29
21:26:00 MQT: tele/pzem016/STATE = {"Time":"2020-02-11T21:26:00","Uptime":"0T00:01:12","UptimeSec":72,"Heap":26,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":327,"MqttCount":1,"Wifi":{"AP":1,"SSId":"WIFI-myownwifi","BSSId":"08:6A:0A:E7:FA:CD","Channel":6,"RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:06"}}
21:26:00 MQT: tele/pzem016/SENSOR = {"Time":"2020-02-11T21:26:00","ENERGY":{"TotalStartTime":"2020-02-10T14:45:14","Total":1.861,"Yesterday":1.858,"Today":0.003,"Period":1,"Power":410,"ApparentPower":462,"ReactivePower":213,"Factor":0.89,"Frequency":50,"Voltage":231,"Current":1.998}}

arendst, 2020-02-12:
pls provide output of command ``gpio``

andker87, 2020-02-12:
GPIO:

> {"GPIO1":{"98":"PZEM016 Rx"},"GPIO2":{"0":"None"},"GPIO3":{"62":"PZEM0XX Tx"},"GPIO4":{"0":"None"},"GPIO14":{"0":"None"}}

arendst, 2020-02-12:
Puzzled.

Pls provide output of command ``status 0``

andker87, 2020-02-12:
> {"Status":{"Module":1,"FriendlyName":["Tasmota"],"Topic":"pzem016","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":" Software/System restart ","Uptime":"0T01:36:31","StartupUTC":"2020-02-12T14:01:11","Sleep":0,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2020-02-12T11:00:42","SaveCount":22,"SaveAddress":"F4000"},"StatusFWR":{"Version":"8.1.0.7(9814468-sensors)","BuildDateTime":"2020-02-10T23:01:11","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"333/699"},"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WIFI-myownwifi",""],"TelePeriod":15,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]},"StatusMEM":{"ProgramSize":598,"Free":404,"Heap":20,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FDAE387","0437A3A4","B7FFBFCD","01DA9BC0","04B6FCC7","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,18,19,20,21,22,24,26,29,30,32","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,34,37,39,40,42,43,51,52,55,56,58,59,62"},"StatusNET":{"Hostname":"pzem016-1963","IPAddress":"192.168.1.171","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:0D:8E:AA:87:AB","Webserver":2,"WifiConfig":4,"WifiPower":17},"StatusMQT":{"MqttHost":"192.168.1.251","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AA87AB","User_mqtt":"user_mqtt","MqttCount":5,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30},"StatusTIM":{"UTC":"Wed Feb 12 15:37:42 2020","Local":"Wed Feb 12 16:37:42 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:04","Sunset":"18:04"},"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0},"StatusSNS":{"Time":"2020-02-12T16:37:42","ENERGY":{"TotalStartTime":"2020-02-12T11:00:42","Total":0.911,"Yesterday":0,"Today":0.911,"Power":264,"ApparentPower":335,"ReactivePower":206,"Factor":0.79,"Frequency":50,"Voltage":231,"Current":1.448}},"StatusSTS":{"Time":"2020-02-12T16:37:42","Uptime":"0T01:36:31","UptimeSec":5791,"Heap":18,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":5,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WIFI-myownwifi","BSSId":"08:6A:0A:E7:FA:CD","Channel":6,"RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:06"}}}

arendst, 2020-02-12:
So that is a totally different config than the first ``status 0`` report.

This looks fine. No seriallog. You might want to switch GPIO1 and GPIO3 configs to
``
{"GPIO1":{"62":"PZEM0XX Tx"},"GPIO2":{"0":"None"},"GPIO3":{"98":"PZEM016 Rx"},"GPIO4":{"0":"None"},"GPIO14":{"0":"None"}}
``
which will provide hardware serial support as you now use software serial. Not sure if it makes sense as your serial data seems to be coming correctly already.

If this is a wemos device you might als want to try other GPIO's (like 4 and 5) to see if it provides a stable solution.

In my case I use a sonoff and it is rock solid, no energy overflows (what you see as 42949.672 is just the max value a float can handle).

andker87, 2020-02-12:
I'm sorry but I'm not sure about the entire situation. The problem it's not just the increase to 42949.672, but also random reset of Today Energy value (please read previuos posts and logs).
If I understand well, the pzem provides real-time data, while the wemos/tasmota elaborates and calculates the total, so the Today Energy value is provided by tasmota, not by pzem. At least, I can let tasmota get the Total Energy value from the pzem by setting SetOption72 to 1, but this won't solve the Today Energy value problems.
Am I right?

arendst, 2020-02-12:
Yes and No. My values are fine all the time using the same software so....

andker87, 2020-02-12:
I don't doubt that it's working for you, but I'm trying to understand (going to logic) where the problem can reside in my situation. if today energy was at the end of pzem and tasmota only performs the readings, I would blame my pzem, but if tasmota reads and then elaborates, then I'm afraid there's some problem in tasmota (sorry if i'm so direct). I don't understand why else the value of today energy suddenly drops/resets.

arendst, 2020-02-12:
As the calculation makes use of the device RTC memory (To save data in case of power outages) it might get corrupted there. Did you ever try another wemos?

andker87, 2020-02-12:
no, because it's the only one I have....

arendst, 2020-02-12:
Oh, my. Perhaps get another one (I buy them by the tens) and check your power supply too.

andker87, 2020-02-12:
I only have 2 brand-new nodemcu... can I use them? can it be powered by the 5v of the pzem?

arendst, 2020-02-12:
You can use them but asking if they can be powered from the PZem makes me wary....

You should not power the wemos/nodemcu from the PZem. You also need a converter between the Pzem and the wemos/NodeMcu converting RS584 signals to 3V3 for the wemos/NodeMcu.

Pls revisit your connections and power environment.

andker87, 2020-02-12:
Sorry but I don't understand. I powered the wemos d1 mini from the 5v output of the pzem-016 to the 5gpio... 
[image]

andker87, 2020-02-27:
Solved by switching cables and gpios for tx-rx.No anomalous values or hardware watchdog restarts since 9 days.
solved from [here ](https://github.com/arendst/Tasmota/issues/5875#issuecomment-591828786)

arendst, 2020-02-27:
Thx for reporting back. Appreciated.

Lantzie, 2020-04-19:
> Solved by switching cables and gpios for tx-rx.No anomalous values or hardware watchdog restarts since 9 days.
> solved from [here ](https://github.com/arendst/Tasmota/issues/5875#issuecomment-591828786)

I have the same problem and I am not clear about your solution: Have you simply used other different gpios?

andker87, 2020-04-19:
> I have the same problem and I am not clear about your solution: Have you simply used other different gpios?

No, I just switched the cables RX-TX in the wemos (and switched gpios tx-rx in tasmota‘s config)

gammerboy, 2020-06-28:
> > I have the same problem and I am not clear about your solution: Have you simply used other different gpios?
> 
> No, I just switched the cables RX-TX in the wemos (and switched gpios tx-rx in tasmota‘s config)

Does this mean changing it to hardware serial fixed the issue  ?

I'm having a issue very similar to this where the energy value went up very high. I've connected it to d1 and d2 pins of a d1 mini. Just wanted to verify if switching to hardware serial gets this issue fixed.