arendst/Tasmota #7676: Backlog Color 1; Scheme + not working as intended

blakadder, 2020-02-05:
### PROBLEM DESCRIPTION
`Backlog Color1; Scheme +` stays stuck on Scheme 2

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): BW-LT27
- [ ] Tasmota binary firmware version number used: 8.1.0.5
  - [ ] Pre-compiled
  - [x ] Self-compiled
    - [x ] IDE / Compiler used: PlatformIO
- [x ] Flashing tools used: _____
```

```
- [ ] Provide the output of this command: `Status 0`:
```
23:37:22 CMD: status 0
23:37:22 RSL: stat/BW-LT27/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"BW-LT27","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:37:22 RSL: stat/BW-LT27/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://10.1.1.2:8000/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T02:04:40","StartupUTC":"2020-02-05T20:32:42","Sleep":50,"CfgHolder":4617,"BootCount":8,"SaveCount":325,"SaveAddress":"F7000"}}
23:37:22 RSL: stat/BW-LT27/STATUS2 = {"StatusFWR":{"Version":"8.1.0.5(tasmota)","BuildDateTime":"2020-02-05T03:12:20","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"287/699"}}
23:37:22 RSL: stat/BW-LT27/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HTPC",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
23:37:22 RSL: stat/BW-LT27/STATUS4 = {"StatusMEM":{"ProgramSize":508,"Free":492,"Heap":28,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","8FCAE187","042683A1","200217C5","010001C0","00007881"],"Drivers":"1,2,3,4,5,6,8,9,10,12,16,18,19,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,26"}}
23:37:22 RSL: stat/BW-LT27/STATUS5 = {"StatusNET":{"Hostname":"BW-LT27-5617","IPAddress":"10.1.1.129","Gateway":"10.1.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.1.1.1","Mac":"50:02:91:CA:15:F1","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
23:37:22 RSL: stat/BW-LT27/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CA15F1","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
23:37:22 RSL: stat/BW-LT27/STATUS7 = {"StatusTIM":{"UTC":"Wed Feb 05 22:37:22 2020","Local":"Wed Feb 05 23:37:22 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:15","Sunset":"17:53"}}
23:37:22 RSL: stat/BW-LT27/STATUS10 = {"StatusSNS":{"Time":"2020-02-05T23:37:22"}}
23:37:22 RSL: stat/BW-LT27/STATUS11 = {"StatusSTS":{"Time":"2020-02-05T23:37:22","Uptime":"0T02:04:40","UptimeSec":7480,"Heap":28,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":100,"Color":"00A6FF0000","HSBColor":"201,100,100","White":0,"CT":300,"Channel":[0,65,100,0,0],"Scheme":2,"Fade":"ON","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"HTPC","BSSId":"50:64:2B:5B:41:59","Channel":6,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_

```
23:31:32 RSL: stat/BW-LT27/RESULT = {"Scheme":4}
23:31:46 CMD: backlog color -; scheme +
23:31:47 RSL: stat/BW-LT27/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF00AA0000","HSBColor":"320,100,100","White":0,"CT":300,"Channel":[100,0,67,0,0]}
23:31:47 RSL: stat/BW-LT27/RESULT = {"Scheme":2}
23:31:47 RSL: tele/BW-LT27/RESULT = {"Time":"2020-02-05T23:31:47","IrReceived":{"Protocol":"UNKNOWN","Bits":50,"Hash":"0x348F82B1"}}
23:31:50 CMD: backlog color 1; scheme +
23:31:51 RSL: stat/BW-LT27/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF00000000","HSBColor":"0,100,100","White":0,"CT":300,"Channel":[100,0,0,0,0]}
23:31:51 RSL: stat/BW-LT27/RESULT = {"Scheme":2}
23:31:56 CMD: backlog color 1; scheme +
23:31:56 RSL: stat/BW-LT27/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF00000000","HSBColor":"0,100,100","White":0,"CT":300,"Channel":[100,0,0,0,0]}
23:31:57 RSL: stat/BW-LT27/RESULT = {"Scheme":2}
23:32:01 CMD: backlog color 1; scheme +
23:32:01 RSL: stat/BW-LT27/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF00000000","HSBColor":"0,100,100","White":0,"CT":300,"Channel":[100,0,0,0,0]}
23:32:01 RSL: stat/BW-LT27/RESULT = {"Scheme":2}
23:32:05 CMD: backlog color 1; scheme +
23:32:05 RSL: stat/BW-LT27/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF00000000","HSBColor":"0,100,100","White":0,"CT":300,"Channel":[100,0,0,0,0]}
23:32:06 RSL: stat/BW-LT27/RESULT = {"Scheme":2}
23:32:09 CMD: backlog color 1; scheme +
23:32:09 RSL: stat/BW-LT27/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF00000000","HSBColor":"0,100,100","White":0,"CT":300,"Channel":[100,0,0,0,0]}
23:32:09 RSL: stat/BW-LT27/RESULT = {"Scheme":2}
```

Scheme - skipping to 0
```
23:39:14 CMD: scheme 4
23:39:14 RSL: stat/BW-LT27/RESULT = {"Scheme":4}
23:39:17 CMD: backlog color 1; scheme -
23:39:17 RSL: stat/BW-LT27/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF00000000","HSBColor":"0,100,100","White":0,"CT":300,"Channel":[100,0,0,0,0]}
23:39:17 RSL: stat/BW-LT27/RESULT = {"Scheme":0}
```

### TO REPRODUCE
use `backlog color 1; scheme +`

### EXPECTED BEHAVIOUR
Increment Scheme number to 3 then 4

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

arendst, 2020-02-06:
As designed.

Scheme -/+ rotate in this case between 0 (fixed color), 2 (auto color up), 3 (auto color down) and 4 (random color).

- Scheme 0 allows a user to set a fixed color using commands like ``color 1``
- Schemes 2, 3, and 4 select their own colors and rotate/randomize between them. 

As soon as you select a fixed color using the ``color`` command the auto schemes will be reset to scheme 0 which is the user selectable color scheme.

So your command sequence of ``color 1`` followed by ``scheme +`` does execute ``color 1`` and reset scheme to 0 as that is what you want, a fixed color, and then increments the scheme to 2.

arendst, 2020-02-07:
To continue our secret comms.

> why does scheme 4 not work if youre in CT mode or on color 12

- CT mode is no RGB mode. Schemes only work in RGB mode.
- Once executing command ``color 12`` (or any ``color`` command) the scheme is set back to scheme 0 as you obviously want to control the color yourselves and NOT by schemes.

arendst, 2020-02-07:
Ah. The ``color 12`` or ``color FFFFFF`` is a nice one. It does fail all schemes indeed.

Investigating...

arendst, 2020-02-07:
And another epiphany. It doesn't seem to work because Saturation in that case is 0. So the schemes work as designed, due to Saturation being set to 0 you won't see any color change. 

This was a design decision to have user control over Saturation during active schemes. So changing Saturation to anything but 0, preferrably 100 you can re-enjoy the features of schemes.

closed 2020-02-10 by blakadder