arendst/Tasmota #7677: Add a function to make it imposible to turn on RGB and W at the same time

DominikEisele03, 2020-02-05:
**Have you looked for this feature in other issues and in the docs?**  
Yep and hope i didn´t overlook anything ;)
**Is your feature request related to a problem? Please describe.**  
Yes and no. It is related to a Problem but it isn´t caused by tasmota.
**Describe the solution you'd like**  
Hi :) I am having the following problem: I succesfully flashed some Tuya lamps with Tasmota and i am realy happy with it! however it seems like the bulbs don´t like if RGB and W is turned on at the same time. The lamps are getting extremely hot until they crash and completly lose connetion. This happens if you combind the white channel with any other color for around 5 minutes. I think it is caused by the transformer inside of the lamps (it cannot produce the power to light up all the channels). I have also heard of some lamps that will imidiatly die if you torn on both at the same time. 
So normaly the channels are seperated already but it would be awsome to have one SetOption to completly lock this. For example with the command "color FFFFFFFFF" i am always able to turn on both channels at the same time. 
I am just afraid to send such a color code by mistake and to kill a lamp that way. Thanks for reading that!

**Describe alternatives you've considered**  
I think the best way would be to add an SetOption command to lock RGB (PWM1,PWM2,PWM3) completly from White (PWM4, maybe PWM5)

**Additional context**  
Thanks for reading all this text and sorry for my bad english XD.
It would be awsome if someone cold make something like that!

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-02-07:
With `SetOption37 0`, all normal commands will automatically switch between RGB and White: `Dimmer`, `CT`, `HSBColor` and all web UI sliders.
The only way to set all channels at once is to use `Color FFFFFFFFFF`. This was an explicit ask from many users. If you don't want to set all channels, just don't send this command.
I'm reluctant to add yet another `SetOption`, configuration is already complex enough.

DominikEisele03, 2020-02-08:
Thanks for your answer! How about a diffrent command name? It wouldn´t necessary need to be a `SetOption` command of course to not make the list any longer (could be like the LedTable command or something like that). Another idea would be to make an option to change the `color` to RGB only (without including a value for WW, CW). That should automaticly solve the problem and would possible take more sense then my first idea to just block it. Have a nice Day!

ascillato2, 2020-02-08:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-02-08 by ascillato2

DerNick, 2020-02-10:
I don’t think it’s answered🤔😕

ascillato, 2020-02-10:
Yes, it is answered at https://github.com/arendst/Tasmota/issues/7677#issuecomment-583594771