arendst/Tasmota #7684: Too many MQTT power reports for no reason

machadof1, 2020-02-06:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
In version 8.1.0, in some Blitzwolf BW-SHP8 I am seeing too many MQTT power reports for no reason(see console output below), with a powerdelta of 20, I get reports with no change of the active power (always 3)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Blitzwolf BW-SHP8_____
- [X] Tasmota binary firmware version number used:8.1.0 _____
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: tasmotizer-1.1_____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
18:43:13 CMD: Backlog Template; Module; GPIO 255
18:43:13 MQT: tomada_este_sala_tv/stat/RESULT = {"NAME":"SHP8","GPIO":[0,56,0,17,134,132,0,0,131,53,21,0,0],"FLAG":0,"BASE":64}
18:43:14 MQT: tomada_este_sala_tv/stat/RESULT = {"Module":{"0":"SHP8"}}
18:43:14 MQT: tomada_este_sala_tv/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"56":"Led1i"},"GPIO2":{"0":"None"},"GPIO3":{"17":"Button1"},"GPIO4":{"134":"BL0937 CF"},"GPIO5":{"132":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"131":"HLWBL SELi"},"GPIO13":{"53":"Led2"},"GPIO14":{"21":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
18:46:29 MQT: tomada_este_sala_tv/stat/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
18:46:30 MQT: tomada_este_sala_tv/stat/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
18:46:30 MQT: tomada_este_sala_tv/stat/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
18:47:22 MQT: tomada_este_sala_tv/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["Tomada Este Sala TV"],"Topic":"tomada_este_sala_tv","ButtonTopic":"0","Power":1,"PowerOnState":4,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:47:22 MQT: tomada_este_sala_tv/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"15T00:28:07","StartupUTC":"2020-01-22T18:19:15","Sleep":50,"CfgHolder":4617,"BootCount":25,"SaveCount":240,"SaveAddress":"F6000"}}
18:47:22 MQT: tomada_este_sala_tv/stat/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":7,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"353/699"}}
18:47:22 MQT: tomada_este_sala_tv/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000AA009","2805C8000100060000005AFF000000000000","00000200","00000000"]}}
18:47:22 MQT: tomada_este_sala_tv/stat/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
18:47:22 MQT: tomada_este_sala_tv/stat/STATUS5 = {"StatusNET":{"Hostname":"xxxx","IPAddress":"xxxxx","Gateway":"xxxxx","Subnetmask":"255.255.255.0","DNSServer":"xxxx1","Mac":"xxxx","Webserver":2,"WifiConfig":4}}
18:47:22 MQT: tomada_este_sala_tv/stat/STATUS6 = {"StatusMQT":{"MqttHost":"xxxx","MqttPort":1883,"MqttClientMask":"xxxxx","MqttClient":"xxx","MqttUser":"xxxx","MqttCount":4,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
18:47:22 MQT: tomada_este_sala_tv/stat/STATUS7 = {"StatusTIM":{"UTC":"Thu Feb 06 18:47:22 2020","Local":"Thu Feb 06 18:47:22 2020","StartDST":"Sun Mar 29 01:00:00 2020","EndDST":"Sun Oct 25 02:00:00 2020","Timezone":99,"Sunrise":"07:13","Sunset":"16:54"}}
18:47:22 MQT: tomada_este_sala_tv/stat/STATUS9 = {"StatusPTH":{"PowerDelta":20,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:47:22 MQT: tomada_este_sala_tv/stat/STATUS10 = {"StatusSNS":{"Time":"2020-02-06T18:47:22","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.048,"Yesterday":0.092,"Today":0.067,"Power":3,"ApparentPower":9,"ReactivePower":8,"Factor":0.30,"Voltage":227,"Current":0.038}}}
18:47:22 MQT: tomada_este_sala_tv/stat/STATUS11 = {"StatusSTS":{"Time":"2020-02-06T18:47:22","Uptime":"15T00:28:07","UptimeSec":1297687,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":4,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxx","BSSId":"xxxx","Channel":9,"RSSI":58,"Signal":-71,"LinkCount":2,"Downtime":"0T00:00:47"}}}
```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

18:21:46 MQT: tomada_este_sala_tv/tele/SENSOR = {"Time":"2020-02-06T18:21:45","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.047,"Yesterday":0.092,"Today":0.066,"Power":3,"ApparentPower":9,"ReactivePower":9,"Factor":0.31,"Voltage":228,"Current":0.041}}
18:21:46 MQT: tomada_este_sala_tv/tele/SENSOR = {"Time":"2020-02-06T18:21:46","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.047,"Yesterday":0.092,"Today":0.066,"Power":3,"ApparentPower":9,"ReactivePower":9,"Factor":0.33,"Voltage":229,"Current":0.041}}
18:21:48 MQT: tomada_este_sala_tv/tele/SENSOR = {"Time":"2020-02-06T18:21:48","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.047,"Yesterday":0.092,"Today":0.066,"Power":3,"ApparentPower":9,"ReactivePower":9,"Factor":0.30,"Voltage":229,"Current":0.039}}
18:21:58 MQT: tomada_este_sala_tv/tele/SENSOR = {"Time":"2020-02-06T18:21:58","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.047,"Yesterday":0.092,"Today":0.066,"Power":3,"ApparentPower":9,"ReactivePower":8,"Factor":0.38,"Voltage":228,"Current":0.038}}
18:21:59 MQT: tomada_este_sala_tv/tele/SENSOR = {"Time":"2020-02-06T18:21:59","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.047,"Yesterday":0.092,"Today":0.066,"Power":3,"ApparentPower":9,"ReactivePower":8,"Factor":0.32,"Voltage":228,"Current":0.038}}
18:22:05 MQT: tomada_este_sala_tv/tele/SENSOR = {"Time":"2020-02-06T18:22:05","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.047,"Yesterday":0.092,"Today":0.066,"Power":3,"ApparentPower":9,"ReactivePower":8,"Factor":0.34,"Voltage":227,"Current":0.039}}
18:22:06 MQT: tomada_este_sala_tv/tele/SENSOR = {"Time":"2020-02-06T18:22:06","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.047,"Yesterday":0.092,"Today":0.066,"Power":3,"ApparentPower":9,"ReactivePower":8,"Factor":0.33,"Voltage":227,"Current":0.039}}
18:22:39 MQT: tomada_este_sala_tv/tele/SENSOR = {"Time":"2020-02-06T18:22:39","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.047,"Yesterday":0.092,"Today":0.066,"Power":3,"ApparentPower":9,"ReactivePower":8,"Factor":0.35,"Voltage":227,"Current":0.039}}
18:22:41 MQT: tomada_este_sala_tv/tele/SENSOR = {"Time":"2020-02-06T18:22:41","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.047,"Yesterday":0.092,"Today":0.066,"Power":3,"ApparentPower":9,"ReactivePower":9,"Factor":0.31,"Voltage":226,"Current":0.042}}
18:22:43 MQT: tomada_este_sala_tv/tele/SENSOR = {"Time":"2020-02-06T18:22:43","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.047,"Yesterday":0.092,"Today":0.066,"Power":3,"ApparentPower":9,"ReactivePower":9,"Factor":0.33,"Voltage":226,"Current":0.041}}
[... cut ...] "ReactivePower":8,"Factor":0.39,"Voltage":226,"Current":0.039}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Nothing needed

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
MQTT reports only active power differs 20%

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-02-07:
Set ``WattRes 2``, ``Voltres 1`` and ``AmpRes 3`` and enjoy the source of your many power reports.

machadof1, 2020-02-07:
I assumed the powerdelta was calculated after applying the Power sensor resolution, but even so the math is off, in the example below, after setting the Power sensor resolution to 2, you can see that the report is triggered with percentages differences below 20% (the powerdelta is set at 20), some as little as 5%.
16:11:36 MQT: tomada_este_sala_tv/tele/SENSOR = {"Time":"2020-02-07T16:11:36","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.135,"Yesterday":0.093,"Today":0.060,"Power":4.80,"ApparentPower":9.35,"ReactivePower":8.00,"Factor":0.51,"Voltage":228,"Current":0.041}}
16:11:38 MQT: tomada_este_sala_tv/tele/SENSOR = {"Time":"2020-02-07T16:11:38","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.135,"Yesterday":0.093,"Today":0.060,"Power":3.60,"ApparentPower":10.56,"ReactivePower":9.90,"Factor":0.34,"Voltage":230,"Current":0.046}}
16:11:46 MQT: tomada_este_sala_tv/tele/SENSOR = {"Time":"2020-02-07T16:11:46","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.135,"Yesterday":0.093,"Today":0.060,"Power":4.20,"ApparentPower":9.86,"ReactivePower":8.90,"Factor":0.43,"Voltage":229,"Current":0.043}}
16:11:47 MQT: tomada_este_sala_tv/tele/SENSOR = {"Time":"2020-02-07T16:11:47","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.135,"Yesterday":0.093,"Today":0.060,"Power":3.60,"ApparentPower":9.86,"ReactivePower":9.20,"Factor":0.37,"Voltage":229,"Current":0.043}}
16:12:37 MQT: tomada_este_sala_tv/tele/SENSOR = {"Time":"2020-02-07T16:12:37","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.135,"Yesterday":0.093,"Today":0.060,"Power":4.40,"ApparentPower":9.58,"ReactivePower":8.50,"Factor":0.46,"Voltage":228,"Current":0.042}}
16:12:39 MQT: tomada_este_sala_tv/tele/SENSOR = {"Time":"2020-02-07T16:12:39","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.135,"Yesterday":0.093,"Today":0.060,"Power":3.70,"ApparentPower":10.04,"ReactivePower":9.30,"Factor":0.37,"Voltage":228,"Current":0.044}}
16:13:11 MQT: tomada_este_sala_tv/tele/SENSOR = {"Time":"2020-02-07T16:13:11","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.135,"Yesterday":0.093,"Today":0.060,"Period":0.31,"Power":3.50,"ApparentPower":9.50,"ReactivePower":8.80,"Factor":0.37,"Voltage":232,"Current":0.041}}
16:13:50 MQT: tomada_este_sala_tv/tele/SENSOR = {"Time":"2020-02-07T16:13:50","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.135,"Yesterday":0.093,"Today":0.060,"Power":4.10,"ApparentPower":9.64,"ReactivePower":8.70,"Factor":0.43,"Voltage":230,"Current":0.042}}
16:13:51 MQT: tomada_este_sala_tv/tele/SENSOR = {"Time":"2020-02-07T16:13:51","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.135,"Yesterday":0.093,"Today":0.060,"Power":3.60,"ApparentPower":9.64,"ReactivePower":8.90,"Factor":0.37,"Voltage":230,"Current":0.042}}
16:14:50 MQT: tomada_este_sala_tv/tele/SENSOR = {"Time":"2020-02-07T16:14:50","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.135,"Yesterday":0.093,"Today":0.060,"Power":4.20,"ApparentPower":9.46,"ReactivePower":8.50,"Factor":0.44,"Voltage":231,"Current":0.041}}
16:14:53 MQT: tomada_este_sala_tv/tele/SENSOR = {"Time":"2020-02-07T16:14:53","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.135,"Yesterday":0.093,"Today":0.060,"Power":3.80,"ApparentPower":9.92,"ReactivePower":9.20,"Factor":0.38,"Voltage":231,"Current":0.043}}
16:15:53 MQT: tomada_este_sala_tv/tele/SENSOR = {"Time":"2020-02-07T16:15:53","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.135,"Yesterday":0.093,"Today":0.060,"Power":4.10,"ApparentPower":10.05,"ReactivePower":9.20,"Factor":0.41,"Voltage":229,"Current":0.044}}
16:15:54 MQT: tomada_este_sala_tv/tele/SENSOR = {"Time":"2020-02-07T16:15:54","ENERGY":{"TotalStartTime":"2019-11-05T00:00:00","Total":9.135,"Yesterday":0.093,"Today":0.060,"Power":3.80,"ApparentPower":10.05,"ReactivePower":9.30,"Factor":0.38,"Voltage":229,"Current":0.044}}
16:16:06 CMD: powerdelta
16:16:06 MQT: tomada_este_sala_tv/stat/RESULT = {"PowerDelta":20}

Watt	Diff
4.8	
3.6	75%
4.2	117%
3.6	86%
4.4	122%
3.7	84%
3.5	95%
4.1	117%
3.6	88%
4.2	117%
3.8	90%
4.1	108%
3.8	93%

arendst, 2020-02-07:
At these low power values do not expect your hardware to be that accurate. The measured values may well be spreading even more.

Try the same with at least a load of 20W and see what happens then.

machadof1, 2020-02-07:
I understand that the values might bee wrong, but aren't the calculations made with the wrong values? Ok instead of 1.2W real value the hardware is reading 1.3 or 1.5 or 2W , isn't the 1.3, 1.5 or 2W the value used in the calculations?
I collected more log values and in 290 readings I got differences as low as 3%.

machadof1, 2020-02-13:
Can you tell me the # of issue that is duplicated ?

ascillato2, 2020-02-13:
> Can you tell me the # of issue that is duplicated ?

Yes, no problem. I can do the search for you:

#7515 
#7509 
#4905 
#7050 
#5901 
#2891 
#5858 
#2157 
#5828
#4813
#4130
#4145
#3795
#3778
#3660
#3648

As Theo has explained, below 5W, the power measurement is not accurate and very noisy (hardware limitation). Also, Tasmota uses the last 3 seconds of power data for reporting.

Workarounds for your case:

https://github.com/arendst/Tasmota/issues/2880#issuecomment-416078047

https://github.com/arendst/Tasmota/issues/5897#issuecomment-497786077

Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-02-13 by ascillato2

machadof1, 2020-02-14:
I didn't know that the reports were an average of the last 3 seconds of data, in that case the behavior makes sense if trigger for the reports is the powerdelta % value of the live data and not the averaged data.