arendst/Tasmota #7686: Blinds are reporting ShutterPosition = 0 during movement, should be sth other

buenyamin-olgun, 2020-02-06:
**Have you looked for this feature in other issues and in the docs?**  
I couldn't found anything similiar.

**Is your feature request related to a problem? Please describe.**  
Currently I'm developing a nodejs plugin which should utilize the HTTP API for blinds movement.
During any movings it should report, that it is currently moving.

**Describe the solution you'd like**  
Right now if you query the endpoint ```/cm?cmnd=ShutterPosition``` during a movement it is reporting ```{"ShutterPosition1":0}```. There's no way to differenciate between movement and that the position is really 0. If it would report sth. like -1 or a string "moving" this would be helpful.

**Describe alternatives you've considered**  
There could be a endpoint more added which shows the current state like "moving", "stopped".

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

schrej-zz, 2020-02-11:
I would suggest a new endpoint publishing the movement direction: UP, DOWN, STOP (or NONE)

buenyamin-olgun, 2020-02-11:
> I would suggest a new endpoint publishing the movement direction: UP, DOWN, STOP (or NONE)

The advantage of a "ShutterTargetPosition" property would be, that you have a accurate target position and you can also get knowledge of the direction very easily.

Example code for getting direction:
```
if(ShutterPosition1 === ShutterTargetPosition1){
   Movement = 'STOP'
} elseif(ShutterPosition1 > ShutterTargetPosition1){
   Movement = 'DOWN';
} elseif(ShutterPosition1 < ShutterTargetPosition1){
   Movement = 'UP';
}

```

schrej-zz, 2020-02-11:
I just saw, that Tasmota is already reporting the direction at least in MQTT:
tele/Rollo/SENSOR = {"Time":"2020-02-11T18:38:28","Shutter1":{"Position":100,"Direction":0}}
where -1 is down,1 is up and 0 is stop.
Maybe this can be just added to the shutterposition command reply.
It's a JSON anyway and would be consistent to the MQTT message.

johan149, 2020-02-14:
same thing here is not reporting the position in realtime just shows the 100% or 0% but not in realtime as before.

schrej-zz, 2020-02-15:
@johan149: I think, your topic is not related to this issue. @buenyamin-olgun is complaining about lack of movement/direction information on HTTP requests. I think, that console command, HTTP request should be identical to the MQTT message.
(JSON position and direction).
Maybe you should open another issue for your topic and exactly describe your issue. In MQTT the driver reports position at every second in stat/RESULT during movement. Not good from my point of view. It should send the tele/SENSOR message. But again. Off Topic from my point of view.

stefanbode, 2020-02-16:
Looks like something broken here. Agree it worked before

stefanbode, 2020-02-16:
ok, the MQTT is still reporting. therefore all fine. what you want is that shutterposition also give the direction in a JSON string. This will break some implementation that relies on the current behaviour. Have you take a look at the rules section there are events triggered AFTER the movement and during the movement. I guess this can solve your problem

schrej-zz, 2020-02-16:
@stefanbode, why should this break implementations? An additional parameter in a JSON should not break anything.
Regarding MQTT: Yes, it is reported every second, but via stat/RESULT. But you cannot use stat/Result as MQTT input, as this will lead to error messages due to other RESULT messages (e.g. "Power2"). I would suggest to either send the tele/SENSOR or the SHUTTERx message every second.
This is a little off topic and might be a topic for another issue.

schrej-zz, 2020-02-17:
> @stefanbode, why should this break implementations? An additional parameter in a JSON should not break anything.
> Regarding MQTT: Yes, it is reported every second, but via stat/RESULT. But you cannot use stat/Result as MQTT input, as this will lead to error messages due to other RESULT messages (e.g. "Power2"). I would suggest to send the tele/SENSOR message.
> This is a little off topic and might be a topic for another issue.
Just to make clear, what I mean. It works as you see the position updated every second e.g. in BasicUI.
But you get permanent error messages in openhab.log due to the lack of the "Shutter1.Position" item in other RESULT messages.
`2020-02-17 18:39:11.639 [WARN ] [.mqtt.internal.MqttMessageSubscriber] - Error processing MQTT message.
org.openhab.core.transform.TransformationException: Invalid path **'$.Shutter1.Position'** in '{"Time":"2020-02-17T18:39:11","Uptime":"0T03:44:36","UptimeSec":13476,"Vcc":3.412,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Schreiner","BSSId":"7C:FF:4D:D0:F3:C7","Channel":11,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:07"}}'
`

buenyamin-olgun, 2020-02-17:
> ok, the MQTT is still reporting. therefore all fine. what you want is that shutterposition also give the direction in a JSON string. This will break some implementation that relies on the current [...]

What I really wanted was that ShutterPosition1 gets unequivocally during movements.

Currently it is reporting ShutterPosition1 = 0 during movement and also if it is at position 0 which makes every blind detection implementation very difficult.

My suggestion was that it could report the real current position during movement sth like:
start: 90; current: 80; target: 70

schrej-zz, 2020-02-18:
@buenyamin-olgun. That doesn't make sense from my point of view.
1. Target position might change during movement.
2. Why reinventing the wheel? Position and Direction are already existing parameters and are updated every second while moving.
The issue I see is, that these parameters have to be made available not just in MQTT and in the logs, but also in the Shutterposition(x) command.

to-scho, 2020-02-24:
@schrej-zz, optional or "sometimes missing" JSON attributes are a common problem not only for shutters. I had same issue in home assistant with a solution stated here (Another Home Assistant integration example with position update while movement (Tasmota versions >= v8.1.0.5)): https://tasmota.github.io/docs/#/Blinds-and-Shutters?id=home-assistant-support.

stefanbode, 2020-02-24:
I will commit an update later this day. Shutterposition without a parameter will the report the last known position. Additionally the rule support will be enhanced. After starting the movement also the target position will be reported in json for using in rules. This report will be new. The final report after stop will be the same.

schrej-zz, 2020-02-24:
@to-scho, your example seems to work with HA. But for example Openhab cannot deal with changing payload for one command (at least not that I'm aware of).
As the current stat/RESULT implementation sends various JSON payloads, OpenHab cannot deal with it and logs error messages.
My suggestion therefore is to send the already existing (and with fixed payload) tele/SENSOR message every second while moving instead of stat/RESULT.
BTW: Also from the MQTT definition of **stat** and **tele**, **tele** would be the right message and not **stat**

ascillato2, 2020-02-24:
Closing this issue as it is solved.

Thanks for sharing your ideas.

closed 2020-02-24 by ascillato2

schrej-zz, 2020-02-24:
New "Target" value ignores SHUTTERINVERT.