arendst/Tasmota #7688: LedPowerX should not do LedState0 for LED that's not assigned

dseven, 2020-02-06:
**Have you looked for this feature in other issues and in the docs?**  

Yes.

**Is your feature request related to a problem? Please describe.**  

I have a Aoycocr SW1 light switch. It has one relay and three LEDs. I would like to use the first LED for the LedLink, the second LED for the relay state, and the third LED for something else (in rules). There doesn't seem to be any way to set the state of the third LED without forgoing LedState functionality - if I do "LedPower2 1", it implies "LedState 0", so *all* LEDs become unmanaged, including the LedLink functionality.

**Describe the solution you'd like**  

When setting LedPowerX Y for an LED that is not associated with a relay and not assigned as the LedLink, it should simply turn that LED on or off - it should not imply "LedState 0"

**Describe alternatives you've considered**  

My current hack is to assign the pin associated with the third LED as "Relay" instead of a "Led", then I can turn it on/off with the "Power2" command... but this has unfortunate side-effects, like changing the MQTT topics to use "POWER1" and "POWER2" instead of just "POWER", and it's just kludgy.

ascillato, 2020-02-07:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-02-07 by ascillato2

dseven, 2020-02-07:
I think you're missing the point. If LedState=0, the LedLink no longer functions (i.e. to alert if the WiFi or MQTT connection is lost). It may be possible to reinvent the LedLink functionality with rules, but I should not have to do that.