arendst/Tasmota #7690: Add WPA2 to build with a default password that can be overridden in user_config

ankurv86, 2020-02-07:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I earlier assumed that it is difficult to implement WPA2 in ESP8266 but I was reading Arduino docs and saw that enabling it just using the overloaded Wifi.SoftAp method away.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Enable WPA2 in default build or release it as a secure build.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
None.

**Additional context**  
_Add any other context or screenshots about the feature request here._
Not enabling WPA2 by default is a huge security flaw. Someone around my house can write a network sniffer that sniffs on http traffic through non-secure wifi networks and use that as seed to break into my wifi ecosystem. What's more if someone uses restore configuration they can leak more than just the ssid credentials. This is further complicated by lack of https in web server so wpa2 is the only way to protect the whole chain. To make matters even worse if I change my home wifi ssid, even then every time I add a new device I will be leaking my latest password out in the open.
My modified build already does this but this should be default even if with a default password.
Documentation : https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/soft-access-point-class.html

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-02-07:
WPA2 is enabled

eorl-the-young, 2020-02-07:
Not as per my experience. If I download one of the release .bin file and deploy it on one of my devices using tuya-convert. The updated device creates a Wireless AP for initial configuration with the name Tasmota-XXXX. This network is unsecured and has no password as in anyone can connect to it.

Please correct me if I am using the wrong bin. Looking in the code I see this line :
https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_01_webserver.ino#L643
which initializes the AP with no password. The build I am using I have replaced this line with 

WiFi.softAP(my_hostname, <Default Password>, channel, false, 1);

The default password is 8 characters or more long to enable WPA2 and max connection is 1 so that at most 1 device(Configuration device) can connect to it.

Jason2866, 2020-02-07:
This AP is just and only for one time initial config.

eorl-the-young, 2020-02-08:
That actually is not my concern. My concern is that the whole chain of events required to securely deploy a Tasmota device has a vulnerability at the root of the whole process. I understand this is true for most of Iot devices available today but still if the solution is this easy why not use it. To explain my concern in detail let me explain the scenario : 

1) A bad entity(let's say X) is somewhere around my house. He runs a wifi sniffer that connects to any wifi AP that is not secured. On top of that as soon as it connects to any such AP it runs Wireshark on that network in promiscous mode(trace all network calls and read every http request in clear text). This is all possible with Tasmota default AP as it is neither secured not does it restrict the number of devices that can connect to this AP to 1.
2) I get a tuya device and use tuya convert to install tasmota bin on the device. It starts a tasmota-xxxx AP and I along with X connects to it.
3) At this point I can do 2 things : 
     a) Put my wifi ssid and password in the webserver that will be transmitted in clear text to be sniffed by X
     b) Restore configuration in which case along with my SSID and Password I can also expose my broker credentials which will allow X to know a lot more about my network. 

At the minimum what could be done is 
1) Default AP starts with WPA2 and a default password. This password should be set using user_config
2) Max connections with default AP should be restricted to 1.

This can all be achieved by just changing softAp initialization to 
WiFi.softAP(my_hostname, <Default password> , channel, false, 1);

This will actually provide more security than adding a password to web interface as the interface does not support SSL and would still be open to the attack defined above.

meingraham, 2020-02-08:
@eorl-the-young 

In your scenario, the bad actor needs to know when you are flashing Tasmota (via serial) or running Tuya-Convert and lurks until the Tasmota device boots into AP mode and then hijacks your device during that 30-60 second window. Yes, this is possible. Probable?

As for restoring the configuration, the Wi-Fi credentials in the binary `.dmp` files are not clear text. If you use the decode-config tool and choose to dump the configuration to a JSON file, decode_config allows you to not include the Wi-Fi credentials in the JSON file.

As with anything, the decision to implement this kind of feature comes to weighing the risk against the benefit and cost (in additional consumed program memory and time to implement). One thing that can alleviate the cost would be for _**you**_ to create a pull request with the feature you are suggesting implemented and fully tested. Then @arendst can assess the implementation and decide whether to merge it into Tasmota.

Regards.

Mike

eorl-the-young, 2020-02-08:
Thanks @meingraham for your response. I was looking more for an explanation as everything is a judgement and thanks for providing that. 
Also thanks for explaining how config works and that is definitely better than setting ssid using the web interface(query parameters over http really scare me).

Will close this issue as I anyways use a custom version of tasmota. Was just curious about why this is not default(but I can say the same about why TLS mqqt is not enabled by default).

Just as an end note, if possible please add somewhere in documentation how to enable WPA2 as I had to read arduino documentation to figure it out.

closed 2020-02-08 by eorl-the-young

joba-1, 2020-02-08:
my 2ct: looks like a good strategy with no cost

meingraham, 2020-02-08:
@eorl-the-young 

The documentation is open source. Anybody can contribute to it. Since you now know how to activate the setting, you could add your knowledge to the documentation.

Cheers!

eorl-the-young, 2020-02-11:
Thanks @arendst. Just limiting number of users to 1 even without password reduces the concern greatly. 
My first ever public issue and a really positive experience from the community.